OpenCPN/OpenCPN #4377: AIS: don't trigger CPA alerts while anchored/stopped.

ocornu, 2025-02-20:
**Is your feature request related to a problem? Please describe.**
I'm currently in harbor, working on OpenCPN, and i get a lot of AIS CPA alerts from incoming vessels. It's annoying in my case, and it seems rather useless in any case.

**Describe the solution you'd like**
Just like there is an option to remove CPA alerts with targets vessels that are anchored/stopped, i'd like a similar option to remove those alerts when my own boat is anchored/stopped.
It would probably be a good idea to turn it this option on by default, as i can't figure a case where one would like to receive such alerts. If such corner cases exist, it seems they would be rather exceptional and could justify manually selecting an optional feature.

Hakansv, 2025-02-20:
There could be a lot such a "corner cases" and it's not easy to detect "Anchored/moored" When is that? SOG below 0.5/0.05/0.005 ??.
I can actually be out sailing at any of these speed. I then may like to be noticed about a CPA alert so I don't think we shall hard code such a undefined occasion.

If you don't wont CPA alerts due this or crowded areas or whatever -- Just switch it of by either shortcut "W" or canvas menu "Hide CPA alarm". As simple as that?

ocornu, 2025-02-20:
> If you don't wont CPA alerts due this or crowded areas or whatever -- Just switch it of by either shortcut "W" or canvas menu "Hide CPA alarm". As simple as that?

Thanks, i'll try it and see if it fits my use case. I'm still interested by the general case though, so let's talk about it…

> There could be a lot such a "corner cases" and it's not easy to detect "Anchored/moored" When is that? SOG below 0.5/0.05/0.005 ??.

There already is such a speed threshold for AIS targets to be considered "anchored" (because we cannot rely on their AIS status for this purpose). How is that any different?

> I can actually be out sailing at any of these speed. I then may like to be noticed about a CPA alert so I don't think we shall hard code such a undefined occasion.

Dude, don't take it the wrong way, but at 0.005kn you're as much sailing as a dead fish is swimming…
Joke aside: what good is a CPA alert to you at that speed? You're not maneuverable anyway, are you?
Can you explain why you'd be "sailing" at that speed and needing CPA alerts? I'm genuinely interested…

Hakansv, 2025-02-20:
> what good is a CPA alert to you at that speed? You're not maneuverable anyway, are you

For example fishing and dreaming. Don't want to be run over by a tanker. Better start the motor?
And, such a threshold is difficult in small boats anchored. You can move as far as the chain allows.

ocornu, 2025-02-21:
> For example fishing and dreaming. Don't want to be run over by a tanker. Better start the motor?

I see… It's a bit of a corner case: in such a situation, it's the job of moving boats to dodge you, rather than the opposite (at least, assuming your AIS can emit). 
But i agree with you: this use case should be covered. Yet, it shouldn't take priority over the other 99% of usages. In other words: OpenOCN should allow both behaviors (and probably defaults on the most common).
Notice i've advocated for an option similar to the one existing for AIS targets (not something hard-coded): you can turn it on/off and tweak the speed threshold as you see fit.

> And, such a threshold is difficult in small boats anchored. You can move as far as the chain allows.

Let not perfect ruin a good enough solution, in my opinion. And the simple threshold implementation can always be improved on later…