OpenCPN/OpenCPN #4379: AIS: ownship sporadically not being correctly filtered out

ocornu, 2025-02-22:
**Describe the bug**
Sporadically, ownship is not filtered out correctly and is thus being displayed on the map as any other AIS target: ownship name is displayed (with AIS target font), _etc_. The bug does not seem to survive the next AIS update.
It does not seem to be an issue with my AIS device (em-trak B100) not flagging ownship as such, as I've never seen it happen on the B&G console fed by the same AIS source.

**To Reproduce**
Run OpenCPN with an AIS source, grab a cup of coffee and be (very) patient…?

**Expected behavior**
Ownship should never be confused with, and displayed as, an AIS target.

**Screenshots**
Too random and short-lived to be caught on a screenshot.

**Desktop (please complete the following information if applicable):**
Ubuntu Linux.

**Smartphone/Tablet (please complete the following information if applicable):**
Not tested.

**Additional context**
This issue _feels_ like a race condition between two concurrent threads. However, i don't even know if AIS data is subject to multi-threading…
From a quick browsing of the code, it seems ownship is correctly filtered out in ais.cpp:AISDrawTarget(), so i fail to see how this issue can arise:
```
 // And we never draw ownship
  if (td->b_OwnShip) return;
```
Is the AIS targets list memory being reused?

Hakansv, 2025-02-22:
Interesting. I've never seen this when receiving N0183 from a AIS transponder. The own ship data should be in AIVDO and thus b_OwnShip = true.
I've seen it when someone uses two AIS devices connected to the same PC, one transceiver and one receiver. The latter collects the VDM from the first and transfers it as N0183. Then one have to use a  MMSI rule to filter it out.

You say: "_AIS device (em-trak B100) not flagging ownship_" I'm not aware of any ownship flag in the VDM message. But I can have missed. Please can you elaborate?

ocornu, 2025-02-22:
Good point: i should have mentioned from the start that i'm pulling AIS data from the N2K bus (specially as ―i believe― this device also speaks N0183). And now, thanks to your feedback, i suppose it's a N2K bug only…

> You say: "AIS device (em-trak B100) not flagging ownship" I'm not aware of any ownship flag in the VDM message. But I can have missed. Please can you elaborate?

My bad: i was talking opencpn-ish to keep things simple (and, to be accurate, i should have written `b_OwnShip`). The corresponding N2K flag i meant to talk about seems to be called `N2kaisown_information_not_broadcast`.

Hakansv, 2025-02-22:
What I can find the em-trak B100 does only use serial N0183. (At least from the manual I found?)
So the N2k message might come from another device in your system?
That device would create PGN129038 or PGN129039 or PGN129794 and also set the flag "tN2kAISTranceiverInfo" to anything greater than zero (0), "4"?.
Now I don't have used a N2k AIS device and can not compare or debug the OCPN code here. But since, what I know, this is not reported by anyone else could it be an error in any of your devices?

ocornu, 2025-02-22:
> What I can find the em-trak B100 does only use serial N0183. (At least from the manual I found?)

This is incorrect. The B100 indeed has two N0183 ports, but [it also has a N2K one](https://www.marine.help/product-page/b100-standard-ais-class-b). We only use N2K on this boat.

[image]

> So the N2k message might come from another device in your system? 

I can confirm there are no other AIS devices on the vessel.

> That device would create PGN129038 or PGN129039 or PGN129794 and also set the flag "tN2kAISTranceiverInfo" to anything greater than zero (0), "4"?

Right. I've investigated the code a bit further, and beside two different types (`tN2kAISTranceiverInfo` and `tN2kAISTransceiverInformation`) for the same AIS field, i couldn't spot any issue. Unfortunately, this error-prone redundancy did not seem to have any impact on the rest of the code.
I got rid of the misspelled one to be sure (see patch below): i'll report if the issue does not appear any longer, but this might take a while…  :-|

> Now I don't have used a N2k AIS device and can not compare or debug the OCPN code here. But since, what I know, this is not reported by anyone else could it be an error in any of your devices?

That's what i thought at first. But such a sporadic issue on an embedded device like the B100 is rather unlikely. I've also never seen the bug happen on the B&G chart plotter which uses the same N2K AIS source. 
Consider as well that this is a sporadic and harmless bug: it's safe to assume that most people would miss it, and among the few that would, most wouldn't go as far as to report it.
This being said, i cannot totally discard that hypothesis yet.

I'm considering running a N2K filter full-time to try to catch any bad message upstream, but the way i'm set up (single NGT-1 N2K to USB converter) i wouldn't be able to use OpenCPN at the same time. :-/
Perhaps i could find a way to multiplex the N2K serial stream in software?

```
    Remove a N2K AIS type redundancy.

---------------------- libs/N2KParser/include/N2KParser.h ----------------------
index b6963bfce..aaac96d16 100644
@@ -382,7 +382,7 @@ bool ParseN2kPGN129794(std::vector<unsigned char> &v, uint8_t &MessageID, tN2kAI
                         uint32_t &IMOnumber, char *Callsign, char *Name, uint8_t &VesselType, double &Length,
                         double &Beam, double &PosRefStbd, double &PosRefBow, uint16_t &ETAdate, double &ETAtime,
                         double &Draught, char *Destination, tN2kAISVersion &AISversion, tN2kGNSStype &GNSStype,
-                        tN2kAISDTE &DTE, tN2kAISTranceiverInfo &AISinfo);
+                        tN2kAISDTE &DTE, tN2kAISTransceiverInformation &AISinfo);
 
 //*****************************************************************************
 // AIS static data class B part A

--------------------- libs/N2KParser/include/N2kMessages.h ---------------------
index 58a2e4cff..ad7194849 100644
@@ -1589,12 +1589,12 @@ void SetN2kPGN129794(tN2kMsg &N2kMsg, uint8_t MessageID, tN2kAISRepeat Repeat, u
                         uint32_t IMOnumber, char *Callsign, char *Name, uint8_t VesselType, double Length,
                         double Beam, double PosRefStbd, double PosRefBow, uint16_t ETAdate, double ETAtime,
                         double Draught, char *Destination, tN2kAISVersion AISversion, tN2kGNSStype GNSStype,
-                        tN2kAISDTE DTE, tN2kAISTranceiverInfo AISinfo);
+                        tN2kAISDTE DTE, tN2kAISTransceiverInformation AISinfo);
 inline void SetN2kAISClassAStatic(tN2kMsg &N2kMsg, uint8_t MessageID, tN2kAISRepeat Repeat, uint32_t UserID,
                         uint32_t IMOnumber, char *Callsign, char *Name, uint8_t VesselType, double Length,
                         double Beam, double PosRefStbd, double PosRefBow, uint16_t ETAdate, double ETAtime,
                         double Draught, char *Destination, tN2kAISVersion AISversion, tN2kGNSStype GNSStype,
-                        tN2kAISDTE DTE, tN2kAISTranceiverInfo AISinfo) {
+                        tN2kAISDTE DTE, tN2kAISTransceiverInformation AISinfo) {
   SetN2kPGN129794(N2kMsg, MessageID, Repeat, UserID, IMOnumber, Callsign, Name, VesselType, Length,
                   Beam, PosRefStbd, PosRefBow, ETAdate, ETAtime, Draught, Destination, AISversion, GNSStype, DTE, AISinfo);
 }
@@ -1603,13 +1603,13 @@ bool ParseN2kPGN129794(const tN2kMsg &N2kMsg, uint8_t &MessageID, tN2kAISRepeat
                         uint32_t &IMOnumber, char *Callsign, char *Name, uint8_t &VesselType, double &Length,
                         double &Beam, double &PosRefStbd, double &PosRefBow, uint16_t &ETAdate, double &ETAtime,
                         double &Draught, char *Destination, tN2kAISVersion &AISversion, tN2kGNSStype &GNSStype,
-                        tN2kAISDTE &DTE, tN2kAISTranceiverInfo &AISinfo);
+                        tN2kAISDTE &DTE, tN2kAISTransceiverInformation &AISinfo);
 
 inline bool ParseN2kAISClassAStatic(const tN2kMsg &N2kMsg, uint8_t &MessageID, tN2kAISRepeat &Repeat, uint32_t &UserID,
                         uint32_t & IMOnumber, char *Callsign, char *Name, uint8_t &VesselType, double &Length,
                         double &Beam, double &PosRefStbd, double &PosRefBow, uint16_t &ETAdate, double &ETAtime,
                         double &Draught, char *Destination, tN2kAISVersion &AISversion, tN2kGNSStype &GNSStype,
-                        tN2kAISDTE &DTE, tN2kAISTranceiverInfo &AISinfo) {
[... cut ...]
   if (ParseN2kPGN129794(v, MessageID, Repeat, UserID, IMOnumber, Callsign, Name,
                         VesselType, Length, Beam, PosRefStbd, PosRefBow,
@@ -721,7 +721,7 @@ bool AisDecoder::HandleN2K_129794(std::shared_ptr<const Nmea2000Msg> n2k_msg) {
 
     pTargetData->b_OwnShip =
         AISinfo ==
-        tN2kAISTranceiverInfo::N2kaisti_Own_information_not_broadcast;
+        tN2kAISTransceiverInformation::N2kaisown_information_not_broadcast;
 
     pTargetData->DimA = PosRefBow;
     pTargetData->DimB = Length - PosRefBow;
```

Hakansv, 2025-02-22:
So my manual was a dated one. Good.
Lack of or ignored report may well be a fact. Very few of all users visit our forum.

To debug this and your patches, can't you connect the NGT-1 N2K directly to an OCPN computer to simplify the test?
I wasn't at all involved to set up the use of "N2kTypes.h" so others might comment your proposed changes.

Hakansv, 2025-02-22:
BTW: Too make more verifying tests:
I friend may have access to a N2k AIS and OCPN to make a similar test. Hopefully tomorrow or a day after.

TwoCanPlugIn, 2025-02-22:
Suggest you grab a dump of the raw data.
Some AIS Transceivers set the transceiver information field to all 1's eg 11111. (The NMEA 2000 definition for any data set to all 1's.is data unavailable. 

The current logic in OpenCPN may incorrectly determine an own ship message.

ocornu, 2025-02-23:
Thanks for your help and suggestions. Much appreciated… :-)

I ended up running a logger on N2K AIS messages all night in order to learn more about this…

For the record i used the canboat project, which provides several useful tools in this regard, filtering out all PGNs but 129038, 129039 and 129794. Of those, i only logged messages containing my own ship's MMSI.
It amounted to 17152 messages overall, all PGN129039 (AIS Class B Position Report) ―obviously, as the em-trak B100 is a Class B device. But only 16963 of those had "AIS Transceiver Information" set to "Own information not broadcast" (_i.e._ value 4).

In other words, _~1% of those messages (189) would have made OpenCPN mistake my own ship for a random AIS target_.

Here's one such message for reference, in raw and json format (relevant fields underlined by me):
```
2025-02-23T00:44:35.204Z 4  43 255 129039 AIS Class B Position Report:  12 A2 9D C6 1F 6B AB F6 3B F5 92 82 02 8F FF FF 00 00 06 00 1E FF FF 00 74 01 FF
```
```json
{"timestamp": "2025-02-23T00:44:35.204Z",
"prio": 4, "src": 43, "dst": 255,
"pgn": 129039,
"description": "AIS Class B Position Report",
"fields":{
    "Message ID": "Standard Class B position report",
    "Repeat Indicator": "Initial",
    **"User ID": <my ship's MMSI>**,
    "Longitude": 100.6021483, "Latitude": 4.2111733, "Position Accuracy": "High",
    "RAIM": "in use", "Time Stamp": 35, "SOG": 0.00, "Communication State": "06 00 06",
    **"AIS Transceiver information": "Channel B VDL transmission"**,
    "Unit type": "CS", "Integrated Display": "No", "DSC": "Yes", "Band": "Entire marine band",
    "Can handle Msg 22": "Yes", "AIS mode": "Autonomous", "AIS communication state": "ITDMA",
    "Reserved": "00 FF"
}}
```
Digging a bit deeper, there is one such message every 3 minutes, and they're strictly alternating between "Channel A VDL transmission" and "Channel B VDL transmission":
```json
{"timestamp":"2025-02-23T00:17:35.698Z" ... "AIS Transceiver information":"Channel A VDL transmission" ... }
{"timestamp":"2025-02-23T00:20:36.781Z" ... "AIS Transceiver information":"Channel B VDL transmission" ... }
{"timestamp":"2025-02-23T00:23:35.684Z" ... "AIS Transceiver information":"Channel A VDL transmission" ... }
{"timestamp":"2025-02-23T00:26:34.820Z" ... "AIS Transceiver information":"Channel B VDL transmission" ... }
{"timestamp":"2025-02-23T00:29:34.853Z" ... "AIS Transceiver information":"Channel A VDL transmission" ... }
{"timestamp":"2025-02-23T00:32:35.122Z" ... "AIS Transceiver information":"Channel B VDL transmission" ... }
```
I don't know the AIS/N2K standards well enough to tell if this is normal behavior, legit interpretation of some under-defined spec, or a plain B100 bug.
There's no mention of firmware updates available on the em-trak website, and i'm gonna need to put my hand on a Windows/MacOS computer to run their proprietary configuration software and see if it offers one.

In the meantime, i checked a bit deeper the B&G chart plotter that's connected on the same N2K network and does _not_ display this "own ship as an AIS target" behavior. Turns out, it uses the user-defined MMSI to filter these suspicious messages out. If i set it to another MMSI, it does display my own ship as a random AIS target as well.

Whether it is an em-trak bug or not, i believe there's a valuable lesson to be learned here in order to make OpenCPN more resilient to this class of issues: go beyond merely checking `AISinfo == N2kaisown_information_not_broadcast` to filter out these messages, use the own ship MMSI as well.
Note that this value (MMSI) does not even need to be set by the user: it can be learned from correct "AIS Transceiver Information"=4 messages (at least with N2K, not sure about N0183/SignalK). This approach could also be used to fill in values like ship length and width (when the user has not defined any)…

TwoCanPlugIn, 2025-02-23:
As I said, you need the raw data, not Cannboat's interpretation. 

To fully understand the problem we need to see the raw bytes (in this case the five bits) that represent the transceiver information.

I had come across a similar problem reported by a user of the TwoCan plugin last year.

leamas, 2025-02-23:
The new data monitor under way will allow this. However, it's not ready.

ocornu, 2025-02-23:
> As I said, you need the raw data, not Cannboat's interpretation.
> 
> To fully understand the problem we need to see the raw bytes (in this case the five bits) that represent the transceiver information.

```
2025-02-23T00:44:35.204Z 4 43 255 129039 AIS Class B Position Report: 12 A2 9D C6 1F 6B AB F6 3B F5 92 82 02 8F FF FF 00 00 06 00 1E FF FF 00 74 01 FF
```

> I had come across a similar problem reported by a user of the TwoCan plugin last year.

Interesting… How did it end up?

ocornu, 2025-02-23:
> The new data monitor under way will allow this. However, it's not ready.

What do you mean exactly by "this"?
And where can i learn more about this new data monitor?

TwoCanPlugIn, 2025-02-23:
If you look at byte 20, it has the value of 0xF8.

Transceiver Information is encoded over 5 bits, eg (data(20) & 0xF8) >> 3.

Probably best way to treat this is assume the NMEA 2000 standard way of handling the maximum value as "data unavailable" and either drop the entire message or treat it as a AIS VDO message. 

What is confusing especially as compared to NMEA 0183 is that notification of an own ship message (VDO) should also indicate what channel (A or B) it was transmitted on. Keep in mind that both NMEA and marine equipment vendors (and for that matter Timo's libraries) are not infallible.

ocornu, 2025-02-23:
> If you look at byte 20, it has the value of 0x1E
> 
> Transceiver Information is encoded over 5 bits, eg (data(20) & 0xF8) >> 3.

Thanks! So, basically: `0x1e>>3 = 3 => channel_B_VDL_transmission` (check!) 
(The bit-mask is redundant as we're shifting all 3 unused bits into oblivion anyway)
 
> Need to verify how Timo's libraries parse this.

I'm not sure who's Timo, but both OpenCPN and canboat parse it the same way you did up there.

> What is confusing especially as compared to NMEA 0183 is that notification of an own ship message (VDO) should also indicate what channel (A or B) it was transmitted on.

Exactly: that's why i assumed in the beginning that `own_information_not_broadcast` was a flag. Assumption reinforced by the fact there are 5 bits available and… 5 possible values in the associated type in the code, which (as bit positions) would have allowed to indicate both at once:
```
enum tN2kAISTransceiverInformation {
                           N2kaischannel_A_VDL_reception=0,
                           N2kaischannel_B_VDL_reception=1,
                           N2kaischannel_A_VDL_transmission=2,
                           N2kaischannel_B_VDL_transmission=3,
                           N2kaisown_information_not_broadcast=4
                          };
```

> Keep in mind that both NMEA and marine equipment vendors (and for that matter Timo's libraries) are not infallible.

No one is, for sure. :-)

Hakansv, 2025-02-23:
In OCPN you can try a similar thing as the MFD. Use MMSI properties. Add a new item and exclude your own MMSI.
Note: This I haven't tested myself for a N2k AIS targets list. I think we would all benefit a verification of this feature for N2k.

[image]

btw:
The used header, ..\libs\N2KParser\include\N2kTypes.h, have all five items:
```
enum tN2kAISTranceiverInfo {
                            N2kaisti_Channel_A_VDL_reception=0,
                            N2kaisti_Channel_B_VDL_reception=1,
                            N2kaisti_Channel_A_VDL_transmission=2,
                            N2kaisti_Channel_B_VDL_transmission=3,
                            N2kaisti_Own_information_not_broadcast=4,
                            N2kaisti_Reserved=5
                          };
```

ocornu, 2025-02-23:
> In OCPN you can try a similar thing as the MFD. Use MMSI properties. Add a new item and exclude your own MMSI. Note: This I haven't tested myself for a N2k AIS targets list. I think we would all benefit a verification of this feature for N2k.
> 
> [image]

Doesn't work: i had my own MMSI flagged as "ignore" all along while i was observing this sporadic bug (i did not know about `own_information_not_broadcast` and assumed this "ignore" was necessary to exclude my own ship from AIS targets).
I'm not sure what this "ignore" does exactly, but it does not remove that MMSI from AIS targets displayed on the chart.

> btw: The used header, ..\libs\N2KParser\include\N2kTypes.h, have all five items:
>
> ```
> enum tN2kAISTranceiverInfo { [...]

Yes, that's the misspelled duplicate version of `tN2kAISTransceiverInformation` that i get rid of with the above patch.

Hakansv, 2025-02-23:
> Doesn't work: i had my own MMSI flagged as "ignore" all along while i was observing this sporadic bug (i did not know about own_information_not_broadcast and assumed this "ignore" was necessary to exclude my own ship from AIS targets).
> I'm not sure what this "ignore" does exactly, but it does not remove that AIS target from the chart.

Thought so. This would be a missed function in AIS decoder for N2k targets. 
For all other source types, N0183 and SignalK, it's working. So we need to work on this part. Simple enough for me to make a try.

Normally we don't need to use MMSI-properties to exclude own MMSI since it would be solved automatically. 
For N0183 the AIVDM should not be transferred by the device but instead AIVDO and that is excluded by OCPN.
For SignalK the server should have a "self" sign so also ignored as an external target by OCPN.
If your device is the culprit not sending the correct info or OCPN code, as you suggest, I can't determine. More experienced programmers will have to look at that. 
But "MMSI-properties" should work anyway.

Hakansv, 2025-02-23:
I think I see what's happening:
A N2k AIS target listed in the MMSI-properties list to be ignored is cleaned out from the AIS targets list but until then every new PGN update for that target is parsed and probably drawn for a while. I suspect that's what "sporadically" refers to.

I've filed a PR #4382 but have only been able to test the general function on SignalK data. Would really appreciate if you @ocornu or anyone else can verify the function by the help of a N2k AIS transponder.

TwoCanPlugIn, 2025-02-23:
I'm away without my dev machine otherwise would generate the PGN's for you. Nonetheless you should be able to feed the above data to an OpenCPN N2K serial connection. Create a pair of virtual ports using socat and echo the data to one of the ports.

Note that it seems only a few N2K AIS transceivers exhibit the behaviour causing this problem so asking others to test may not exercise your fix. For TwoCan the bug was reported for a Garmin AIS transceiver, I haven't had reports from users with other manufacturers' units.

The workaround registering the own-ships MMSI should be OK, but I personally think it hides a fundamental bug.

i don't have access to the standard, nor does Kees (author of canboat) but I believe Timo, author of the N2K libraries that OpenCPN uses, does have access to the standard in his official  work capacity.

I don't know if it is an error by NMEA, a misinterpretation by AIS vendors or an incorrect enum or decoding of the 5 bits in the library, but a VDL Channel A or Channel B Transmission implies an own-ship transmission. Should the 5 bits be used as an equality match as they are today, or should they be used to test individual bits?

ocornu, 2025-02-24:
> I'm away without my dev machine otherwise would generate the PGN's for you. Nonetheless you should be able to feed the above data to an OpenCPN N2K serial connection. Create a pair of virtual ports using socat and echo the data to one of the ports.

Ah, good point. Thanks for the tips, much appreciated!

> Note that it seems only a few N2K AIS transceivers exhibit the behaviour causing this problem so asking others to test may not exercise your fix. For TwoCan the bug was reported for a Garmin AIS transceiver, I haven't had reports from users with other manufacturers' units.

Good to know. We now know there's (at least) another one: the em-trak B100. :-]

> The workaround registering the own-ships MMSI should be OK, but I personally think it hides a fundamental bug.
>
> i don't have access to the standard, nor does Kees (author of canboat) but I believe [...]

It is rather confusing indeed…
However, no straightforward flag/bit position model seems to fit either. Here are the 3 values i'm observing for PGN129039 with `UserID == my_MMSI`:
```
- 0x16:    00010110    =>    00010 = 2    (channel_A_VDL_transmission)
- 0x1e:    00011110    =>    00011 = 3    (channel_B_VDL_transmission)
- 0x26:    00100110    =>    00100 = 4    (own_information_not_broadcast)
```
By the way, what are the 3 right-most bits used for?

In the meantime (while we wait for someone with proper access to double-check this), are you advocating processing `channel_A_VDL_transmission` and `channel_B_VDL_transmission` as implicit `own_information_not_broadcast`?
Are you aware of any situation where this approach could come back to bite us? Like, are their any kind of AIS relays that are susceptible to repeat/broadcast our own ship data, and that we'd still want to distinguish those from the ones we are broadcasting ourselves?

> Just a comment on the above,
> 
> > Thanks! So, basically: 0x1e>>3 = 3 => channel_B_VDL_transmission (check!)
> > (The bit-mask is redundant as we're shifting all 3 unused bits into oblivion anyway
> 
> There are many instances where multiple values may be encoded within a byte or range of bytes, so the bit mask while in this case may be redundant it makes it explicitly clear to the reader how many bits are used for the value. For example look at PGN 129284 and values for arrival circle entered, perpendicular crossed, bearing reference, calculation type etc. TwoCan does not rely on third party libraries, for me it is my "documentation" of the NMEA 2000 standard.

Yes, i understand that.
However, being redundant in this context (of a simple and concise explanation), it made my brain go into a bunch of side-quests about potential reasons that took me minutes to discard one by one. A bit as if you had written 0x1E*1 (instead of 0x1E): i'd be like "why would he add this *1?", "perhaps he's using *2 as a left-shift and he made a typo, and really meant 0x1e<<1?", "or perhaps this is taken from generic code that could be left-shifting but isn't, so it's been passed 1 instead if anything bigger?", _etc_. :-D
It truly was confusing to me and did make my life unnecessarily harder. It's probably just my autistic brain tho: you can just ignore such comments. I realize now you must have assumed i was just being a smart-ass. Happens a lot…  :-D
Please accept my apologies.

ocornu, 2025-02-24:
> I think I see what's happening: A N2k AIS target listed in the MMSI-properties list to be ignored is cleaned out from the AIS targets list but until then every new PGN update for that target is parsed and probably drawn for a while. I suspect that's what "sporadically" refers to.

Ah, good catch! That would explain this "race condition feel" it had…

> I've filed a PR [#4382](https://github.com/OpenCPN/OpenCPN/pull/4382) but have only been able to test the general function on SignalK data. Would really appreciate if you [@ocornu](https://github.com/ocornu) or anyone else can verify the function by the help of a N2k AIS transponder.

Sure, i'll run it as soon as i'm sailing and report. It might take a while though, considering how sporadic this is: always difficult to prove a negative… :-p 

One question: as i understand your patch (i quickly browsed through it and am not familiar with that part of the code, so please correct me if i'm wrong), you've opted for removing the target from the AIS target list altogether.
How about keeping an AIS targets database as exhaustive as possible, and leaving it to each user-code to discard what they don't need (like `own_ship` for drawing purposes, _e.g._)? This way, other parts of the code base could still do something useful with this information…

ocornu, 2025-02-25:
I can confirm that your commit does prevent own ship from being displayed as a target on the map, if its own MMSI is in the ignore list. :-)

> No, that is "old" code that I have not changed. Probably consistent with MMSI prop: "Ignore this target". The philosophy might be that you don't need own(ignored) ship to be listed. It's expected that you know that you are there? :)

Sure: i don't want my own ship to be displayed on the screen as a tiers AIS target, nor do i want collision alerts between myself and i. What can i say? I'm a weirdo… :-]
By the way, there's never any CPA alert with myself, even though my (buggy) AIS target is obviously pretty close to me: another bug in hiding?

Back to our discussion: the trouble with your ignore-based approach is that it assumes what is being done now with the AIS targets database (target display, collision detection…) is all that will ever be done with it; Thus it does not matter to remove targets entirely and rely on this mechanism downstream.
Consider this:
- What if i wanted to export my AIS target list  to a website/app so that nearby sailors could see how crowded the place is? Then i'd want my own ship (and most likely other ignored targets) to be featured in that list.
- What if i imported a list of AIS targets in the area (from, say, MarineTraffic) to see beyond what my limited VHS antenna can receive? Then my (ignored) own ship would be in that list, and your code would fail.
- What if i wanted to fill-in certain values using AIS data (like ship name, MMSI, size, position of the GPS/transceiver, etc…) that the user hasn't set himself? Then i'd need own ship to be in my AIS targets database…

These are just a couple ideas out of the top of my head, but other people i'm sure will come up with better ideas to exploit this data. Therefore the correct approach, at least as far as own ship is concerned, is to keep doing what we're already doing: maintain an exhaustive targets list, set `b_OwnShip` and let user code rule it out where it makes sense to do so. 

One simple way to do this would be to follow @TwoCanPlugIn's suggestion: assume that `channel_A_VDL_transmission` and `channel_B_VDL_transmission` imply `own_information_not_broadcast`.
For what it's worth, i've sniffed AIS traffic for a while to check this: every single instance of VDL transmission indeed had my UserID/MMSI (_i.e._ it was an `own_information_not_broadcast`).
Then this short patch is all that's needed (for own ship):
```diff

-------------------------- model/src/ais_decoder.cpp --------------------------
index b56deccc3..9a5b9bce8 100644
@@ -126,6 +126,12 @@ static inline double GeodesicRadToDeg(double rads) {
 
 static inline double MS2KNOTS(double ms) { return ms * 1.9438444924406; }
 
+static inline bool isN2kAisOwnShip(tN2kAisTransceiverInformation info) {
+  return info==tN2kAisTransceiverInformation::N2kaisown_information_not_broadcast ||
+         info==tN2kAisTransceiverInformation::N2kaischannel_A_VDL_transmission ||
+         info==tN2kAisTransceiverInformation::N2kaischannel_B_VDL_transmission;
+}
+
 int AisMeteoNewMmsi(int, int, int, int, int);
 int origin_mmsi = 0;
 
@@ -477,9 +483,7 @@ bool AisDecoder::HandleN2K_129038(std::shared_ptr<const Nmea2000Msg> n2k_msg) {
     //
     //     pTargetData->ROTIND = round(rot_dir * pow((ROT / 4.733), 2));
 
-    pTargetData->b_OwnShip =
-        AISTransceiverInformation ==
-        tN2kAisTransceiverInformation::N2kaisown_information_not_broadcast;
+    pTargetData->b_OwnShip = isN2kAisOwnShip(AISTransceiverInformation);
     pTargetData->b_active = true;
     pTargetData->b_lost = false;
     pTargetData->b_positionOnceValid = true;
@@ -575,9 +579,7 @@ bool AisDecoder::HandleN2K_129039(std::shared_ptr<const Nmea2000Msg> n2k_msg) {
     pTargetData->b_lost = false;
     pTargetData->LastPositionReportTicks = pTargetData->PositionReportTicks;
     pTargetData->PositionReportTicks = now.GetTicks();
-    pTargetData->b_OwnShip =
-        AISTransceiverInformation ==
-        tN2kAisTransceiverInformation::N2kaisown_information_not_broadcast;
+    pTargetData->b_OwnShip = isN2kAisOwnShip(AISTransceiverInformation);
 
     pSelectAIS->DeleteSelectablePoint((void *)(long)mmsi, SELTYPE_AISTARGET);
     CommitAISTarget(pTargetData, "", true, bnewtarget);
@@ -743,10 +745,7 @@ bool AisDecoder::HandleN2K_129794(std::shared_ptr<const Nmea2000Msg> n2k_msg) {
     Name[sizeof(Name) - 1] = 0;
     pTargetData->b_nameValid = true;
     pTargetData->MID = 124;  // Indicates a name from n2k
-
-    pTargetData->b_OwnShip =
-        AISinfo ==
-        tN2kAisTransceiverInformation::N2kaisown_information_not_broadcast;
+    pTargetData->b_OwnShip = isN2kAisOwnShip(AISinfo);
 
     pTargetData->DimA = PosRefBow;
     pTargetData->DimB = Length - PosRefBow;
```
Note that this patch assumes the patch further above (consistent transceiver info type) has been first applied .

Hakansv, 2025-02-25:
1. Many thanks for the confirmation. That will close this issue's main cause?

2. The "What if" discussion may be moved to another (discussion)Issue? But the "own ship" handling is the same for all types of sources be it N0183, N2k or SignalK. (If you import from MT, Marine Traffic, it would be N0183? and thus will MMSI-properties still work.)

3. I can agree that the VDL info status whether it says transmission(2 || 3) or own_..._not...(4)  would mean OwnShip. What else would be transferred from a AIS transponder? Also likely implied [here](https://ttlappalainen.github.io/NMEA2000/group__group__msg_types.html#gae7f8142668a3c9396ada7d2e725eb15a)
But all this is another issue. May I suggest you create a PR about suggested code changes to be revived TwoCan and Dave, bdbcat.

TwoCanPlugIn, 2025-02-25:
> What else would be transferred from a AIS transponder

The problem I encountered with TwoCan and the Garmin AIS, IIRC was TxInfo was set to all 1's (eg. 0x1F). For both 0x1F and 0x1E the message could probably be dropped as in NMEA 2000 terminology it means "data unavailable" or "error".

Agree with Hakan, your three "What If" scenarios should be moved to a discussion. As far as core OpenCPN is concerned there is no concept of importing or exporting AIS Data. OpenCPN receives AIS data from any of it's supported connections (NMEA 0183 or 2000 and SignalK). Even traffic received from Marine Traffic or other aggregators is in NMEA 0183 format, so OpenCPN will ignore any VDM message in its "ignore" list. And of course you can output received NMEA 0183 AIS sentences to an outside connection. For an outbound connection you could filter VDO sentences or leave it to the aggregator to handle.

Hakansv, 2025-02-25:
I'll later issue a discussion about own ship data, e.g. MMSI, as many MFDs apparently do. Now we can achieve it using the MMSI properties interface but .... Let's see.

ocornu, 2025-02-26:
I'm afraid you guys are missing my point entirely.

My point is: the AIS targets list is a representation of the world (from an AIS perspective). If you keep it as is, you can exchange it rather seamlessly with other actors (import it, export it, _etc_).
The ignore lists are just masks put over it, tailored to some specific usage (like bit-masks in our raw PGN example earlier). The "friend-ship" list will mask member ships from collision detection algorithms, the "ignore" list will do the same and also mask member ships from the drawing algorithm, and so on. But none of those masks alter the underlying representation itself.

However, if you start hacking through the targets list, for example by removing some ships altogether, you break this metaphor for no good reason and, ultimately, you're shooting yourself in the foot.
Your architecture becomes difficult to understand because it has no obvious equivalent anymore: it's just a pile of ad-hoc "solutions" to a bunch of arbitrary corner cases, that no-one can guess unless they know the whole code and its historical context. "But it works!" is no excuse.
It does not matter either that this specific website (MarineTraffic) uses that specific protocol (N0183) so "it would still work so far". On the contrary: taking this into account at this stage amounts to tightly coupling your architecture with some arbitrary external website. Tomorrow you'll have to deal with json (or another kind of input), and you'll need to make even more ad-hoc exceptions here and there to try to keep your "design" standing. And the code will get even more obscure, messier, harder to understand and maintain.

It's not an AIS or an OpenCPN specific point. It's a general design point about the importance of picking the right data structures and architectures for a specific task, rather than rushing for a quick fix.
This series of "what ifs" was supposed to illustrate that point. I'm not sure what kind of discussion you guys want to start from them: as far as i'm concerned, their sole intent was to show that the chosen course to fix the "ignore list does not honor its contract" bug was not, according to me, the right one. Even if it "just works".

> 1. Many thanks for the confirmation. That will close this issue's main cause?

Originally this bug was not about the fact that the AIS ignore list contract was no honored. Admittedly, after the AIS layer let some own_ship messages slip through, i did try to use the ignore list ―and it also failed.
We should probably have opened another bug, that your commit did target, and that we could now close; But we didn't. Now that second bugged is fixed, but the first one still isn't.
As it is, e-trak B100 users would have to be an exception, being the only ones (we know of) having to ignore their own ship to get standard OpenCPN AIS behavior.
According to me, the original bug has not been addressed and this ticket shouldn't be closed. But you're the boss: do as you wish.

ocornu, 2025-02-27:
> Does OpenCPN have some questionable internal structures and constructs? Of course it does.

Quite the understatement! Get your code-base externally reviewed (a free automated tool will do) and weep.

> But first and foremost, we are committed to supporting our practical existing base, with a very small "developer/LOC" metric.

I'm not sure i get what you mean by "small developer/LOC metric"… Doesn't it mean that, the more bloated the code is, the better your metric gets?
Consider all the time you spent trying to fix #4376 before you threw in the towel and slashed the zoom-to-mouse feature instead… That's the price you gotta pay in the end for this privileging-the-short-term philosophy: this _should_ have been a quick ans easy fix.
If you yourself can't locate your kittens in that code, who's supposed to? Maybe that's the reason why the project does not have more developers? The state of the code base probably being a pretty efficient walled garden?

> Anything else architecturally related should be a discussion, or RFE, or specific PR to be tested against all known user platforms.

Sure. Rush half-assed commits straight into master, and send back everything else to die in a comity… </sarcasm>

Look, i've tried to contribute productively for days now, and all i got to show for the time i spent is a loss of a critical feature (look-ahead zoom-to-mouse) according to me, and what i consider a degradation of the code-base overall ―even though it does successfully camouflage the issue behind an ad-hoc work-around.
To be honest, and at the risk of sounding cocky: it would have been quicker for me to (actually) fix those bugs myself…
It's obviously unrealistically challenging for us to work on the same project, so it's better that i just work on my things on the side and leave you guys alone ―i can hear those sighs of relief! :-D
No hard feelings. Cheers! :-)

closed 2025-02-27 by bdbcat

rgleason, 2025-02-27:
@ocornu  Yikes. What happened to thoughtful collaboration? Sure its a community or family, but for the most part it gets along.