OpenCPN/OpenCPN #4380: AIS: render targets in real size when size on map is larger than AIS target icon

ocornu, 2025-02-22:
**Is your feature request related to a problem? Please describe.**
There's an AIS option to show targets in "real size" (as reported by AIS) on the map, and it seems to be triggered by some arbitrary map scale threshold. Which means, big vessels can be larger than their representation on the map, despite this option being turned ON. 
Out of safety considerations, OpenCPN should try its hardest to represent AIS targets in ways that are not misleading. In other words, OpenCPN should never represent/draw AIS targets smaller than it knows them to be.

**Describe the solution you'd like**
When the "show targets real size" option is turned ON, no target on the map should appear smaller than it actually is ―or rather: than its AIS declares it to be.
Which means switching from icon to drawing the ship (at scale) when the latter is wider than the former, regardless of current map scale. 

**Describe alternatives you've considered**
I haven't considered any.

**Additional context**
I've sailed the Malacca straight lately and encountered this problem on multiple occasions. Obviously, no one should maneuver on OpenCPN and AIS data alone, yet that is no excuse not to do the right thing when we can.

Hakansv, 2025-02-22:
The size of the AIS icon is the same all time. Following S52 rules if not user adjusted.
The actual size of the ship is indicated if known and that size is larger then a programmed limit. What I can see the size is drawn correct. See shot.

Do you suggest that the ship size should be shown on a smaller chart scale? I think the size is drawn only if it's bigger than actual icon size, wide or length, on screen? 
(See the pgm comment "// drawing too small does not make sense")

[image]

ocornu, 2025-02-23:
> The size of the AIS icon is the same all time. Following S52 rules if not user adjusted. The actual size of the ship is indicated if known and that size is larger then a programmed limit. What I can see the size is drawn correct. See shot.

Yes, i understand this. I haven't noticed any discrepancy in the ship size drawing (although i wasn't specifically checking this).

> Do you suggest that the ship size should be shown on a smaller chart scale? I think the size is drawn only if it's bigger than actual icon size, wide or length, on screen? (See the pgm comment "// drawing too small does not make sense")

If real-size ship drawing was triggered by this constraint, it'd be perfect (that's what i'm advocating for in this enhancement request); However, my experience is that it is not (thus this post).
To be fair, i don't believe i've checked the code ―or if i did, i forgot about it… :-D

bdbcat, 2025-02-23:
Code, ais.cpp:948
```
      if (ais_real_size[4].x - ais_real_size[0].x < 16 ||
          ais_real_size[2].y - ais_real_size[0].y < 30)
        bcan_draw_size = false;  // drawing too small does not make sense

```
This code inhibits real size rendering if the rendered dimensions are less than the default target icon size.  And it works fine by my testing.

Happy to review any screenshots showing failure of this logic.

ocornu, 2025-02-24:
I'm working from memory: i was sailing around Singapore a couple weeks ago, and there was quite a lot of heavy traffic and anchored tankers to slalom between over there, as well as significant current (15° of lateral drift, making it difficult to see clearly, by eye only, if you're gonna clear that moored tanker or not). More specifically, as i remember it, zoomed-out it looked like the COG-based prediction was dodging the vessel, yet zoomed-in it showed i was running right through it.

Perhaps it had to do with the nature of the vessel? Those are very long ships with the GPS/AIS transceiver located at the very back of the ship, very far from the bow and not at all in the middle. Like in @Hakansv's picture:
[image]
I should note however that there was no issue either at such "extreme" level of zoom. The issue was around the zoom-level where the real-size should/would trigger. 

It's a rather vivid memory, but perhaps i got it wrong then? Or perhaps i remember it wrong? Unfortunately, there are no such vessel where i am right now: impossible for me to test it again IRL at the moment.

I haven't dug into the code either…

> ```
>       if (ais_real_size[4].x - ais_real_size[0].x < 16 ||
>           ais_real_size[2].y - ais_real_size[0].y < 30)
>         bcan_draw_size = false;  // drawing too small does not make sense
> ``` [...]

Thanks, i'll look into it further.
Is there a possibility, you think, that this code does not behave as intended, or does not cover some unforeseen corner case?

bdbcat, 2025-02-24:
Possible, I suppose, that the ship dimensions as received are bogus.
But far more likely is the situation you noted.  These large ships have their GPS receivers near the stern, usually.  So passing closely by the bow of such a ship could be misleading.  Coupled with heavy current, and, well...
I understand the current thing. I regularly sail in the Gulf Stream.  Determining a comfortable CPA at night while crossing the stream was pretty scary before AIS and MARPA.

ocornu, 2025-02-24:
> Possible, I suppose, that the ship dimensions as received are bogus.

No, the ship dimensions looked ok and appeared to be drawn correctly. And once zoomed enough to trigger real-size, it was obvious we were on a collision course.

> But far more likely is the situation you noted. These large ships have their GPS receivers near the stern, usually. So passing closely by the bow of such a ship could be misleading.

Yes, that was the situation. GPS near the stern (correctly reflected by real-size drawing, afaict) and we were passing closely by the bow…

> I understand the current thing. I regularly sail in the Gulf Stream. Determining a comfortable CPA at night while crossing the stream was pretty scary before AIS and MARPA.

I bet it was!  :-p

Hakansv, 2025-02-24:
As I usually mention to course participants here. Avoid passing in front of large ships. Set CPA alarm to at least 1.5 Nm and 20 minutes.

ocornu, 2025-02-24:
> As I usually mention to course participants here. Avoid passing in front of large ships. Set CPA alarm to at least 1.5 Nm and 20 minutes.

Sounds like good general advice!
The boat i mentioned was anchored though: passing in front is just as safe as passing behind.  
I'm afraid those settings wouldn't work around Singapore: you'd have constant CPA alerts ―which amounts to no CPA alert at all! I had to tighten settings significantly in order to muzzle as many false positives as safety could cope with… :-p

[image]

The above image does not make it justice though: it does not show how tight and busy the international rail is, both sides being forbidden territorial waters if you're just passing by… 

[image]

Anyway, i'm digressing…

ocornu, 2025-02-24:
> Time to close this? If needed you can reopen it.

Sure! For lack of evidence (if only temporarily)…

closed 2025-02-24 by ocornu

ocornu, 2025-02-26:
So, i've been thinking about this again, trying to clarify my memories, and imagining what close phenomenon i might have confused with the one i described in the original bug description above…

I see two possibilities so far:
1. being a simple contour ("transparent" fill) the real-size ship might no have stood up enough to be immediately obvious, thus giving the impression the collision course was not properly displayed when i finally realized it was there.
Why not replacing the AIS target icon by its real-size counter part altogether, instead of drawing a low-visibility contour around its usual icon? In other words, filling in this contour with the target appropriate color (and adding all the other indicator overlays that would be present on a usual target icon)? The idea behind it being: to really honor the real-size metaphor and improve visibility.
2. a CPA alert might have been misleading, claiming for example we would come to within 100m of each other, except the tanker was longer than 100m with the AIS transceiver located at the very back (the bow was 150m further _e.g._).
Does the CPA algorithm take ships dimensions into account, or is the minimal distance announced measured transceiver-to-transceiver?

Hakansv, 2025-02-26:
My thoughts: 
The above discussion again emphasizes the recommendation not to use shorter CPA range than 1 Nm. Anything less than that can also make the helmsman on ships nervous unnecessarily.