OpenCPN/OpenCPN #4386: OpenCPN 5.10.2 PPA and backport do not include pypilot plugin

luu5, 2025-02-28:
OpenCPN 5.10.2 PPA and backport do not include pypilot plugin. Flatpak does. Also Windows OpenCPN has pypilot.

**To Reproduce**
Install OpenPlotter on Raspberry using PPA
Go to plugin manager
There is no pypilot plugin

**Expected behavior**
There would be pypilot available

nohal, 2025-03-01:
Not an OpenCPN issue. If a third party plugin does not support any platform, it is an issue of that plugin, not the OpenCPN application developed in this project.

In this particular case the problem is the plugin developer did not submit the Bookworm ARM64 package, which he actually built and made available at https://cloudsmith.io/~opencpn/repos/pypilot-prod/packages/detail/raw/pypilot_pi-0.53.2.0-debian-arm64-12-bookworm-arm64-tarball/v0.53.2.0/#metadata to the plugin catalog.

closed 2025-03-01 by nohal

nohal, 2025-03-01:
Please try to stay in contact with the reality, the package was not submitted to the catalog, which you can trivially confirm by actually reading https://github.com/OpenCPN/plugins/blob/master/ocpn-plugins.xml.

rgleason, 2025-03-01:
Pavel, my reality is I do too much on Opencpn already. I am not preparing PPA.  Sorry

nohal, 2025-03-01:
This has absolutely nothing to do with any PPA.

sailoog, 2025-03-02:
Therefore, it seems that the package is already built but was not sent to the package manager. Who can do this? @rgleason ? @seandepagnier ?

leamas, 2025-03-02:
Or more  exactly, the  missing build(s) are not part of the plugin catalog.  Technically, this is a problem with pypilot and should be reported at https://github.com/rgleason/pypilot_pi where @rgleason  perhaps could handle it.

bdbcat, 2025-03-02:
Rick is away from his desk for a month or so.
I'll manually add the plugin to the catalog.

bdbcat, 2025-03-02:
Done.
There is apparently some problem with the plugin build script for this platform.  The bare metadata .XML file was not uploaded to Cloudsmith, so the automated script adding plugin to catalog did not find it.  I extracted the required metadata file from tarball, renamed, and pushed to catalog.
tl;dr;  We may see this again...

bdbcat, 2025-03-03:
Do you know if any previous version worked on this platform?

mgrouch, 2025-03-03:
Oh. Actually both are working. I had to change the pypilot host to localhost in the second case. It's fine now.