OpenCPN/OpenCPN #4387: Optimized Land Crossing Detection via Distance Field Approach

sebastien-rosset, 2025-03-01:
# Problem Description

The current implementation of OSMSHP land crossing detection significantly underperforms compared to the GSHHS version, even when using smaller data files. 

# Propose Solution: Optimized Land Crossing Detection via Distance Field Approach 

Implement a distance field technique to achieve substantial performance improvements while maintaining or enhancing accuracy. This approach pre-computes the minimum distance to land from any point in a multi-resolution grid, enabling near-constant time queries regardless of coastline complexity.
The distance field will be calculated on-demand when entering a new osmshp tile that hasn't been visited before, and will persist the distance field for that tile. This will ensure that subsequent requests are in O(1). 

The system would initially build the distance field for tiles where `CrossesLand` requests are made, such as where weather routing operations are performed. The results are cached to disk for future use. This approach distributes the computation cost across sessions while ensuring that frequently sailed areas benefit from consistently high performance. The distance field would use adaptive resolution - high precision near coastlines where detailed navigation decisions matter, and progressively lower resolution farther from shore - significantly reducing memory requirements while maintaining accuracy where it matters. 

A key optimization involves appropriate compression techniques, including quantization of distance values (storing exact distances only up to common safety margins), quadtree-based spatial indexing, and run-length encoding for vast ocean areas. 

I think this approach could deliver microsecond-level query performance. This represents a potential 10-100x speedup for weather routing applications which perform thousands of crossing checks during route calculation. 

The distance field cache would have to be invalidated when new tiles are downloaded. This would happen when the user is updating the osmshp files. 

The nice thing about this approach is that this can be retrofitted in the existing plugin API while providing a blazingly fast response. This is in contrast to the existing implementation which visits every polygon in a tile, and some tiles have tens of thousands of polygons. 

Additionally, this implementation could support a new API function that returns the actual distance to land, not just a binary crossing determination. This would enable more sophisticated routing constraints, proper handling of various safety margins. This would also provide a further 4X performance improvement by removing the need for the weather routing plugin to make four calls to the crossesland function. 

For the weather routing plugin specifically, this approach aligns perfectly with its propagation pattern, where geographically adjacent points are evaluated in sequence, allowing for exceptional cache coherence and memory locality that further enhances performance. 

This approach could be extended to take other types of obstructions in consideration. For example, this could be used to establish a safety margins around buoys, which can be tricky to see, especially during the night. And yes, there could be over-the-top advanced capabilities beyond basic land avoidance. These are not necessarily useful in practice but they are just fun to think about.

**Navigation Hazards Distance Field**
Similar to the land distance field, a comprehensive hazard distance field could incorporate underwater obstructions, wrecks, rocks, shallow areas, waves. This would create a single unified "danger distance" metric for route planning, accounting for vessel draft.

**Dynamic Depth Contour Analysis**
By extracting bathymetric data, the system could dynamically generate safe passages based on the vessel's draft plus safety margin. This would automatically route vessels through deeper channels and away from shoaling areas, especially valuable in complex archipelagos or river deltas.

bcn2, 2025-03-02:
OpenCPN is missing a "Route Checker" today. Perhaps it's most important blind spot.
Do you think your approach can facilitate an implementation?

sebastien-rosset, 2025-03-02:
> OpenCPN is missing a "Route Checker" today. Perhaps it's most important blind spot.
> Do you think your approach can facilitate an implementation?

Do you mean the ability to automatically validate a route against navigation hazards and constraints? Can you elaborate? Here is what maybe you are thinking about:

Checks:
   - Land crossings
   - Shallow water areas based on vessel draft
   - Navigation hazards (rocks, wrecks, obstructions)
   - Restricted areas or exclusion zones
   - Traffic separation schemes
   - Bridge clearances for vessels with height restrictions
   - buoys checks, e g port side, starboard side ..

There are potentially many types of checks. This RFE could help for land proximity, safety distance, depth, but not for checks like bridge clearance.

bcn2, 2025-03-02:
_"the ability to automatically validate a route against navigation hazards and constraints"_
Exactly - your listing is quite complete. In the case of buoys, case of cardinal marks as well.
And yes, this should include the "safety envelope" of the vessel, from draft and lateral clearing.

Ideally a routing generator like weather routing would check those objects as well. Or as a second steps once a route selected.
What does it help when the most covenient route crosses an exclusion zone or is running non-orthogonal through a TSS?

sebastien-rosset, 2025-03-02:
> _"the ability to automatically validate a route against navigation hazards and constraints"_
> Exactly - your listing is quite complete. In the case of buoys, case of cardinal marks as well.

Makes sense.

> And yes, this should include the "safety envelope" of the vessel, from draft and lateral clearing.
>
> Ideally a routing generator like weather routing would check those objects as well. Or as a second [...]

The distance field could be used in any situation where there is a safety distance to maintain. It could potentially be used both in route planning and while navigating. However it wouldn't help for:
1. Tides 
2. Bridge clearance 
3. TSS with mandatory direction of travel
4. One-way channels or fairways
5. Recommended tracks with specific headings
6. Speed restriction zones
7. Temporary notice to mariners
8. Proper entry/exit points for canals, locks, harbors, channels.
9. And other types of constraints that are not based on distance.

Eventually, multiple constraint-based operations (including distance field) could be combined to provide a comprehensive route checker. There could be day time versus night time variants, bad weather, etc

bc38, 2025-03-06:
More about Route Check and Anti Grounding

https://youtu.be/AQxTkI4JOCw?si=uXlXbMLouD5rfeLv

https://youtu.be/hV0wOFdldUE?si=gqJvDs_GsIDIzT13

bcn2, 2025-03-06:
Searching for ECDIS Route Checker:
https://www.linkedin.com/pulse/ecdis-automatic-route-check-function-explained-part-1-caroletti-
or
https://www.youtube.com/watch?v=7p-wW0mqgYA
for some impressions/ideas

quinton-hoole, 2025-03-22:
I'm no expert on these matters, but the proposal seems very sound to me. 
Two relatively minor things I've run into recently, that may be worth considering (although I'm almost certain you have already):

1. I've come across areas on some charts (in this case in the Tuomotus and Marquesas islands in French Polynesia) labelled as "No Hydrographic Information Available" or similar. It's probabably safest to route around these areas where possible (or issue very loud warnings otherwise).  Hopefully the depth data in the charts will be represented in a way that makes this happen by default.

2. It's not that uncommon (e.g. in the Tuamotus) to see currents (e.g. in passes into atolls) where the current speed (sometimes far) exceeds boat speed. I don't think those particular currents are possible to predict with any accuracy, but in other cases the data may be available in tide and current forecasts. Either way, these sorts of current can dramatically affect the safety or otherwise of a route.

rgleason, 2025-04-12:
https://github.com/rgleason/weather_routing_pi/issues/279