OpenCPN/OpenCPN #4389: Add own MMSI management to avoid interference with AIS (self) target

Hakansv, 2025-03-02:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

leamas, 2025-03-02:
Overall, the idea to add MMSI handling looks good to me, FWIW. However, the GUI makes me a bit uneasy

There is no natural relation between units & format on one side and  the MMSI number on the other. The heading "Units, formats and MMSI" thus lumps unrelated stuff together.

For the initial configuration a new, separate page labeled "Own Ship" might be better. Besides the MMSI number it could perhaps also house the overall width & length settings.

Following the same logic it might make sense to move the MMSI setting next to the length/width settings in the configuration page. All of these are own ship  properties and belongs together. The overall organization of this page could probably need an overhaul, but that's another issue.

bdbcat, 2025-03-02:
"Following the same logic it might make sense to move the MMSI setting next to the length/width settings in the configuration page. All of these are own ship properties and belongs together."

I agree.  In the "Settings->ownship" properties page, Just slide the MMSI entry down a bit.

Hakansv, 2025-03-02:
Thank you for the "acceptance" of the feature and the idea to house it in Options->Ships.
Of course, the MMSI control should be among other own ship options, no problem.

A PR to come..

About the Wizard:  I was hesitant at first to add it there, not least because of the comment about created with wxformbuilder - "Do not edit" and the source file was not found. But still I dared to edit and await a comment from the author?? :)
I selected to place it among units and changed the header.  The wizard runs infrequently and only for a new user, so all options are new and missing context may not be the first reaction. Also, from what I've seen, very few users change the ship icon so the ship's dimensions are rarely used.

bdbcat, 2025-03-02:
On further thought, I now would remove from first start wizard.  
I can hear:
"Honey, do you remember what our MMSI number is?  I can't install OpenCPN without it..."

Adds a speed-bump to new user startup which is not necessary for most installations.  Available later if required.

Hakansv, 2025-03-02:
Options now like the shot?

Wizard option is out. Agree it may confuse.

[image]

Hakansv, 2025-03-02:
Or for the most users:
[image]

Hakansv, 2025-03-02:
The mmsi is omitted from first time Wizard. May confuse

PR #4390 (2025-03-02): Add own ship MMSI to options Own Ship. Don't use that as an AIS target
Closes #4389
closed 2025-03-02 by bdbcat