OpenCPN/OpenCPN #4399: Remove the 0183 "ALL" message channel /type

leamas, 2025-03-14:
**Is your feature request related to a problem? Please describe.**
Currently, n0183 drivers posts each received message twice; one copy using the message type as key and another using the ALL key.

The ALL key was introduced in order to make it possible to listen to all 0183 messages and transfer them to plugins using the "old" way where consumers listens to all data instead of listening to explicit keys. It's kind of a hack meaning that messages are handled twice. Also, the ALL message type occasionally is visible in logging etc. instead of the correct n0183 type causing confusion.

**Describe the solution you'd like**
Using the new API  `GetActiveMessages()`  and  ` EventVar new_msg_event`  in  *comm_navmsg_bus{.h,.cpp}* it is possible to enumerate all current messages received.  If sinks, notably the multiplexer, used this API the ALL message channel could be abandoned.  This would simplify  the input code paths considerably when each message is handled once instead of twice.

TwoCanPlugIn, 2025-03-15:
Will it have any impact on the SetNMEASentence Plugin API?

leamas, 2025-03-15:
No

bdbcat, 2025-04-23:
N2K drivers also use the "ALL" target as well.
Time to fix this, I think.

leamas, 2025-04-23:
Question just is if it should be done to 5.12 or postponed. I have basically dropped ideas of major refactorings like this in the 5.12 cycle, we are close to beta freeze.

bdbcat, 2025-04-23:
How big a job is it?

leamas, 2025-04-23:
Not really sure.  The problem is not so much the main effort as plugging all sorts  of inevitable holes which will be left after the first round.  And personally, I have a yacht to launch...

bdbcat, 2025-04-24:
OK, lets table this for O 5.14, or so.
Not so pretty, but surely functional for now.

leamas, 2025-04-24:
Done.