OpenCPN/OpenCPN #4402: comm_drv_n2k_serial.cpp fails to send frames on the N2k bus

ocornu, 2025-03-16:
**Describe the bug**
`comm_drv_n2k_serial.cpp` fails to send frames on the N2k bus, using an Actisense NGT-1 (0403:d9aa).
After investigation, it turns out `CommDriverN2KSerial::SetTXPGN()` fails to set up the NGT-1 for sending PGNs.

The legacy code (which allowed TX on a per-PGN basis) has been `#ifdef 0`'ed in favor of a wildcard approach (clearing the TX list with `NGT_STARTUP_SEQ`), allegedly allowing all TX PGNs:
```cpp
/* Copied from canboat Project
 * https://github.com/canboat/canboat
 *
 * The following startup command reverse engineered from Actisense NMEAreader.
 * It instructs the NGT1 to clear its PGN message TX list, thus it starts
 * sending all PGNs.
 */
static unsigned char NGT_STARTUP_SEQ[] = {
    0x11, /* msg byte 1, meaning ? */
    0x02, /* msg byte 2, meaning ? */
    0x00  /* msg byte 3, meaning ? */
};

int CommDriverN2KSerial::SetTXPGN(int pgn) {
  // ...
  SendMgmtMsg(NGT_STARTUP_SEQ, sizeof(NGT_STARTUP_SEQ), 0x11, 0, NULL);

#if 0
  //  Enable PGN message
  unsigned char request_enable[] = { 0x47,
                      0x00, 0x00, 0x00,           //pgn
                      0x00, 0x01,
                      0xFF, 0xFF, 0xFF, 0xFF};
```
In its current state, this wildcard approach does not work on my hardware (double-checked with logs from another device on the same N2k network). Further investigation is needed to figure out why.
However, reverting back to the per-PGN legacy code _does fix_ PGN sending…

**To Reproduce**
Unfortunately, there does not seem to be any N2k PGN sending code in the current code-base to easily reproduce this bug.

**Desktop (please complete the following information if applicable):**
 - Linux 6.8.0-45

**Additional context**
As a side note, this class should be renamed after the device it's targeting (Actisense NGT-1) as, contrary to what its name suggest (`CommDriverN2KSerial`), it is _not_ a generic N2k serial driver.
While doing so, and at the risk of sounding pedantic, please write N2k (not N2K): the SI abbreviation for 10³ is a lower-case k (kilo); the upper-case K is the abbreviation for 2¹⁰, therefore N2K literally means N2048. :-D

bdbcat, 2025-03-16:
" Unfortunately, there does not seem to be any N2k PGN sending code in the current code-base to easily reproduce this bug."
Correct.  So, for our study, how did you discover and confirm this fault?

TwoCanPlugIn, 2025-03-17:
> "Unfortunately, there does not seem to be any N2k PGN sending code in the current code-base to easily reproduce this bug."

I don't own an Actisense NGT-1 so I can't replicate or test. However when using the NGT-1, does OpenCPN not transmit PGN 126996 (Product Information) as it does for the SocketCAN interface?

For example, when using SocketCAN, upon reception of a PGN 59904 (ISO Request) for PGN 126996, OpenCPN does respond correctly by transmitting the product information.

bdbcat, 2025-03-17:
PGN 126996 is automatically transmitted by NGT-1 (and Wifi cousins) on power up, independent of OCPN.  It does this as part of the address-claim dance.

SocketCAN interfaces are not so smart, so we have to perform the dance in OCPN driver side.

ocornu, 2025-03-18:
> " Unfortunately, there does not seem to be any N2k PGN sending code in the current code-base to easily reproduce this bug." Correct. So, for our study, how did you discover and confirm this fault?

I'm coding a dashboard instrument to drive my Simrad NAC-2 autopilot. My MFD (B&G Zeus 3S) records every N2k frame and can export the last 3 hours (iirc) of pcap logs.
Used virtual serial ports to confirm my code was outputting correct bytes, but the NGT-1 wasn't forwarding them (that's also how i stumbled on #4403).

TwoCanPlugIn, 2025-03-19:
> . It does this as part of the address-claim dance.

PGN 126996 has nothing to do with the address claim dance which is performed solely by PGN's 60928 & 65240.

PGN 126996 may be transmitted upon startup or in response to a request from another device to report product, hardware, software, model versions, serial number etc.

As I don't own a NGT-1 I cannot verify if a NGT-1 generates it's own PGN 126996 or whether it is the responsibility of the application.

> I'm coding a dashboard instrument to drive my Simrad NAC-2 autopilot. 

FYI, NAC-3 autopilot controller PGN's have been reverse engineered and documented on Canboat Presumably similar to your NAC-2.
https://github.com/canboat/canboat/issues/309

ocornu, 2025-03-20:
To answer your first question: no, before reverting to the legacy code, i didn't see any PGN 126996, nor any other PGN from source 0 (my Actisense NGT-1) in the logs for that matter. However, the pcap logs come split in files covering one minute each, and i was only checking the files around the time when i had sent PGNs (i never had the need to go back as far as boot-up time).
Considering my MFD N2k network explorer does identify the Actisense NGT-1 correctly, i assume a PGN 126996 must have gone through at some point in time. As it makes no mention of OpenCPN, i also assume the Actisense is managing those autonomously by default…

> FYI, NAC-3 autopilot controller PGN's have been reverse engineered and documented on Canboat Presumably similar to your NAC-2.

Thanks for the link! i did use Canboat code and documentation as a starting point, but i wasn't aware of this particular ticket.
Canboat documentation is incomplete and sometimes erroneous (note that it does not pretend otherwise), but more importantly it lacks information about how to use these PGNs (when to send what to get what you want, _e.g._). In the end i was able to fill the gaps to properly monitor and drive my NAC-2 from OpenCPN. I've even added a "COG mode" because the native "No drift mode" is zig-zaging too much (as if gain was too high ―except it's not). All good. :-)

ocornu, 2025-03-20:
Note that there might be collision between this bug and #4403: i can't remember in which order i've fixed those, maybe the #4403 fix would make the wildcard PGNs TX code work too…
As far as i'm concerned, i kind of like the idea of rigorously maintaining an explicit list of output PGNs anyway.

bdbcat, 2025-03-21:
ocornu....
Digging into this report.  I have an NGT-1 setup running on OpenCPN, with external bus monitor hardware.
Can you provide some pseudo code showing how you are attempting to send A/P messages through the NGT-1?
Specifically:
1.  Are you using the plugin API calls?
2.  What PGNs are sent?

For reference, using the OpenCPN Javascript plugin, and the existing code, I can successfully send some PGNs onto the bus.  Here is minimal js code, to be  entered into the js plugin script editor to send one PGN 128267 (depth report).  This works for me.  The js plugin exercises the published OpenCPN plugin API.

```
Nmea2000 = require("NMEA2000");
nmeaSend = new Nmea2000(128267);
nmeaSend.push();

```
Thanks

bdbcat, 2025-03-21:
BTW...
The NGT-1 **does** autonomously send PGN 126996 on startup, and when requested.  And it identifies itself on the N2K bus as an "Actisense gateway".  This of course means that when using this device, OpenCPN will not be specifically identifiable on the N2K bus as a message "source".  O is hiding behind the gateway.

ocornu, 2025-03-22:
> ocornu.... Digging into this report. I have an NGT-1 setup running on OpenCPN, with external bus monitor hardware. Can you provide some pseudo code showing how you are attempting to send A/P messages through the NGT-1? Specifically:
> 
>     1. Are you using the plugin API calls?

Yes. 

>     2. What PGNs are sent?

Currently: 65305 and 130850. 

> For reference, using the OpenCPN Javascript plugin, and the existing code, I can successfully send some PGNs onto the bus. Here is minimal js code, to be entered into the js plugin script editor to [...]

I'm using C++ and rely on the comm API declared in `ocpn_plugin.h`:
- First i fetch the proper `DriverHandle` for N2k, using  `GetActiveDrivers()` and `GetAttributes()`.
- Then i set up the NGT-1 TX list, using `RegisterTXPGNs()`.
- Finally, anytime i later need to send a PGN, i craft it up (OpenCPN's N2k parser does not support these PGNs) and call `WriteCommDriverN2K()`.

I insist on the fact this code is working fine with legacy `RegisterTXPGNs()`.
I've checked along `WriteCommDriverN2K()` call-path that things were fine at every step (i went so far as double-checking the checksum calculation): they were, all the way to the ultimate `m_serial.write()`. 
Lastly, I've checked using a virtual serial port (in place of the the NGT-1's) that the code was outputting the desired payload. Once #4403 was taken care of: it did (before that fix, payload was amputated of its last bytes).

bdbcat, 2025-03-23:
After some experimentation, I find this.

Actisense NGT-1:
1.  Resetting all TX filters allows the NGT-1 to transmit all PGNs _that it knows about_.  The device's internal microcode does not know anything about the proprietary Simrad PGNs you are using, so the SetTXPGN() call is definitely required in your use case.

YachtDevices YDNU-02:
1. This device is also supported by the OCPN N2KSerial device driver.  The device can be configured to use Actisense format for communication with the PC, and that is done during driver initialization.
2. The device can use Rx/Tx filters.  The filters may be configured by a specific set of commands, different from Actisense protocol.
3. On reset, the device filters are all clear.  When operating in N2K mode, the device will receive and transmit all PGNs without objection.
4. Accordingly, there is no need to use SetTXPGN() on this device.  Indeed, it does not process the Actisense command sequences.  Any errors reported from attempting to use Actisense commands are ignored.

tl;dr;
The driver code has been corrected to support proprietary PGNS (e.g. Simrad) on both supported devices.

Just for reference, here is a code snippet used for testing:

```
        int pgn = 65305;
        std::vector<DriverHandle> drivers = GetActiveDrivers();
        std::vector<int> pgns;
        pgns.push_back(pgn);
        RegisterTXPGNs(drivers[0], pgns);

        auto buffer = std::make_shared<std::vector<unsigned char>>();
        std::vector<unsigned char>* vec = buffer.get();
        for (int i=0; i < 8; i++) vec->push_back(0);

        WriteCommDriverN2K(drivers[0], pgn, 255, 7, buffer);

```
I connected this code to a hotkey (F12), for debug testing.

rgleason, 2025-03-23:
Dave, is there a reason to have these notes somewhere in the documentation?

bdbcat, 2025-03-24:
I suppose so, though I do not know where...

TwoCanPlugIn, 2025-03-24:
Presumably on NMEA 2000 connections other than am Actisense NGT-1 a call to SetTXPGN is a null op ?

From what I recall, the GetActiveDrivers and GetAttributes API's don't enable the plugin to differentiate between an Actisense NGT-1 and any other serial/usb device such as Yacht Devices YDNU-2

Should the advice for a plugin developer be to always call SetTXPGN when transmitting proprietary PGN's irrespecitive of the connection ?

bdbcat, 2025-03-24:
"Presumably on NMEA 2000 connections other than am Actisense NGT-1 a call to SetTXPGN is a null op ?"
Yes.  If you look at the code, the SetTXPGN() for Actisense requires three steps.  If the first step fails, we declare "probably YD", and report success.

"Should the advice for a plugin developer be to always call SetTXPGN when transmitting proprietary PGN's irrespecitive of the connection ?"
Yes.  Required for Actisense, NOP for YD.  Seems the safe bet, with the result being that the plugin needs to know nothing about the actual hardware interface in use.

FWIW, Tony's N2K support library in Javascript plugin always calls SetTXPGN() prior to TX, even for well known non-proprietary PGNs.
No harm to call SetTXPGN() once per session for all anticipated PGNs during plugin init method, just to be sure.

"I thought Sebastien already added appropriate comments to ocpn_plugin.h"
Sebastien did a great job documenting the API at functional level.  The notes above might better be called "Device Interface Notes".

ocornu, 2025-03-24:
> Resetting all TX filters allows the NGT-1 to transmit all PGNs _that it knows about_. The device's internal microcode does not know anything about the proprietary Simrad PGNs you are using, so the SetTXPGN() call is definitely required in your use case.

That is sooo weird. Have you seen that list of PGNs the NGT-1 firmware "knows about"?

> "Should the advice for a plugin developer be to always call SetTXPGN when transmitting proprietary PGN's irrespecitive of the connection ?"
Yes. Required for Actisense, NOP for YD. Seems the safe bet, with the result being that the plugin needs to know nothing about the actual hardware interface in use.
(...)
No harm to call SetTXPGN() once per session for all anticipated PGNs during plugin init method, just to be sure.

Except that is only guaranteed to work if there's _at most one_ N2k sender per OpenCPN instance.
If two plugins (or more) intend to send PGNs, for example, they will invariably step on each other's toes…
1. either at the TX-list level, if they're sending _different_ PGNs: their `RegisterTXPGNs()` calls overwriting each other,
2. or at a higher level, if they're sending _the same_ PGNs: two sources unknowingly sending two concurrent versions of the same PGN is almost guaranteed _not_ to have the expected results.

(2) is not a TX-list issue _per se_ (so a bit outside of this bug's original scope), yet it deserves to be taken into account as well. More generally it showcases why promiscuous mode, while acceptable on the input side, is not a good idea on the output side…
Most likely the N2k driver should monitor which sender is sending which PGNs, manage the TX list accordingly, and alert in case several senders intend to TX the same PGN.

bdbcat, 2025-04-24:
Pending further bug reports, closing this issue.

closed 2025-04-24 by bdbcat