OpenCPN/OpenCPN #4403: CommDriverN2KSerialThread fails to flush the serial output buffer after writing a message.

ocornu, 2025-03-16:
**Describe the bug**
`CommDriverN2KSerialThread::WriteComPortPhysical()` methods fail to flush the serial output buffer after writing a message, therefore it is legit for the serial layer to withhold bytes until further writes (which can potentially be: never!).

**Expected behavior**
Messages sent on the serial bus are expected to be sent ASAP. Therefore the output buffer must be flushed after each message.

**Fix**
```diff
@@ -716,39 +722,41 @@ void CommDriverN2KSerialThread::ThreadMessage(const wxString& msg) {
 size_t CommDriverN2KSerialThread::WriteComPortPhysical(
     std::vector<unsigned char> msg) {
   if (m_serial.isOpen()) {
     ssize_t status = 0;
 #if 0
     printf("X                ");
     for (size_t i = 0; i < msg.size(); i++) printf("%02X ", msg[i]);
     printf("\n");
 #endif
     try {
       status = m_serial.write((uint8_t*)msg.data(), msg.size());
+      m_serial.flushOutput();
     } catch (std::exception& e) {
       WARNING_LOG << "Unhandled Exception while writing to serial port: "
                   << e.what();
       return -1;
     }
     return status;
   } else {
     return -1;
   }
 }
 
 size_t CommDriverN2KSerialThread::WriteComPortPhysical(unsigned char* msg,
                                                        size_t length) {
   if (m_serial.isOpen()) {
     ssize_t status;
     try {
       status = m_serial.write((uint8_t*)msg, length);
+      m_serial.flushOutput();
     } catch (std::exception&) {
       //       std::cerr << "Unhandled Exception while writing to serial port: "
       //       << e.what() << std::endl;
       return -1;
     }
     return status;
   } else {
     return -1;
   }
 }
 
```

closed 2025-03-24 by bdbcat

ocornu, 2025-03-24:
How does f6ca869 take care of both `CommDriverN2KSerialThread::WriteComPortPhysical()` methods?

bdbcat, 2025-03-25:
"How does https://github.com/OpenCPN/OpenCPN/commit/f6ca8696570279bfd2007192cf357dce947aca72 take care of both CommDriverN2KSerialThread::WriteComPortPhysical() methods?"
Because they both end up at the same place in the std_serial_io.cpp module.

ocornu, 2025-03-25:
> Because they both end up at the same place in the `std_serial_io.cpp` module.

I wasn't aware of that…
Can you please explain how `CommDriverN2KSerialThread::WriteComPortPhysical()` calling `m_serial.write()` ends up in `StdSerialIo::WriteComPortPhysical()`?

bdbcat, 2025-03-25:
Hmmm....
Looks like Windows, in its delightful difference-ness, follows a different model for com port output.
We will need to add a flush() call in to Windows lower level write write method.

If you have a repeatable, testable, failure case that seems like it needs a flush, please try this:

libs/serial/src/impl/win.cc : 451

+ flush();
  return (size_t) (bytes_written);

Thanks

ocornu, 2025-03-25:
> Hmmm.... Looks like Windows, in its delightful difference-ness, follows a different model for com port output. We will need to add a flush() call in to Windows lower level write write method.
>
> If [...]

I'm sorry: there are no Windows machine onboard… 

> Can you please explain how `CommDriverN2KSerialThread::WriteComPortPhysical()` calling `m_serial.write()` ends up in `StdSerialIo::WriteComPortPhysical()`?

Would you be so kind and clear that up for me, please?

bdbcat, 2025-03-25:
OK, you are correct.
The code path for binary writes from N2K serial drivers uses a different low-level output module, avoiding the "0" problem.
I have added the correct flush() invocations in github  https://github.com/OpenCPN/OpenCPN/commit/0144534cb9e238d98d3e25eec6fd0dc416df794c

Thanks for helping us track this down.

bdbcat, 2025-03-26:
After some more testing and analysis on the various platforms, I have concluded that your original suggested correction is the bast approach to ensuring binary data packets are fully flushed on serial connection,available.  And it has been tested in the wild.  Good.  :)
Committed to github.

ocornu, 2025-03-27:
While we're on this piece of code: improved version. Fixes a couple bugs (returning -1 when return type `size_t` is _unsigned_) and improves general handling (only flushes if there's something to be flushed, exception handling consistent with `StdSerialIo::WriteComPortPhysical()`, _etc_).

```c++
size_t CommDriverN2KSerialThread::WriteComPortPhysical(
    std::vector<unsigned char> msg) {
  return WriteComPortPhysical(msg.data(), msg.size());
}

size_t CommDriverN2KSerialThread::WriteComPortPhysical(unsigned char* msg,
                                                       size_t length) {
  if (!m_serial.isOpen()) return 0;
  try {
    size_t status = m_serial.write((uint8_t*)msg, length);
    if (status) m_serial.flushOutput();
    return status;
  } catch (std::exception& e) {
    DEBUG_LOG << "Unhandled Exception while writing to serial port: "
                << e.what();
  }
  return 0;
}
```

commit f6ca8696 (2025-03-24): Flush serial ports on each write operation.  Closes #4403
commit 920f882c (2025-03-26): Revert "Flush serial ports on each write operation.  Closes #4403"

This reverts commit f6ca8696570279bfd2007192cf357dce947aca72.