OpenCPN/OpenCPN #4404: Vessel AIS targets appear as buoys

ocornu, 2025-03-16:
**Describe the bug**
Actual vessels appear as buoy AIS targets.
I haven't looked into it, but i assume this is because of their MMSI...

**Expected behavior**
Do not classify AIS targets going 30+ knots as "buoys". 

**Screenshots**
A pretty fast "buoy": 
[image]

All these "buoys" are in reality ships:
[image]

More "buoys":
[image]

**Additional context**
This is pretty common in south-east Asia for some reason... Buoys AIS emitters are cheaper, perhaps?

Hakansv, 2025-03-16:
It has been classified as a "Buoy or similar" because of their invalid and perhaps fabricated MMSI numbers, that's correct.
So not a bug, It's intentional. 
In my opinion, speed should not be a parameter for any classification of objectives. A net buoy could very well be on the deck of a fishing boat traveling at 30 knots?
The discussion that led to the introduction of the BUOY class can be found [here](https://github.com/OpenCPN/OpenCPN/discussions/3998).

ocornu, 2025-03-18:
> It has been classified as a "Buoy or similar" because of their invalid and perhaps fabricated MMSI numbers, that's correct. So not a bug, It's intentional. In my opinion, speed should not be a parameter for any classification of objectives.

That wasn't my point: OpenCPN detects this buoy as "sailing with engines on". Surely buoys don't do that, nor do they broadcast that sort of things, right?
For what it's worth, my MFD does not confuse these ships with buoys. And i assume it has certifications in the AIS realm (_i.e._ it's not ignoring the "MMSI buoy class"). Which means there is (at least) one way to classify these AIS targets correctly (and yes: cutoff won't be SOG-based)…

Hakansv, 2025-03-18:
009119050 is not a valid MMSI number. It is made up, most likely by the user who has chosen not to certify this radio equipment and get a proper registration.
Instead of just ignore such class B MMSI we call it a buoy or similar not really identified target.

The "Underway using engine" is though interesting. That should not occur for a class B transmitter.
In the ais_decoder we set NavStatus = UNDEFINED for all those to be classified as Class BUOY and all such targets I've seen have the "Undefined" status. Both when received as N0183 or from SignalK.
Could it be you have received this from a N2k AIS /transponder/receiver? If so we don't override the NavStatus and force it to undefined but rather use the received message in PGN 129039. In this case I suppose some author or converter in you data chain sets this nav-status. 
Maybe we should override this so if the MMSI is not valid and we call it a buoy or similar we should also force the NavStatus to be Undefined? 
Thoughts?

rgleason, 2025-03-19:
It is not simple. @ocornu what would you suggest? 
Perhaps you should check your setup too?

ocornu, 2025-03-20:
> Thanks for the catch.

You're very welcome! Although i did not catch anything: you did all the work, and this bug request is about something else: vessels appearing as buoys.

Another example, see from OpenCPN:
[image]

... vs seen from a B&G MFD:
[image]

The "buoy"/vessel in question is a 20m long, 30+ tons commercial passenger ship; The one featured in the OG post was another commercial passenger ship going 30+ knots. These are not harmless buoys: they both are mobile and could cut my ship in two. They deserve to be displayed as actual ships, just like the B&G MFD does…
To be honest, i couldn't care less that their MMSI numbers are forged, or that they haven't paid their subscription this year (i'm not a cop). However, displaying them as buoys is both misleading and dangerous, and i do care about that.
As far as i'm concerned, those ships have gone through the trouble of getting equipped with an AIS transponder, and i need to be able to trust OpenCPN to represent them accordingly, instead of getting in the way (for what it's worth, my own copy will do the right thing, whatever ends up being decided here).

rgleason, 2025-03-20:
Why don't you cooperate and submit a PR since you"ve already figured it out?

Hakansv, 2025-03-20:
> OpenCPN to represent them accordingly

If someone installs equipment in an unsanctioned way, you can't count on proper handling, of course. In these cases, OCPN tries to show that there is at least something in the water. 
If you have a detailed suggestion for other handling, please formulate it. But I feel that we are repeating ourselves and nothing new is being added.

ocornu, 2025-03-21:
> Why don't you cooperate and submit a PR since you"ve already figured it out?

Because, as you can see, this is not a technical issue: it's a matter of "policy". I'm not wasting time cooking up a PR that will most certainly be rejected.
FWIW, i'm not claiming i've figured it all out. I was pointing out OpenCPN is not doing a good job in my use-case, and was assuming i was missing the big picture. It has failed to emerge so far… :-]

> If someone installs equipment in an unsanctioned way, you can't count on proper handling, of course.

And yet their transponders don't mind being set that way. And yet my MFD handles them just fine…
Is there a chance you're trying to be more catholic than the pope here?

In my experience, about half the class B ships in Thailand and Malaysia broadcast such a number. In other words, the ships that are, by far, the most numerous on these seas. *Half*.
Also, they're all commercial ships (fishermen mainly, but also passenger transportation, tourist tour ships, _etc_). I've never seen a private ship appear as an "OpenCPN buoy"…

As far i can remember, all the ones i've checked (a few dozens) started with "009" ―which does not seem to be a legit MMSI indeed. Yet, the sheer number of them alone begs the question: is there anything to it?
If it was only random, or a cheap ploy to save MMSI registration fees, what are the chances that they would all pick the same three first digits?
As far i can tell, 009xxxxxx (with the exception of [009990000: VHF coast station](https://en.wikipedia.org/wiki/Maritime_Mobile_Service_Identity#Group_coast_station_call_identities)) is an unused MMSI numbers group. Could it be that it's being used by some local authority for some local purpose?

On the other hand, i've never come across an actual buoy with such a MMSI (they all had conventional numbers). Has anybody?
I don't reckon that it's any more legit for a buoy to broadcast a fake MMSI than it is for a ship, is it? Therefore, what's the rationale behind classifying those fake MMSIs as "buoys" to begin with?

To be honest, i'd challenge the idea that this information ("unlawful MMSI") belongs on a chart (it should at best be an asterisk in the AIS details window). But, if it has to appear, why not a dedicated ship icon for unconventional MMSI instead of buoy?

> If you have a detailed suggestion for other handling, please formulate it.

I have. As this ticket's title explicitly conveys: just show those ships as ships!
Get rid of that extra code creating a virtual buoys class that does not seem to serve any/much purpose. What's wrong with that?
If this does not work, recognize that 009xxxxxx (excluding 009990000) group number as a known (although unconventional) group of class-B ships.
Basically, anything but the current _status quo_ that isn't working in those waters…

nohal, 2025-03-21:
Stop being dramatic, please. All these ships are using the cheap Chinese AIS transponders and don't give a damn about the regulations, that is all there is about it. There is probably no other region in the world this is possible as all of them would be fast punished by the authorities pretty much everywhere else in the world.
There is nothing that can be deduced from these MMSI numbers, all that's left is some heuristics distinguishing between a "ship" and an "object" based on something else than the MMSI. If you have such an algorithm, submit a PR and we are done.

nohal, 2025-03-21:
And to be more explicit, everywhere else but where you are, these MMSI numbers are abused by "objects", not "ships".

rgleason, 2025-03-22:
Ok, Should users be permitted to make exceptions? This is actually also dangerous, but a user could enter an exception of '009××××..' to be accepted as a ship. Would you then be happy with the operation of AIS?  

What if this "bug feature" were only enabled by adding in the config file under AIS  a line  "screwyAISuseat yourownrisk=1"?

Would you be happy?

Hakansv, 2025-03-22:
Aside from the lack of undefined NavStatus for Class B received by an N2k receiver this issue is not a bug. 
OCPN follows standards. With the lack of a general recommendation on how different countries should classify the increasingly common ATONs for net buoys, we have taken the liberty of clarifying, via our own Class BUOY, how these ATON's should be presented. 
The fact that some not so scrupulous boat owners install equipment in an irregular way cannot reasonably be taken into account. They are at least shown as "something is there in the water". Take care!

commit 5c300337 (2025-03-28): Correct N2k AIS Class B Navstatus. Closes #4404
commit 7282f611 (2025-03-28): Merge pull request #4407 from Hakansv/navstatus

Correct N2k AIS Class B Navstatus. Closes #4404
commit 9d959e10 (2025-03-31): Correct N2k AIS Class B Navstatus. Closes #4404
closed 2025-03-28 by bdbcat