OpenCPN/OpenCPN #4405: When the user pans a chart, the wind barbs do not move with the chart

sebastien-rosset, 2025-03-16:
**Describe the bug**
When the user pans a chart, the wind barbs do not move with the chart

**To Reproduce**
Steps to reproduce the behavior:
1. Open the GRIB plugin.
2. Load a GRIB, for example a HRRR GRIB.
3. Ensure the wind barbs and wind overlay are displayed.
4. Do Left click to pan the chart.

Notice the overlay moves with the chart (as expected) but not the wind barbs.

**Expected behavior**

The wind barbs should "move" with the chart. Instead, the magnitude and angle of the wind barbs are based on the data before panning the chart.

**Screenshots**

See attached .mp4

**Desktop (please complete the following information if applicable):**
 - MacOS 15.3.1
 - OpenCPN master build 3/16

**Additional context**
Discovered while testing #4370

sebastien-rosset, 2025-03-16:
https://github.com/user-attachments/assets/d1cc1831-c132-43b0-b1d1-839c02816fe4

bdbcat, 2025-03-16:
How completed?

sebastien-rosset, 2025-03-16:
This part of the grib plugin keeps tricking the mind. When you pan a chart, the origin points of the barbs are fixed on the screen, which makes the user believe the barbs are not panning like the chart. I'm addition, I had a grib with fairly consistent magnitudes and angles in a fairly large area, which compounded the effect.
The barb calculations are correct, so it's not a bug per se. Perhaps an RFE would be more appropriate to have the barbs follow the panning movement.

rgleason, 2025-03-16:
So the barbs stay in position and turn?

sebastien-rosset, 2025-03-16:
> So the barbs stay in position and turn?

Yes. The barbs are set on a grid at fixed coordinates. In other words, the barbs don't mov while panning the chart. So it tends to make you believe they are stuck at the original position where panning started.

Hakansv, 2025-03-16:
You're right. I just haven't noticed or thought about it before. 
My words for my own understanding: A barb has a position on the screen. It is controlled by the tightness specification in Grib settings. So each bar stays at the same screen position but its value, angle and wind speed, is constantly recalculated and updated for the new geographic position as the underlying chart moves. Amazing! No need to change. But it's fun to study.

sebastien-rosset, 2025-03-17:
> A barb has a position on the screen. It is controlled by the tightness specification in Grib settings. So each bar stays at the same screen position but its value, angle and wind speed, is constantly recalculated and updated for the new geographic position as the underlying chart moves.

Personally I find this behavior counter intuitive. When you pan, it looks as if the barbs are stuck, and the effect is more pronounced when the visible area has similar TWD and TWS values. Other weather forecast software tend to emulate a paper chart, where the wind barbs move with the chart panning function, which gives a more natural look.

bdbcat, 2025-03-17:
"Personally I find this behavior counter intuitive."
I agree.  probably originally implemented this way for expediency, and the requirement to interpolate between GRIB points in order to calculate values at specific lat/lon.
Seem reasonable to reopen some part of this Issue as an RFE, or "Enhancement".

sebastien-rosset, 2025-03-17:
In areas of low wind, or when the TWD changes within a small area, it also tends to create weird behaviors, where some arrows tend to rotate in a pattern that seems random. It's obviously not random, and it's the interpolation at play, but the experience is not at all intuitive.

ocornu, 2025-03-20:
Agreed. This has bugged me for a while as well.

sebastien-rosset, 2025-03-29:
Fixed in #4416

closed 2025-03-29 by sebastien-rosset