OpenCPN/OpenCPN #4406: Connection dialog: N2000 "driver not avaiable"

Hakansv, 2025-03-18:
**Describe the bug**
In Options->Connect. List of connections.
A N2000 connection won't display "Data flowing" status but instead "No driver statistics available" although data is flowing normally.
Compare now solved issue #4351 
It's the same for a N2k connection.

**To Reproduce**
Steps to reproduce the behavior:

    Connect a N2k data source
    Study status - "No driver..."
    Open NMEA Debug window and see that data is actually flowing.

**Expected behavior**
A similar solution as 291628b9a 

**Desktop (please complete the following information if applicable):**
 -Win10, Rpi Debian

**Screenshots**

[image]

leamas, 2025-03-18:
Not "Driver not available" but "Driver stats not available". 

The message is correct, reflecting that the patches to provide stats are yet not implemented in the n2k driver

bdbcat, 2025-03-18:
For reference and testing, what N2K gateway/interface do you have connected to COM3 ?

Hakansv, 2025-03-18:
I have a YDNU-02 in N2k-mode

Hakansv, 2025-03-19:
Corrected by 227888bb 
Thanks

closed 2025-03-19 by Hakansv