OpenCPN/OpenCPN #4412: Using arrow keys to pan chart has inconsistent movement (missing keystrokes?)

davidsanner, 2025-03-20:
When panning on a chart using keyboard arrows keys the charts doesn't pan consistently as if it is missing some key strokes.

This seems to only effect my linux/debian bookworm instance only as I don't see this behavior on my Mac.

Sometimes a single keystroke pans the chart, other times two or three keystrokes in a row are missed resulting in no movement.

When holding an arrow key down (repeating) it does pan but it can pan in fits/starts.

Finally, I've tried sending arrow keystrokes with xdotool and see a similar behavior where OC seems to be missing / not responding to some of the keystrokes.

I'm testing on a Raspberry Pi 5 w/ Debian & latest Bookworm with latest OC master branch.
I've seen this behavior on other builds for years including running on my older Pi4.

rgleason, 2025-03-20:
Can you possibly determine if there is some processing of the CPU taking place?

davidsanner, 2025-03-20:
The CPU is not being maxed out or anything if that's what you're wondering.   Is there a specific test I might try?   I was thinking it could have been my keyboard but my xdotool ruled that out.

I tried several versions of loop below with different delays and the chart panned irregularly like it was missing keys.
> while true ; do xdotool key --window 44040437 Left ; sleep .25 ; done

Also I should mention this is w/ CP on fullscreen mode using std X (not Wayland) with a std keyboard (wireless w/ dongle , not bluetooth).  I see no other delays w/ arrow keys in other apps/terminal.

I have another handheld bluetooth remote that I use for panning that sends arrow keys and it has behavior, CP seems to drop key strokes.   I think I've seen this w/ one and two charts display mode.

Thanks

Hakansv, 2025-03-21:
You could test to disable Options->Display->General: "Smooth Panning/Zooming" and check for a difference.
If that feature is enabled, is there a timer involved that could potentially cause quick keystrokes to be missed?

davidsanner, 2025-03-21:
@Hakansv I think you are on to something, thanks.   With that box unchecked the panning jumps are larger but consistent.  With it checked, it's more irregular.  I'm testing via VNC through a reverse ssh tunnel but the difference is pretty clear.    Locally on an iMac  it's less noticeable but at a closer viewing I see that with that box checked the arrow keys produce inconsistent behavior.  eg.  3 left-key presses followed by 3 right-key presses doesn't return to the same location,  even if the presses are slow (1/3 of sec a part).
Thanks again.

Hakansv, 2025-03-21:
Good! 
Now I wouldn't call this discussion a bug. Would you mind close this issue to reduce the burden of unresolved bugs for developers. :)

leamas, 2025-03-21:
So, making it to a discussion