OpenCPN/OpenCPN #4415: Proposed Sun Position and Day/Night State API

sebastien-rosset, 2025-03-22:
# Is your feature request related to a problem? Please describe

There are multiple use cases to obtain information for sunset, sunrise, dawn and dusk. This includes:
1. #2427 
2. https://github.com/rgleason/weather_routing_pi/pull/226

##  Weather Routing plugin Night Efficiency

In the Weather Routing plugin, I am adding efficiency factors that are used during isochron propagation.
1. **Upwind** efficiency modifies boat speed when sailing toward the wind source.
2. **Downwind efficiency** adjusts performance when sailing with the wind.
3. **Night efficiency** applies a cumulative performance factor during darkness hours.

For night efficiency, that means the isochron propagation algorithm needs to know whether it's day or night at each specific geographical position and time being evaluated, requiring accurate sunrise, sunset, and twilight calculations based on coordinates and UTC time. Without this capability, the algorithm cannot properly apply the configured night efficiency percentage that accounts for reduced sailing performance during darkness.

# Describe the solution you'd like

The WR plugin could do the sunrise/sunset calculation on its own, but this would create unnecessary code duplication. A centralized API in OpenCPN's core ensures consistent sun position calculations across all plugins and allows improvements in calculation accuracy to benefit the entire application.

In addition, plugins may benefit from being able to query and receive events related to day/night transitions, enabling them to automatically respond to changing light conditions. An event-based system would allow plugins to register for sunrise, sunset, or twilight notifications at the current location, triggering automatic actions like switching chart modes, adjusting instrument backlighting, or modifying routing parameters without requiring continuous polling.

# Proposed Sun Position and Day/Night State API

I propose adding a set of functions to OpenCPN's plugin API that would provide standardized access to sun position data and day/night state information. These functions would:

1. Calculate sun position (elevation and azimuth) for any given location and time
2. Determine sunrise, sunset, and solar noon times for any location on a given date
3. Provide day/night/twilight state determination with configurable twilight definitions
4. Optionally offer an event notification system for day/night transitions

These functions would benefit not just the Weather Routing plugin but also dashboard displays, navigation tools, and the proposed automatic night mode switching. The API would handle all astronomical calculations internally, ensuring accuracy and consistency across the application.

# Proposed API Code

This would be added in `ocpn_plugin.h`.

```c++
/**
 * Calculate sun position for a specific location and time
 * 
 * @param lat Latitude in degrees (-90 to 90)
 * @param lon Longitude in degrees (-180 to 180)
 * @param datetime Date and time in UTC
 * @param elevation Output: Sun elevation angle in degrees above horizon (negative when below)
 * @param azimuth Output: Sun azimuth in degrees from true north (0-360)
 * @return true if calculation was successful
 */
bool GetSunPosition(double lat, double lon, wxDateTime datetime, 
                    double* elevation, double* azimuth);

/**
 * Calculate times of sunrise, sunset and solar noon
 * 
 * @param lat Latitude in degrees (-90 to 90)
 * @param lon Longitude in degrees (-180 to 180)
 * @param date Date in UTC (time component is ignored for calculation)
 * @param sunrise Output: Time of sunrise as seconds since midnight UTC (or -1 if no sunrise)
 * @param sunset Output: Time of sunset as seconds since midnight UTC (or -1 if no sunset)
 * @param noon Output: Time of solar noon as seconds since midnight UTC
 * @param twilight_angle Angle of sun below horizon to use for twilight definition in degrees
 *                      (6° for civil, 12° for nautical, 18° for astronomical)
 * @return true if calculation was successful
 */
bool GetSunEvents(double lat, double lon, wxDateTime date,
                  double* sunrise, double* sunset, double* noon,
                  double twilight_angle = 6.0);

/**
 * Determine the day/night state at a specific location and time
 * 
 * @param lat Latitude in degrees (-90 to 90)
 * @param lon Longitude in degrees (-180 to 180)
 * @param datetime Date and time in UTC
 * @param twilight_angle Angle of sun below horizon to use for twilight definition in degrees
 * @return DAY_NIGHT_STATE enum value (DAY, TWILIGHT, or NIGHT)
 */
enum DAY_NIGHT_STATE {
    DAY_NIGHT_STATE_DAY = 0,
    DAY_NIGHT_STATE_TWILIGHT = 1,
    DAY_NIGHT_STATE_NIGHT = 2
};

DAY_NIGHT_STATE GetDayNightState(double lat, double lon, wxDateTime datetime,
                                 double twilight_angle = 6.0);

/**
 * Invoked when there is a transition.
 */
virtual void OnDayNightStateChanged(DAY_NIGHT_STATE new_state);
```

# Example usage to automatically switch color mode.

```c++
 void OnDayNightStateChanged(DAY_NIGHT_STATE new_state) override {
        switch (new_state) {
            case DAY_NIGHT_STATE_DAY:
                SetColorScheme(GLOBAL_COLOR_SCHEME_DAY);
                break;
                
            case DAY_NIGHT_STATE_TWILIGHT:
                SetColorScheme(GLOBAL_COLOR_SCHEME_DUSK);
                break;
                
            case DAY_NIGHT_STATE_NIGHT:
                SetColorScheme(GLOBAL_COLOR_SCHEME_NIGHT);
                break;
        }
    }
```

# Example usage in weather routing

Pseudo-code:

```c++
// Inside the isochron propagation loop where we propagate from position to new positions
void IsoChron::Propagate(Position current, time_t time) 
{
    // Calculate boat speed based on wind conditions
    double boat_speed = GetBoatSpeed(current.lat, current.lon, time);
    
    // Apply upwind/downwind efficiency factors
    if (IsUpwind(current.course, wind_direction))
        boat_speed *= m_upwindEfficiency / 100.0;
    else if (IsDownwind(current.course, wind_direction))
        boat_speed *= m_downwindEfficiency / 100.0;
    
    // Apply night efficiency if needed using the new API
    DAY_NIGHT_STATE state = GetDayNightState(current.lat, current.lon, 
                                            wxDateTime(time), 
                                            6.0);  // Using civil twilight definition
    
    if (state == DAY_NIGHT_STATE_NIGHT || state == DAY_NIGHT_STATE_TWILIGHT)
        boat_speed *= m_nightEfficiency / 100.0;
    
    // Continue with position propagation using the adjusted boat_speed
    PropagatePosition(current, boat_speed, time);
}
```

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

sebastien-rosset, 2025-03-22:
@bdbcat , @leamas , what do you think about this proposal? I could implement it.

sebastien-rosset, 2025-03-24:
1. Solar Position Calculations:  calculate sun elevation or azimuth angles
2. Terminator Functions: calculate the day/night boundary
3. Twilight Definitions: civil, nautical, or astronomical twilight boundaries
4. Date/Time Calculations: Functions for converting between calendar dates and Julian day numbers

Algorithms to consider: Jean Meeus' Algorithms, NOAA Solar Calculator, US Naval Observatory's VSOP87, Almanac for Computers, 1990" from the US Naval Observatory

sebastien-rosset, 2025-03-24:
> To me, seems a bit too much for a plugin API function. Consider adding to the opencpn-libs repo, thus automatically available to all plugins. I know, adds a little complication to the proposed auto dark mode option, but this is what the opencpn-libs repo is intended to be used for.

Thanks for the guidance.
If I add a sun calculator capability in `opencpn-libs`, I could use this library in the weather plugin to differentiate routing based on conditions with a point in polygon check.

Then down the road, we could piece multiple capabilities together:
1. Add timeline widget in OpenCPN core as described in https://github.com/OpenCPN/OpenCPN/discussions/4126
2. Add a day/night/twilight overlay as a function of time - integrate with the above sun calculator library and timeline widget. Example visualization below.
3. GRIB plugin would eventually use the  timeline widget instead of its own custom UI.

[image]

rgleason, 2025-03-24:
Sebastien, you probably have this well in hand, but we've been working on Celestial Navigation, and I'll bet that Stelian who has been doing the programming has some code that you could use. Also see the Number 92 Issue under Cel Nav pi which has C++ Program sources.

sebastien-rosset, 2025-03-24:
> Sebastien, you probably have this well in hand, but we've been working on Celestial Navigation, and I'll bet that Stelian who has been doing the programming has some code that you could use. Also see the Number 92 Issue under Cel Nav pi which has C++ Program sources.

As @bdbcat  has pointed out, it makes sense to have this kind of code in `opencpn-libs`. Multiple components (core and plugins) need this kind of library. I'd rather use an existing library rather than add all this in the weather routing plugin.

rgleason, 2025-03-25:
Yes, I know. Stelian could direct you to that part of the code, or alternatively we could also use your  files added to opencpn-libs for day-night, in Celestial Navigation, because that is one of the things Sean had on his to do feature list.

@stelian42  Maybe Sebastien will have some opencp-lib files to use for day-night.