OpenCPN/OpenCPN #4418: Deviation Plugin in MacOS 5.10.2 does not load when selected

Arne2024, 2025-03-23:
**Describe the bug**
Deviation Plugin in MacOS version 5.10.2 does not load when selected, and can not be loaded from plugin utility

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'plugins'
2. Click on 'Deviation'
3. Launch 'OpenCPN'
4. See error message:     '' The following plugin found errors during startup               Deviation.      This plugin will be reinstalled. You might want to reinstall after updating the catalog''. 

**Expected behavior**
after updating the catalog, to versions dated 2025-03-09 15:08, (more recent seems not to work today is March 23) 
one would expect the Deviation plugin can be downloaded,  but not,  so the same error when starting openCPN

**Operating system):**
 - MacBook Pro M1 
 - OS: [macOS] Sequoia 15.3.2
 - Version [15.3.2]

rgleason, 2025-03-23:
This should be closed and moved to @RooieDirk s deviation Issues.

I cannot transfer it there.
It should be closed here in Opencpn by the OP.

Arne2024, 2025-03-23:
can you advise how to move this to @RooieDirk - thanks

Btw, I wonder if I am the only one, this Deviation error keeps popping up, whenever I launch openCPN and it cannot be installed or uninstalled from PlugInn's : not responsive.

closed 2025-03-23 by Arne2024

rgleason, 2025-03-23:
look for a repo like github.com/rooiedirk/deviation_pi
once in there, look for Issues in the top menu.

If you don't find it then you will have to use the Issues in my fork of rooiedirk/deviation_pi  under github.com/rgleason