OpenCPN/OpenCPN #4423: libunrar-dev appears to be mispelled in the ci/control dependancies Build-Depends

darsys, 2025-03-30:
I struggled to get the system the system to install build dependencies and found a typo (I believe) in the ci/control file.   

Line 83:  
libunarr-dev | base-files (<< 12),

I believe the corrected line should be:
libunrar-dev | base-files (<< 12),

nohal, 2025-03-30:
No, `libunarr` is the correct decompression library.
Also, the current https://github.com/OpenCPN/OpenCPN/blob/master/ci/control sure does look different than the one you are using as it does not even have 83 lines in total.

nohal, 2025-03-30:
@leamas what's the status of `libunarr` in future Debian releases? I can see it removed from unstable and testing since 9/2024

nohal, 2025-03-30:
If the system you are using does not have `libunarr` and does have new enough `libarchive` (> 3.7.3), `libunarr` is not needed anymore, because these new enough versions of `libarchive` provide the needed functionality directly.
Libunarr IS needed on all the systems based on current stable Debian up to Bookworm. New enough libarchive seems to be available in the future Debian Trixie (and anything derived from that).
LibunRAR is NOT needed by OpenCPN as a dependency.

You may see https://github.com/OpenCPN/OpenCPN/blob/master/plugins/chartdldr_pi/CMakeLists.txt for reference

darsys, 2025-03-30:
I apologize for not mentioning this and for the assumptions. I am using Debian Testing (Trixie). I could not find a package for libunarr which makes sense now.

leamas, 2025-03-31:
@nohal  It is removed. 

Current master  is patched to use libarchive  if it's version is high enough, it has working rar support. See 4e9c1ee1e.  Same patch is applied to the Debian 5.10 package.

leamas, 2025-03-31:
The entry reads `libunarr-dev | base-files (<< 12)`.   Looks like a typo, it basically means that it's satisfied if unarr-dev is available or Debian  < 12.

But, it should be the other way around:  `libunarr-dev | base-files (>> 12)`. That is, either unarr-dev or Debian 13+. So, a typo.

OTOH, why don't we state the actual condition, it should be much more clear: `libunarr-dev | libarchive-dev(>=3.7.4)`

TL;DR. This is a bug we should somehow fix

leamas, 2025-03-31:
... have  test build running.

leamas, 2025-03-31:
Fixed in  9a84cb9f6, closing.

@darsys  Thanks for filing!

closed 2025-03-31 by leamas