OpenCPN/OpenCPN #4431: Route point GetETD is incorrect

nocodehummel, 2025-04-01:
Route point `GetETD()` incorrectly returns `GetETA()` (in route_point.cpp). The `GetETA()` must be invalid for the function to return an `etd`.

```
if (!GetETA().IsValid() || m_seg_etd > GetETA()) {
  return m_seg_etd;
} else {
  return GetETA();
}
```

Should be after check that `GetETD().IsValid()`.

```
if (GetETA().IsValid() && m_seg_etd < GetETA()) {
  return m_seg_etd;
} else {
  return GetETA();
}
```

sebastien-rosset, 2025-04-01:
I will investigate this issue.

sebastien-rosset, 2025-04-01:
Hmm, it makes sense to me that if ETA is not valid `!GetETA().IsValid()`, then it's ok to return the manually set ETD. That can happen if for some reason, OpenCPN is unable to calculate the ETA. For example, the speed is very low.
In that case, there is uncertainty about the ETA, or the ETA is unknown. However, the user has specified the ETD manually, so IMHO OpenCPN decides to trust the user judgement.

Perhaps @bdbcat has some insights.

I did make a few changes in #4434.
AFAIK, if a waypoint is the last waypoint of a route, the ETD does not make sense.

nocodehummel, 2025-04-01:
I just cannot get my head around; when the the ETA is valid do NOT return the ETD - `!GetETA().IsValid()`.

sebastien-rosset, 2025-04-01:
> I just cannot get my head around; when the the ETA is valid do NOT return the ETD - !GetETA().IsValid().

Suppose we have the following scenario:
1. You are sailing at 6 knots and you have activated a route.
2. The next active waypoint is named `Tahiti`
3. After the `Tahiti` waypoint, the next waypoint is `006`.
4. Based on your VMG, the ETA for the next waypoint (`Tahiti`) is estimated to be 3 days from now.

**Scenario 1:**
You have not specified the ETD for `Tahiti`. OpenCPN assumes that immediately after you arrive at `Tahiti`, you will depart for `006`. So the ETD is the same as the ETA.

**Scenario 2:**
You have specified a ETD at `Tahiti`, but it is before the ETA. In that case, the ETD is ignored. OpenCPN assumes you will depart from `Tahiti` at the ETA.

**Scenario 3:**
You have specified a ETD at `Tahiti`, which is 5 days after the ETA. So effectively you are telling OpenCPN that you want to enjoy a good time in Tahiti, and then you will depart for `006`.

So IMHO, it makes sense that `GetETD` returns the manual ETD even if the ETA is valid.

nocodehummel, 2025-04-02:
> I just cannot get my head around; when the the ETA is valid do NOT return the ETD - !GetETA().IsValid().

The code is written as such that when the ETA is valid, don't return the ETD. Which means that a route point with a valid ETD and ETA will never return the ETD. That is why I reported this issue.

sebastien-rosset, 2025-04-02:
> The code is written as such that when the ETA is valid, don't return the ETD. Which means that a route point with a valid ETD and ETA will never return the ETD. That is why I reported this issue.

Hmm
```c++
if (!GetETA().IsValid() || m_seg_etd > GetETA()) {
```

The code is written such that:
IF the ETA is NOT valid
OR
The ETD is greater than the ETA

which is the same as:

If the ETA is NOT valid
OR
The ETA is valid AND the ETD is greater than the ETA

To be sure I verified with a debugger and testing. Or maybe I misunderstand what you are asking.

nocodehummel, 2025-04-02:
> To be sure I verified with a debugger and testing. Or maybe I misunderstand what you are asking.

Yes. This is not a critical issue so it can wait for someone who knows the code to respond.

nocodehummel, 2025-04-05:
> I'm not following. The route manager is designed such that the ETD can be any time after the ETA, so ETD is not necessarily the same as ETA.

@sebastien-rosset given the meaning of ETD and ETA the above does not make sense.

nocodehummel, 2025-04-06:
Okay, I was slow on getting the IF clause, but with some testing figured it out. The code is correct.

closed 2025-04-06 by nocodehummel