OpenCPN/OpenCPN #4432: Route point next tide event has incorrect time

nocodehummel, 2025-04-01:
The route points list in the route manager shows incorrect times for the next tide events (when compared to the tide chart). Also the 09.49 LW event is not identified as next tide event, when route point ETA is 09.18.

Steps to reproduce the behavior:
1. Open a route in the route manager (UTC).
2. Open way-point properties for a route point.
3. Set tide station (extended properties).
4. Open tide dialog for same day as route point (UTC).
5. Active way-point dialog and click 'OK'  to close the dialog.
6. Show column  'Next tide event' in route point table.
7. See difference in HW/LW times.

The expected behavior is that the correct HW/LW times are displayed in the route. 

 - OS: Windows
 - Version: master (6d6521360613e751b76fbbb54a7e48def531b5eb)

[image]

In the screenshot all times are in UTC. Since the different is in whole hours it is likely related to timezone conversion.

nocodehummel, 2025-04-01:
The results are different on 52bf97a12, which could mean that the issue is related to #4052. Also the ETA is shown different in UTC between the two software versions.

[image]

nocodehummel, 2025-04-01:
@sebastien-rosset, any comments?

nocodehummel, 2025-04-01:
Portsmouth HW/LW values in the tide dialog match published times (and should be considered correct).

sebastien-rosset, 2025-04-01:
I didn't change this part of the code as part of #4052. The challenge is to find all code locations where time is displayed, and use the new date/time API function to ensure time is formatted according to the centralized date/time configuration.

The formatting of the ETD attribute below should follow the centralized options (local time versus UTC):

[image]

Relevant Code:
https://github.com/OpenCPN/OpenCPN/blob/310453809db872a8548a707dd1a095258f0fd8b5/gui/src/MarkInfo.cpp#L638
https://github.com/OpenCPN/OpenCPN/blob/310453809db872a8548a707dd1a095258f0fd8b5/gui/src/MarkInfo.cpp#L650
https://github.com/OpenCPN/OpenCPN/blob/310453809db872a8548a707dd1a095258f0fd8b5/gui/src/MarkInfo.cpp#L1677
https://github.com/OpenCPN/OpenCPN/blob/310453809db872a8548a707dd1a095258f0fd8b5/gui/src/tcmgr.cpp#L4174

nocodehummel, 2025-04-01:
> I didn't change this part of the code as part of [#4052](https://github.com/OpenCPN/OpenCPN/pull/4052). The challenge is to find all code locations where time is displayed, and use the new date/time API function to ensure time is formatted according to the centralized date/time configuration.

Does that mean that the time display was not changed in the route dialog? How can it be that the time is displayed different before and after the merge commit?

sebastien-rosset, 2025-04-01:
> > I didn't change this part of the code as part of [#4052](https://github.com/OpenCPN/OpenCPN/pull/4052). The challenge is to find all code locations where time is displayed, and use the new date/time API function to ensure time is formatted according to the centralized date/time configuration.
> 
> Does that mean that the time display was not changed in the route dialog? How can it be that the time is displayed different before (correct) and after (incorrect) the merge commit?

Sorry I wasn't specific enough. I didn't change the date/time formatting of the tide stations, but I did modify the ETE/ETA columns in the routes. I'll take a look.

sebastien-rosset, 2025-04-01:
[image]

nocodehummel, 2025-04-01:
> Sorry I wasn't specific enough. I didn't change the date/time formatting of the tide stations, but I did modify the ETE/ETA columns in the routes. I'll take a look.

Thanks, as mentioned above the Portsmouth HW/LW are correct (compared to the online published times).

sebastien-rosset, 2025-04-01:
[image]

nocodehummel, 2025-04-01:
> > Also the 09.49 LW event is not identified as next tide event, when route point ETA is 09.18.

On line two the ETA is displayed as 9.18 UTC in the initial screenshot (partial visible) which means the next tide event should be LW 09.49 UTC. The next tide event is however HW 03:05:44 UTC. It could be that the ETA is displayed incorrect because the screenshot with the correct time shows 11.18 UTC. Than the next tide event is correct.

sebastien-rosset, 2025-04-04:
This should be fixed in #4434

rgleason, 2025-04-24:
@nocodehummel   this should be fixed, if so please test and close this Issue. Original Person closes.