OpenCPN/OpenCPN #4435: When accelerated graphics is turned off there is a crash.

rgleason, 2025-04-01:
**Describe the bug**
When accelerated graphics is turned off there is a crash.
Then when it is Opened again there is a message that there was a crash yes no.
Then it appears that accelerated graphics is turned off and things are normal again without acel graphics.

**To Reproduce**
Turn off accelerated graphics.

**Expected behavior**
If Opencpn must close and then be reopened, it should not crash first.

**Desktop (please complete the following information if applicable):**
 - OS:Win11
 - Version O 5.10.2

transmitterdan, 2025-04-01:
I've been running latest master code with OpenGL turned off for a while. I test each commit that I can. I've never seen a crash this year.

Can you give some hint as to anything that led to the crash? Only using the 4 in-built plugins or some others?

bdbcat, 2025-04-01:
On O5.0.2, I expect the trouble is coming from a plugin.

rgleason, 2025-04-01:
I have 

- Climatology
- Grib
- Ocpn_Draw
- Weather_routing

enabled.  Nothing else.  Once it crashes and I start O back up, and renable the plugins, everything works ok.
It is just closing OpenGL acceleration causes the crash.  Upon restart things are ok.

PS: Also Dave, we need to remove rtlsdr_pi  it is causing crashes and misbehaving. I will add an issue for that too.

bdbcat, 2025-04-01:
Rick...
Please disable the plugins, one-by-one, and verify the fault still occurs.
If you still get a fault with no plugins activated, then we may need more information.

rgleason, 2025-04-01:
Quite certain it is Climatology.

rgleason, 2025-04-01:
https://github.com/rgleason/climatology_pi/issues/61

closed 2025-04-01 by rgleason