OpenCPN/OpenCPN #4436: Nothing is displayed

wangdemon, 2025-04-02:
**Describe the bug**
I use the Windows 10 system, VS2019 compiler, and OpenCPN version 5.10.0

When I started the program for the second time and set the charts directory, nothing was displayed anymore

I use the Win10 system and VS2019 compiler to compile OpenCPN source code

The OpenCPN version is 5.10.2

**Screenshots**

[image]

**Desktop (please complete the following information if applicable):**
 - OS: Wins10
 - Version: 5.10.2

leamas, 2025-04-02:
I'd suggest that you first try the windows package available at https://opencpn.org/OpenCPN/info/downloadopencpn.html. If this doesn't show this behaviour the problem is about your build, otherwise not. That might be a starter

rgleason, 2025-04-02:
Alec is right, do what he suggests. 
but this might just be a path setting for charts, etc. 
But shapefiles does not seem to be found.
You don't provide enough information.

leamas, 2025-04-06:
@wangdemon  Ping ?

wangdemon, 2025-04-07:
Maybe my English is not good and I didn't express my viewpoint clearly

I downloaded the opencpn.exe program and it runs normally

After compiling the opencpn source code, I was able to see the map for the first time

On the second run, I can't see the map anymore

https://github.com/user-attachments/assets/44a3a50a-3858-422d-97c1-7f9399d55564

Hakansv, 2025-04-07:
If the normal installation runs as expected the fault can be your debug use?
Does the Debug run have all (basemap) files available? Try RelWithDebInfo mode instead?

wangdemon, 2025-04-07:
I understand the reason, 

it's because the opencpn.ini configuration did not correctly record the path where the map is located

What confuses me is that when I press the Ctrl+F5 shortcut key in Visual Studio 2019, the configuration file opencpn.ini is automatically generated in the path where the opencpn.exe file is located

When I double-click the opencpn.exe file, the configuration file opencpn.ini is generated in the "C:\ProgramData\opencpn" directory

Moreover, every time the opencpn.exe program is run, the values of "Directors/BasemapDir" and "Directors/BaseShapefileDir" in the configuration file opencpn.ini will change

nohal, 2025-04-07:
You don't say it, but I suppose you used `winconfig.bat`. If it is so, Ctrl+F5 starts opencpn in portable mode (with -p switch, you may check the properties of your project to confirm), while double clicking opencpn.exe of course does not.
There is no bug, everything works as it should as far as I read your information.

leamas, 2025-04-09:
This is not really an opencpn bug. That is **not** to say that this should not be discussed, these are important issues. But a discussion is a better place to do this. 

So, converting to a discussion. To repeat, this is absolutely about discouraging further discussion.