OpenCPN/OpenCPN #4437: Function "Set Scale" is (recently?) destroyed

Hakansv, 2025-04-02:
**Describe the bug**
The function "Set scale" does initially scale to desired scale but immediately zoom out again to a very small scale.

**To Reproduce**
Steps to reproduce the behavior:
1. Click on the chart canvas bar the scale info
2. In the "set scale" dialog enter for example: 1000
3. The new chart scale is shown but immediately it zooms out to very small scale instead.
4. Also the AIS target list "Center-Info-close" uses the same ChartCanvas::DoZoomCanvas() function and the behavior is consequently the same; Sets and show a large scale. The it's zooming out to a very small scale.

**Expected behavior**
Both "Set scale" and "Center-Info-close" should show desired chart scale and stay there.

**Desktop (please complete the following information if applicable):**
 - OS: Win10
 - Version:Today's Git master code

Hakansv, 2025-04-02:
For info.
This issue is not all time reproducible. Clear dependencies are hard to find. Try a couple of times with different view port.
The AIS target list "Center-Info-close" is more reliable for reproducing than "Set scale".

ironpowertga, 2025-05-08:
Yes i confirm the issue

sebastien-rosset, 2025-05-11:
I had a similar issue yesterday. I was running OpenCPN master. I was trying to use `+` and `-` keys to zoom in and out. I could see OpenCPN was trying to zoom in and out, but then immediately got back to the initial scale. Same effect with the zoom in/out button clicked with a mouse.

bdbcat, 2025-05-11:
Probably quilt turned OFF due to errant "q" keystroke?

sebastien-rosset, 2025-05-11:
> Probably quilt turned OFF due to errant "q" keystroke?

Yes, quite possible. 

For this issue, I wonder if there's a conflict between different movement/zoom operations, i.e. with `DoMovement`.

sebastien-rosset, 2025-05-12:
It seems there is a conflict between:
1. `ChartCanvas::SetViewPoint`, which sets the viewport scale directly; and
2. `DoMovement()` which aims to gradually change the scale factor.

When `SetVPScale()` is called to set a new scale, it does so directly, but:

1. If there's already an active movement timer running (started by mouse wheel zoom, keyboard zoom, or other UI interactions)
2. Then `DoMovement()` will continue to be called by the timer
3. Inside `DoMovement()`, it checks the current scale against `m_zoom_target`
4. Since `m_zoom_target` wasn't updated when `SetVPScale()` was called, `DoMovement()` will try to "fix" the scale to reach the previous target
5. This causes the immediate zoom out that users are experiencing

The issue is effectively a race condition between direct scale setting and the smooth animation system. When you set a scale directly but the animation system is still active and targeting a different scale, the animation "corrects" what it sees as a deviation from its target.

To fix this, whenever `SetVPScale()` is called directly, it should either:
1. Call `StopMovement()` to cancel any ongoing animations, or
2. Update `m_zoom_target` to match the new scale so the animation system doesn't try to change it

There are about 20 callers to `ChartCanvas::SetViewPoint`, and it's possible call paths were changed.

bdbcat, 2025-05-12:
Got it, thanks for the analysis.
I'll probably fix DoMovement to update the target...maybe

Hakansv, 2025-05-17:
On Win10:
Now both AIS target list "Center-info-close" and "Set scale" stays at the desired scale. 
So far so good.

But, for the AIS target list function:
The used chart file scale is not changed when zoomed. We need to update the viewport to show appropriate detailed chart.
Now I afterwords have to make a zoom action to show correct chart details.

Hakansv, 2025-05-18:
Now solved by bb28e66 
I'm closing. Pls comment if other thoughts.

closed 2025-05-18 by Hakansv