OpenCPN/OpenCPN #4439: TTG calculation not correct with SOG & VMG

svMigration, 2025-04-04:
**Describe the bug**
The TTG calculation in the ROUTE box should change when it is ETA @VMG or ETA @SOG. However, at times the calculation does not seem to be happening.

**To Reproduce**
1. Travel along a route
2. Head in a direction that makes your VMG to the next WP very slow while your SOG stays the same
3. Click on the ROUTE box to change from @VMG to @SOG and compare the TTG values.
4. Both values seem to be based on VMG

**Expected behavior**
TTG should be distance RNG / VMG and RNG / SOG. 

**Screenshots**
Notice the COG compared to the BRG. These screenshots were taken just a few seconds apart. VMG of 1.18-1.21kts. SOG of 4.0-4.1 kts.

SOG

[image]

VMG

[image]

**Desktop (please complete the following information if applicable):**
Edition	Windows 11 Pro
Version	24H2
OS build	26100.3476

Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.

rgleason, 2025-04-24:
@sebastien-rosset   Is this related to the global time?  Can it be fixed for the release?

sebastien-rosset, 2025-04-24:
> @sebastien-rosset   Is this related to the global time?  Can it be fixed for the release?

No, this is a separate issue.

leamas, 2025-04-28:
@svMigration : I have been able to reproduce this and have a patch at hand in  #4507.  Still, if possible I would appreciate if you could test it.

Please let me know if you could do such a test.

svMigration, 2025-04-28:
@leamas: Sorry. We're in the boatyard so I can't test now. But glad you found it.

leamas, 2025-04-28:
OK, let's merge it anyway. I'm pretty confident.

Thanks for reporting!

PR #4507 (2025-04-28): concanv: Don't toggle ETA visibility - #4439
I can't see that the visibility of ETA should be affected by the change from using SOG to VMG or the other way around. The bug is caused by this, triggered by a @VMG -> @SOG change every second time.

Closes: #4439
commit 89dfe94d (2025-04-28): concanv: Don't toggle ETA visibility - #4439

I can't see that the visibility of ETA should be affected by the change
from using SOG to VMG or the other way around. The bug is caused by this,
triggered by a @VMG -> @SOG change every second time.

Closes: #4439
closed 2025-04-28 by leamas