OpenCPN/OpenCPN #4441: Changed function on "Safe mode" dialog. Keyboard key: "Enter" will now enter safe mode

Hakansv, 2025-04-04:
**Is your feature request related to a problem? Please describe.**
The new "Safe mode" goes default to Safe mode instead of previous no action. "Safe mode"  is highlighted instead of "Cancel".

**Describe the solution you'd like**
"Cancel" would be the default action as before.

**Additional context**
The new 15 seconds timer is good and defaults to "Cancel".

nocodehummel, 2025-04-05:
@Hakansv thanks I will have a look.

nocodehummel, 2025-04-05:
@Hakansv I had some doubts about the name of the cancel button. In principle following UI guidelines, but in this case *Cancel* is maybe a bit misplaced. What do you think?

Hakansv, 2025-04-05:
I think a user perceives Cancel as "Do nothing, continue as usual", so it probably works well.  
For example, "Start normally" is more specific, but I not sure it increases understanding.

Hakansv, 2025-04-05:
@nocodehummel 
Your new commit looks fine to me. 
A thought, did you checked the expired timer still end up in a "Cancel" function?
Thanks

nocodehummel, 2025-04-05:
> A thought, did you checked the expired timer still end up in a "Cancel" function? Thanks

Yes, the timer is working as before -> normal mode.

TwoCanPlugIn, 2025-04-05:
Given the dialog text asks the question "Do you want to run in safe mode without plugins and other possibly problematic features?", I would have thought Yes and No are unambiguous.

Some users may find "Cancel" to mean cancel starting in Safe mode.

rgleason, 2025-04-05:
Either way is clear IMHO.

Hakansv, 2025-04-06:
So a string like "Start normally" may be a good advice correctly interpreted by everyone?

nocodehummel, 2025-04-06:
> Some users may interpret "Cancel" to mean cancel starting in Safe mode.

Indeed *Cancel* will cancel starting in *Safe Mode*.

nocodehummel, 2025-04-06:
This was changed to adhere to GUI guidelines. The affirmative button should be labelled with an imperative verb that is more clear than: OK or YES. *Safe Mode* is a button with a clear label. As I wrote above I had some doubt about *Cancel*, but that is recommended in the guideline.

bcn2, 2025-04-06:
Guidelines are that, not a solution for every case - if the context recommends a different option, take it.

leamas, 2025-04-06:
@nocodehummel  When referring to guidelines in a discussion like this, a link  to the relevant section is customary and useful.

nocodehummel, 2025-04-07:
> Label the affirmative button with a specific imperative verb, for example: Save or Print. This is clearer than a generic label like OK or Done.

Guidelines for GNome HIG [Dialog](https://developer.gnome.org/hig/patterns/feedback/dialogs.html)

leamas, 2025-04-07:
Thanks.

Well, looking into that the advise on using imperative verbs are for a Action Dialog. However, this is to me more like a Error Dialog,  so he advise does IMHO not really apply here.

Anyway, changing the current Yes/No to something else will if course cause some user confusion. This means that we IMHO needs stronger arguments than presented so far to change it.

nocodehummel, 2025-04-07:
@leamas IMO this dialog does not have the characteristics of an Error Dialog as per [Dialogs](https://developer.gnome.org/hig/patterns/feedback/dialogs.html) definition. This dialog asks the users decision before executing an action (start in safe-mode). 

> Action dialogs present options and/or information about an action, before it is carried out.

> Error dialogs are another type of alert dialog. These present an error message to the user and often include just a single button to close the dialog. Error dialogs should be avoided where possible, since they are disruptive. For simple non-critical errors, [toasts](https://developer.gnome.org/hig/patterns/feedback/toasts.html) can be a good alternative.

The *Yes*/*No* alternative requires more effort from the user. It is an answer to a question, so the user should have understood the question to be able to take a decision. The purpose of using a specific "named" button is that it is clearer and requires less effort. 

As any changes existing users need to get used to them, and reactions to changes are mixed. The ambition for following UI guidelines is to improve usability in the long run (for existing and new users).

leamas, 2025-04-07:
Well , we should perhaps not spend too much time splitting hairs over interpreting the GL. I insist that this is not an Action dialog, but  Error dialog  was admittedly also wrong. This is more of a confirmation dialog. Yje nre result is the same.

Lets agree to disagree. 

This  means that you could either use Yes/No or try to convince Dave about you arguments.  I am certainly not.

nocodehummel, 2025-04-07:
The other alternative is the [*Alert Dialog*](https://developer.gnome.org/hig/patterns/feedback/dialogs.html), for which IMO the same applies to label the buttons, although that is not written specifically in the GL, the example is clear.

[image]

> Alert dialogs present a message or question, along with between one and three buttons with which to respond. Alert dialogs are an appropriate choice when it is essential that the user sees and responds to a message.

Furthermore I think we should have good reasons to deviate from the GL (otherwise it looses value).

@bdbcat what are your thoughts on this?

bdbcat, 2025-04-08:
My thoughts are coming from the standpoint of casual computer users who are using OCPN to navigate their boat.
So I would forget the Guidelines here, and actually guide the user to an expected result.
I would like to see:
"Normal startup"   "Safe mode startup"
"Normal" being the default, activated by "Enter" key.

nohal, 2025-04-08:
I have been missing for some time so excuse me if I will repeat something obvious, but a choice between "Cancel" and "Safe mode" that I have just had after killing opencpn is the most unintuitive I have seen in ages, guidelines or not.

[image]

I am not a native English speaker,  but does anybody really seriously think that the answer "Do you want to run in safe mode without plugins and other possibly problematic features?" can be anything but "Yes" or "No"?

bdbcat, 2025-04-08:
Sadly, I fear we cannot assume that the user will read and grok the message, however nicely phrased in English "Do you want...".

nohal, 2025-04-08:
The user will keep hitting Enter and maybe making the application crash until he eventually reads the message if hitting Enter "does not work"
The user will also usually never click on the known problem link, but that will not stop me from updating that page, because I have seen more than zero of them who did.
Anyway, what we have now maybe looks better and conforms to some guideline, but is worse than what was there before as far as I'm concerned.

bcn2, 2025-04-08:
> My thoughts are coming from the standpoint of casual computer users who are using OCPN to navigate their boat. So I would forget the Guidelines here, and actually guide the user to an expected and useful result. **I would like to see: **"Normal startup" "Safe mode startup"** "Normal" being the default, activated by "Enter" key**.

That's would I prefer, a clear indication about the choices.

"Cancel" implies for me "undo", "abort", "go-back". Back to where?

nocodehummel, 2025-04-08:
Thanks for the feedback and will implement an explicit label.

commit ca3abfaa (2025-04-10): libs/gui: Safe mode dialog: update button labels - #4441

Closes: #4441
closed 2025-04-10 by leamas