OpenCPN/OpenCPN #4445: Route point ETA is incorrect.

nocodehummel, 2025-04-06:
The route point ETA is incorrect after PR #4052. 

Steps to reproduce the behavior:
1. Start the specific OpenCPN version.
2. Open a route in the route manager (UTC).
3. See changed ETA.

Screenshot  on master (after #4052).
[image]

Screenshot on 52bf97a12 (before #4052).
[image]

 - OS: Windows
 - Version: master (6d6521360613e751b76fbbb54a7e48def531b5eb)

Additional info: linked to #4432.

nocodehummel, 2025-04-06:
@sebastien-rosset I think you have this covered in PR #4434, but important enough to track as separate issue.

sebastien-rosset, 2025-04-06:
Yes, this is fixed by #4434

sebastien-rosset, 2025-04-08:
This should have been resolved with #3334

nocodehummel, 2025-04-08:
@sebastien-rosset I will verify and close when done with PRs review.

bdbcat, 2025-04-26:
Ready to close?

sebastien-rosset, 2025-04-26:
I think so but it would be better if the submitter could close it

sebastien-rosset, 2025-05-02:
@nocodehummel , can we close this issue?

rgleason, 2025-05-03:
@nocodehummel   You are the OP.

closed 2025-05-14 by nocodehummel