OpenCPN/OpenCPN #4449: "Send to GPS" does not work and turns off any filters on the outgoing port.

Hakansv, 2025-04-06:
**Describe the bug**
"Send to GPS"  don't work.

**To Reproduce**
Steps to reproduce the behavior:
1. Key "E" to watch the Debug window.
2. Right click a route and select "Send to GPS" (Any previous used port) or a new port.
3. The "new" port can be any serial or network port on the drop-down list.
4. Click OK
5. The error message: "Error on Route Upload. GPS driver not available"occurs.
6. The Debug window shows no route transfer.
7. If the selected (serial) port has a output filter it's by-passed from now on until the connection is toggled off-on.
8. See in the Debug window that the output port now transfer all sentences received.

**Expected behavior**
The "Send to GPS" would transfer the route sentences on the selected port.
The function should not disturb any output filter on the selected port.

**Screenshots**
See shots and more descriptions here: https://github.com/OpenCPN/OpenCPN/pull/4440#issuecomment-2781293105

**Desktop (please complete the following information if applicable):**
 - OS: Windows and RPi Debian
 - Version: Git Master as of today. (96ea394df6)

**Additional context**
The error text "GPS driver not available" is not really clear? "Driver" for what? The serial converter? The GNSS device driver for any reason? Or does it refer to the connection source in any form?

Hakansv, 2025-05-06:
The "Send to GPS" is now working again.

The "bypass output filter" is "partly" solved:
When the route messages, xxWPL,xxRTE, are sent also whatever existing N0183 input messages are sent for a while. The difference now is that it is restored to normal after a second or two.
My test scenario is a virtual COM5-COM6 pair. In OCPN I send to COM6 and can use Putty to read COM5 and see what's transferred.
So here can be seen that the route messages are sent but also some "scrap" possible what's in some kind of connection buffer or similar. The "extra" output is interrupted at the  same time as the Message "Route Transmitted" is shown. Then the output to COM6 is again filtered according to the connection settings.
(The same can of course also be see by the Data Monitor. I tried to make a DM-loging but I can't say it's very usable for  this kind of investigation. See attached csv-file. [monitor.csv](https://github.com/user-attachments/files/20053886/monitor.csv))

So -- I'd say this issue is solved although some "scrap" is also sent to the "GPS" in addition to the route sentences. But I suppose the "GPS" can handle this and sort out what's not relevant?
If nothing else comes up I'll close this issue tomorrow.

```
Serial COM5 traffic read by Putty:

$ECWPL,5744.355,N,01130.745,E,001*6F
$ECWPL,5744.355,N,01130.745,E,001*6F
$ECWPL,5743.451,N,01131.467,E,002*6A
$ECWPL,5743.451,N,01131.467,E,002*6A
$ECWPL,5742.572,N,01132.070,E,003*6B
$ECWPL,5742.572,N,01132.070,E,003*6B
$ECRTE,2,1,c,1,001,002*17
!AIVDM,1,1,,A,13uObR0OhkPm9RtPvR09sow<0@G2,0*5F
!AIVDM,1,1,,B,13u4BP0000PngL6Q1AH8LCE:08G6,0*0F
!AIVDM,1,1,,B,13u><8g0000mJ:PQ10LIOoW@0L3>,0*51
!AIVDM,1,1,,A,13tfHj0P00Pm<6dQ3c?>4?w>28GA,0*69
!AIVDM,1,1,,A,1772MP0wQm0nHGTPfVh6Em3>0@GB,0*28
!AIVDM,1,1,,B,13u=gF02000m4V0Q7pW8lDc>0HGC,0*33
!AIVDM,1,1,,A,8@2R5Ph0GhENJAb8wliS2QT?1r06EuOwgwl?wnSwe7wvlOwwsAwwnSGmwvwt,0*3B
!AIVDM,1,1,,B,13u7El0000Pnbg6Q1630O`9>0L29,0*4B
!AIVDM,1,1,,B,33u?er08Q>Pn01BPuuR:W`WF00U@,0*69
!AIVDM,1,1,,A,13Q;;JPvi`Pmg>0PmJquK:m<00RH,0*0D
!AIVDM,1,1,,B,33u?DL3P00PnHDvQ5G2v4?w@RC4S,0*6D
!AIVDM,1,1,,B,33uwo003A3PlKa0PoFs`pG3>20q0,0*3C
!AIVDM,1,1,,B,H3uAaiP5E8u8400000000000003,2*02
!AIVDM,1,1,,A,1815VbP1Af0jWv@Q07bE;l=B04hp,0*27
!AIVDM,1,1,,B,13MC2UG0000hR?fQ1k:Tj:7@00S=,0*46
!AIVDM,1,1,,A,13uB3eP0000nF4PQ0HS01Bw@0<2c,0*0A
!AIVDM,1,1,,A,33u=L?P015Pn<oVPwat<Dqrl00v0,0*21
!AIVDM,1,1,,A,33u<pK001VPm7KHPk=c:`pS>0000,0*18
!AIVDM,1,1,,A,33uGQtPOj8PlcOtPrR4p@Vc@00Si,0*23
!AIVDM,1,1,,B,H3uMcq4TD816;jBC:4I0000h6040,0*12
!AIVDM,1,1,,B,13@ouT0Okf0n4ApPv;:WfF=@00S?,0*41
!AIVDM,1,1,,B,13u=j<gP000mHIbQ1B8P0?w>08Gq,0*02
!AIVDM,1,1,,A,33uF31PPAEPmvaNPuQD2CAa<0P00,0*4A
!AIVDM,1,1,,A,33u=dR0P@2PnfmNQ1@ajeRQ80000,0*44
!AIVDM,1,1,,A,13ul1<00000mILFQ05uofU3@08H7,0*57
!AIVDM,1,1,,B,13u?eoP000Png3HQ1AtJE5k@08H8,0*19
!AIVDM,1,1,,A,33u><=g01oPmO1FQ1:5r9`;D00th,0*47
!AIVDM,1,1,,B,13uOjMP001PnfjTQ1A1@931@04hp,0*0A
!AIVDM,1,1,,B,13uO0iP01>Pn2d@Q0:TB;ik@08H?,0*48
!AIVDM,1,1,,B,13v02f0000PnDWdQ0aNeWV;D08HB,0*2B
!AIVDM,1,1,,B,33A0In001CPnIt0PlmsVBDu@0483,0*7A
!AIVDM,1,1,,A,13thdV002m0i9LBQ5sqs18k>0D36,0*03
!AIVDM,1,1,,B,13uObR000lPm9QHPvQvItWu@00S2,0*53
```

bdbcat, 2025-05-06:
Just to be sure of the setup:
Is the output port in active use before sending the route?  Or is this a "new" connection just for the route output?

Hakansv, 2025-05-10:
Non-relevant excess of messages sent is unlikely to be a problem

closed 2025-05-10 by Hakansv