OpenCPN/OpenCPN #4458: Wrong text color in safe mode dialog

nohal, 2025-04-08:
[image]

The text color in the safe restart dialog is black even with dark system theme. That is wrong, text should be light.

nocodehummel, 2025-04-09:
@nohal thanks, I will have a look.

nocodehummel, 2025-04-10:
How to reproduce? On Windows in dark-mode the old and new dialog show the same.

**Screen-shot old version**
[image]

**Screen-shot new version**
[image]

leamas, 2025-04-10:
Actually the same for me on Linux.

@nohal  Is this on Linux for you?

nocodehummel, 2025-04-10:
> Actually the same for me on Linux.

Same as the results on Windows?

leamas, 2025-04-10:
same as windows. There is more than that lurking around, but we first need Pavel's input.

leamas, 2025-04-10:
But in the meantime, you need at least a Linux box so you can test also Linux. Some things **are** platform dependent, and Windows is in many ways an oddity  in the platform landscape. More extensive gui work like this needs to be tested on more than one platform.

The easiest is probably that you install VirtualBox on your windows host and a Ubuntu or Debian guest.

nocodehummel, 2025-04-10:
I do have Linux boxes, but headless. Can have a look to add a desktop.

nohal, 2025-04-10:
Yes, up to date Fedora (normal Gnome version) with dark theme. BTW also the "not for navigation" dialog at first startup has the same problem.

nocodehummel, 2025-04-10:
@nohal could you post a screenshot from the route delete confirmation? Cause that's a AlertDialog with a normal message.

nohal, 2025-04-10:
There the text color is correct both on Linux and macOS:
[image]

nocodehummel, 2025-04-10:
That should narrow it down to the wxHTMLWindow.

nocodehummel, 2025-04-11:
Reproduced in Xfce on Debian 12 with Adwaita-dark. Also buttons do not have the correct color.

[image]

nocodehummel, 2025-04-11:
The alert dialog with a text message instead of HTML content is correct (incl. buttons).

[image]

nocodehummel, 2025-04-11:
The safe restart dialog buttons are also not visible in normal mode in Xfce on Debian 12.

nocodehummel, 2025-04-19:
> same as windows. There is more than that lurking around, but we first need Pavel's input.

@leamas what Linux do you use? X or Wayland?

leamas, 2025-04-20:
Wayland for now. Can test X11 tomorrow.

nocodehummel, 2025-04-20:
> Wayland for now. Can test X11 tomorrow.

It was a question on the wxWidgets forum.

nocodehummel, 2025-04-29:
Missing buttons issue is resolved by #4511.

nocodehummel, 2025-04-29:
@nohal could you confirm the dark mode issue is resolved by #4512?