OpenCPN/OpenCPN #4462: Data minitor; Still no output source

Hakansv, 2025-04-08:
**Describe the bug**
The output messages, filtered or not, have no source

**To Reproduce**
Steps to reproduce the behavior:
1. Open Data monitor
2. Arrange output by for example Navigate to here.
3. In this case all blue output sentences belongs to COM6 or UDP 10115
4. See screen shot.

**Expected behavior**
Show source.

**Screenshots**

[image]

**Desktop (please complete the following information if applicable):**
 - OS: Win10
 - Version [e.g. 22]

**Additional context**
A long time bug. Strange it's still here

bdbcat, 2025-04-09:
This is all out broadcast N0183 messages, as used with autopilot output sentences like RMB and XTE.
Corrected by https://github.com/OpenCPN/OpenCPN/commit/ab4d8cdb0ca159d07f7b408801c2f4e4e684572e

Hakansv, 2025-04-09:
Corrected by ab4d8cdb

closed 2025-04-09 by Hakansv