OpenCPN/OpenCPN #4477: Watchdog Course Alarm is > 180, instead of max 180.

pa2wlt, 2025-04-11:
Watchdog generates incorrect course alarms.

Go to the WD alarm screen
Set any course alarm, with the course to the current course and off course > 10 degrees.
Port or Starboard.
Use GPS.
Condition triggered for 10 seconds
Enable the alarm

WD will then easily generate alarms saying 355 degrees, whereas no alarm should be generated, since the actual of course is just 5 degrees.

**Expected behavior**
Trivial, no alarm when condition is not met. Off course is always between -180 and 180.

**Screenshots**
n/a

**Desktop (please complete the following information if applicable):**
Win 11 on Intel NUC.
OCN 5.10.2-0 +21è5712 (please make that number copy-able in the about screen)
PluginAPI 118.

**Smartphone/Tablet (please complete the following information if applicable):**

**Additional context**
with this not being in place, the complete course alarm is useless. :-(
Might have to do with instrument source selection? Since previously (years ago) the course alarm worked well..

rgleason, 2025-04-11:
Thank you. This plugin needs work and verification. More bug reports such as this one will be a big help.

rgleason, 2025-04-24:
@pa2wlt we need to move this Issue to watchdog_pi  and have done so

https://github.com/rgleason/watchdog_pi/issues/116

Please close this one, it is in the wrong place.  Thanks. (Following your High Pressure pump rebuild)

leamas, 2025-04-28:
Closing this, moved to https://github.com/rgleason/watchdog_pi/issues/116