OpenCPN/OpenCPN #4481: OpenCPN Crashes when panning CM93 across large distance

svMigration, 2025-04-12:
**Describe the bug**
Crash when trying to pan CM93 charts 

**To Reproduce**
See video

https://www.mediafire.com/file_premium/legq9omtytxrwxb/CM93_crash.mp4/file

**Expected behavior**
Shouldn't crash

CRASH REPORTS
[CrashRpt-Log-20250412-210724-{72876aa3-84ce-4a22-8caf-83a2cfb37357}.zip](https://github.com/user-attachments/files/19718974/CrashRpt-Log-20250412-210724-.72876aa3-84ce-4a22-8caf-83a2cfb37357.zip)

[af6a7b81-0bc5-41ba-a144-502f3568e226.zip](https://github.com/user-attachments/files/19718976/af6a7b81-0bc5-41ba-a144-502f3568e226.zip)

soggygeek, 2025-07-27:
I have experienced this too.  For me it seemed to be only on a new/clean install (or maybe when panning across a totally new area?)  Maybe could recreate it by deleting whatever cache OpenCPN creates to render the CM93 chart.  I reported it a long time ago but could never reproduce it on purpose, though I have seen it occur at least twice over the past couple of years.  v5.10 or below on Windows.

svMigration, 2025-09-27:
Hi All,
I just downloaded 5.12.4 and panned all over the world with both mbTiles and CM93. I did not get a crash. Sherry, have you tried?  Dave, did something get changed that would have solved this issue? If so, I'll close this and the others that may be related.
Thanks,
Bruce

svMigration, 2025-11-16:
New computer. New install. Can no longer reproduce, so closing.

closed 2025-11-16 by svMigration