OpenCPN/OpenCPN #4484: Crash on Windows when pressing Enter in waypoint manager filter field

nohal, 2025-04-13:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Waypoint manger
2. Press Enter key while in the filter textbox
3. See application crash

Windows specific, not reproducible for me on Linux or macOS

rgleason, 2025-04-13:
Yes, I get this crash with  DEv version and  5.10.2 as well.  Put cursor in the filter area and press enter and I get a crash.

>	opencpn.exe!RouteManagerDialog::OnTrkMenuSelected(wxCommandEvent & event) Line 1970	C++
 	[External Code]	
 	[Frames below may be incorrect and/or missing, no symbols loaded for wxbase32u_vc14x.dll]	
 	opencpn.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 767	C++
 	[External Code]	

[image]

rgleason, 2025-04-13:
```
'opencpn.exe' (Win32): Loaded 'C:\Windows\SysWOW64\netutils.dll'. Symbol loading disabled by Include/Exclude setting.
Exception thrown at 0x76BFD424 (KernelBase.dll) in opencpn.exe: 0x000006BA: The RPC server is unavailable.
onecore\vm\dv\storage\plan9\rdr\dll\util.cpp(99)\p9np.dll!64BE2186: (caller: 64BDAA17) LogHr(1) tid(3d6c) C0000034     Msg:[NtCreateFile(&device, SYNCHRONIZE, &attributes, &ioStatus, nullptr, FILE_ATTRIBUTE_NORMAL, (FILE_SHARE_READ | FILE_SHARE_WRITE | FILE_SHARE_DELETE), FILE_OPEN, FILE_SYNCHRONOUS_IO_NONALERT, nullptr, 0)] 
The thread 15724 has exited with code 0 (0x0).
Exception thrown at 0x76BFD424 (KernelBase.dll) in opencpn.exe: 0x000006BA: The RPC server is unavailable.
onecore\vm\dv\storage\plan9\rdr\dll\util.cpp(99)\p9np.dll!64BE2186: (caller: 64BDAA17) LogHr(2) tid(470) C0000034     Msg:[NtCreateFile(&device, SYNCHRONIZE, &attributes, &ioStatus, nullptr, FILE_ATTRIBUTE_NORMAL, (FILE_SHARE_READ | FILE_SHARE_WRITE | FILE_SHARE_DELETE), FILE_OPEN, FILE_SYNCHRONOUS_IO_NONALERT, nullptr, 0)] 
The thread 1136 has exited with code 0 (0x0).
Exception thrown: read access violation.
mergeList._Mypair._Myval2.**_Myfirst** was nullptr.

```

================

>	opencpn.exe!RouteManagerDialog::OnTrkMenuSelected(wxCommandEvent & event) Line 1970	C++
 	[External Code]	
 	[Frames below may be incorrect and/or missing, no symbols loaded for wxbase32u_vc14x.dll]	
 	opencpn.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 767	C++
 	[External Code]	

=================

[image]

Hope that helps.

Hakansv, 2025-04-17:
This ugly hack solves the Win problem. If no other prof solution I will PR?
```
@@ -1964,8 +1964,13 @@ void RouteManagerDialog::OnTrkMenuSelected(wxCommandEvent &event) {
         Track *track = (Track *)m_pTrkListCtrl->GetItemData(item);
         mergeList.push_back(track);
       }
 
+      if (!mergeList.size()) {
+        ::wxEndBusyCursor();
+        break;
+      }
+
       std::sort(mergeList.begin(), mergeList.end(), CompareTracks);
 
       targetTrack = mergeList[0];
       lastPoint = targetTrack->GetLastPoint();
```

bdbcat, 2025-04-18:
Patch looks good.  Awaiting PR.

bdbcat, 2025-04-24:
OP:  Ready to close?

closed 2025-04-25 by bdbcat