OpenCPN/OpenCPN #4489: Bug LaunchDefaultBrowser_Plugin OpenCPN API1.14 function, is missing DECL_EXP in the ocpn_plugin file

rgleason, 2025-04-16:
**Describe the bug**

LaunchDefaultBrowser_Plugin is supposed to be an OpenCPN API function since API 1.14, but, probably because it is missing DECL_EXP in the ocpn_plugin file, it seems to not work in Windows builds (but it does on all others).
Well well, I have revert to the older way...

**To Reproduce**
See https://github.com/rgleason/celestial_navigation_pi/issues/142#issuecomment-2809485892

**Expected behavior**
That it work for all OS MacOS and Windows.

**Screenshots**
Right now there are no screenshots. The plugin does not compile for Windows using this code, but it does compile for MacOS, but not sure if it works yet.

**Desktop (please complete the following information if applicable):**
 - OS:Win10/11
 - Opencpn 5.10.2

leamas, 2025-04-22:
It would probably be possible to add the missing DECL_EXP to current ocpn_plugin.h without breaking the ABI. Technically it would for Windows, but since it does not work anyway: who cares.

The more interesting question is why anyone would want to use this function when there is `wxLaunchDefaultBrowser()`  in wxWidgets. Am I missing something?

rgleason, 2025-04-22:
It's ok to have broken stuff lying around waiting to be used?

leamas, 2025-04-22:
As you know, changing the  API is not done lightly. Perhaps the solution here could be to add a comment saying "Use wxLaunchDefaultBrowser() instead"

bdbcat, 2025-04-23:
"The more interesting question is why anyone would want to use this function when there is wxLaunchDefaultBrowser() in wxWidgets. Am I missing something?"

Very old leftover code.  In the early wx2.x days, wxLaunchDefaultBrowser() only reliably worked on Windows.

I can see no reason not to simply add the "DECL_EXP" declaration to ocpn_plugin.h

leamas, 2025-04-23:
OK, seems reasonable to fix then.  

Pushed 9ac9546f1 fixing this. From my side this means that we should close this issue

leamas, 2025-04-23:
@rgleason  Up to you to close this

rgleason, 2025-04-23:
Yes, Thank you Leamas

closed 2025-04-23 by rgleason