OpenCPN/OpenCPN #4499: Safe restart dialog: Rpi: No buttons visible

leamas, 2025-04-23:
**Describe the bug**
After a crash the "Safe Restart"' dialog's buttons are not visible on RPi and look alike devices.

**To Reproduce**
Steps to reproduce the behavior:
1. Rebuild  master at 07702d2b8 on RPI4/raspbian or OrangePi 5/Debian
2. Start opencpn from command line
3. Kill it using for example ctrl-C
4. Restart opencpn
5. See unusable dialog

**Expected behavior**
Buttons should be visible

**Screenshots**
Rpi
[image]

OrangePi:
[image]

**Desktop (please complete the following information if applicable):**
 - Raspbian, gnome

**Smartphone/Tablet (please complete the following information if applicable):**
 - Rpi 4, Orange Pi 5

**Additional context**
Work around: just wait for the timeout.

leamas, 2025-04-23:
@nocodehummel  ^

nocodehummel, 2025-04-26:
@leamas, would it be better to implement this as wxFrame instead of wxDialog as there is no parent? Alternative to change the initiation after that there is a parent?

nocodehummel, 2025-04-26:
A quick solution could be to set a minimum height when there is no parent window.

```
if (!parent) {
  footer->SetMinSize(wxSize(-1, GetCharHeight() * 3));
}
```

leamas, 2025-04-26:
> would it be better to implement this as wxFrame instead of wxDialog as there is no parent? Alternative to change the initiation after that there is a parent?

I have been thinking in similar ways. Not so much about wxFrame/wxDialog as  using RunModal() vs Show().

Making it non-modal will not work without major refactoring of current  big and complicated start up function. If possible we should probably avoid this, especially now when we  are more or less in beta freeze..

To run in later might be an option, but it is complicated -- the dialog basically governs how to start the main window.

In short, this is a mess. Perhaps we could create a temporary top window, run the modal dialog under it and remove everything when done. It would probably be more consistent with wxWidgets ideas of life. Perhaps we could make this top level invisible to avoid user confusion.

Will make a try with you patch.

leamas, 2025-04-27:
@nocodehummel  ^

nocodehummel, 2025-04-29:
Verifying/trying some different solutions are the above patch gives different results on different platforms.

nocodehummel, 2025-04-29:
@leamas the hard-coded button gives the best results across Windows and Linux for what I could verify. Does this work for the Rpi?

leamas, 2025-04-29:
Yes.

leamas, 2025-04-29:
Well... using `4 * GetCharHeight()` is actually better overall on my small machines.

nocodehummel, 2025-04-29:
I first tried `GetCharHeight()` (see below on Win) and later changed to hard-coded for more consistent results between Linux and Windows. After testing scenarios to set the footer height and/or button height the hard coded buttons are the best solution (on my screens). 

[image]

nocodehummel, 2025-04-29:
Could potentially improve the hard-coded size using `GUI::GetScreenSize()`.

leamas, 2025-04-29:
Hard coded dimensions just cannot work on different resolutions and should not be used. No need to test that.

On your windows screenshot above, how many  char heights did you use, so to speak?

nocodehummel, 2025-04-29:
> On your windows screenshot above, how many char heights did you use, so to speak?

3x

closed 2025-04-30 by leamas

leamas, 2025-04-30:
The buttons like like crap on Gnome, too high. However, thay are functional we live with this for now.

commit 7e906cbe (2025-04-30): libs: gui: Refactor platform footer fix

Using GetCharHeight() * 4 is obviously "wrong" and gives too high
buttons on native GTK. However, it is necessary to close #4499.

This needs a better long term solution.