OpenCPN/OpenCPN #4502: WTF @#$$%$#@

TwoCanPlugIn, 2025-04-24:
No need for words, other than to say, I'm sick of regressions and you guys throwing stuff over the fence without even performing the most basic of unit tests.

Platform, Windows.

Settings

[image]

[image]

Output.
```
PS D:\code\powershell> ./nmea-udp-listen.ps1

cmdlet nmea-udp-listen.ps1 at command pipeline position 1
Supply values for the following parameters:
Address: 127.0.0.1
Port: 1234
Listening On:  + 127.0.0.1:1234
A185702.000 0CFF4 1F200 00401FFFFFFF7F
A185703.000 0CFF4 1F200 00401FFFFFFF7F
A185704.000 0CFF4 1F200 00401FFFFFFF7F
A185705.000 0CFF4 1F200 00401FFFFFFF7F
A185706.000 0CFF4 1F200 00401FFFFFFF7F
A185707.000 0CFF4 1F200 00401FFFFFFF7F
A185708.000 0CFF4 1F200 00401FFFFFFF7F
A185709.000 0CFF4 1F200 00401FFFFFFF7F
A185710.000 0CFF4 1F200 00401FFFFFFF7F
A185711.000 0CFF4 1F200 00401FFFFFFF7F
A185712.000 0CFF4 1F200 00401FFFFFFF7F
A185713.000 0CFF4 1F200 00401FFFFFFF7F
A185714.000 0CFF4 1F200 00401FFFFFFF7F
A185715.000 0CFF4 1F200 00401FFFFFFF7F
```
Expected Output (as from v5.10.2)
```
PS D:\code\powershell> ./nmea-udp-listen.ps1

cmdlet nmea-udp-listen.ps1 at command pipeline position 1
Supply values for the following parameters:
Address: 127.0.0.1
Port: 1234
Listening On:  + 127.0.0.1:1234
$ECRMB,A,0.009,L,001,Bells ,3830.779,S,14410.470,E,8.792,220.961,15.590,V,A*1B
$ECAPB,A,A,0.009,L,N,V,V,221.016,T,Bells ,220.961,T,220.961,T*78
$ECXTE,A,A,0.009,L,N*46
$ECRMB,A,0.009,L,001,Bells ,3830.779,S,14410.470,E,8.792,220.961,13.192,V,A*1B
$ECAPB,A,A,0.009,L,N,V,V,221.016,T,Bells ,220.961,T,220.961,T*78
$ECXTE,A,A,0.009,L,N*46
$ECRMB,A,0.009,L,001,Bells ,3830.779,S,14410.470,E,8.792,220.961,9.594,V,A*22
$ECAPB,A,A,0.009,L,N,V,V,221.016,T,Bells ,220.961,T,220.961,T*78
$ECXTE,A,A,0.009,L,N*46
$ECRMB,A,0.002,L,001,Bells ,3830.779,S,14410.470,E,8.784,221.003,10.793,V,A*1F
$ECAPB,A,A,0.002,L,N,V,V,221.016,T,Bells ,221.003,T,221.003,T*73
$ECXTE,A,A,0.002,L,N*4D
$ECRMB,A,0.008,L,001,Bells ,3830.779,S,14410.470,E,8.779,220.965,9.594,V,A*22
$ECAPB,A,A,0.008,L,N,V,V,221.016,T,Bells ,220.965,T,220.965,T*79
$ECXTE,A,A,0.008,L,N*47
```

Hakansv, 2025-04-24:
I don't doubt your findings but it works here for a serial port:

[image]
COM5 and COM6 are connected virtual. (No heading on the stream)
[image]

bdbcat, 2025-04-24:
TwoCan...
Will you please be more explicit with your report?
Which output is O5.10, and which is O5.11?
Not clear to me.
Thanks

TwoCanPlugIn, 2025-04-24:
I thought "output" and "expected output" were self explanatory?

The "output" from 5.11 is Ascii N2K. It should be NMEA 0183.

The "expected output" that I've displayed was generated from 5.10 and is what 5.11 should be generating.

And for some reason the Ascii N2k are PGN 127488, Engine Rapid Update messages ! 

Aside from the incorrect protocol, if it was intended to be an output related to an active waypoint/route, it should be PGN's 129283 & 129284.

bdbcat, 2025-04-24:
Just to be clear, as I can see from the screenshots:  You have no N2K connections active.  Correct?

bdbcat, 2025-04-25:
TwoCan...
Sorry, cannot reproduce anything like this locally.
We could use a little debug help.  You might like to start by showing the results reported by the DataMonitor.
If those results do not match what is actually going out to localhost UDP, then for sure this is a bug, somewhere.

TwoCanPlugIn, 2025-04-25:
Sincere apologies, my mistake.

I was testing some stuff in order to close #4242 and had forgotten I had left in some debug spew.

closed 2025-04-25 by TwoCanPlugIn

Hakansv, 2025-04-25:
Easy done when a lot testing cables on the table. I know how my own head can mess things up .. 👍