OpenCPN/OpenCPN #4503: Problem to build OpenCPN with GCC 15.1.0

filochard, 2025-04-26:
Hi Alec
I talked about problems to build opencpn plugins with GCC15.1.0  because of a missing #include inside opencpn-libs
https://github.com/OpenCPN/opencpn-libs/issues/21

You have commited an addition of #include `<cstdint>` to OpenCPN/include/ocpn-plugin.h that will correct this for the next version of OpenCPN

But
We  have tried to rebuild OpenCPN 5.10.2 with GCC 15.1.0 official release with our autobuild server in a testing repository

Lots of errors cause the abortion of the build of OpenCPN

OpenCPN needs at least globally #include `<cstdint>`

I will attach an extract of the build.log (NB the build aborted at 24%)

leamas, 2025-04-26:
This is just way too early. It's a long time before gcc 15 hits us. Our current focus is to get next version 5.12 out of the doors, -15 will not be used anywhere for this.

filochard, 2025-04-26:
Thanks for your reply : I understand that it is too early for you now.

I am gonna dig to know where to apply patches to build the actual version of OpenCPN in the next (cooking) version of our distribution (Mageia Cauldron)

I have no problem with our current version of Mageia9 (everything works with a rpm installation and I can update the plugins and OpenCPN itself) but I anticipate for the future !

leamas, 2025-04-29:
It's not just early, it's way too early. Looking at this per platform:

  - Windows is built using msvc, not affected.
  - Android and Macos are using clang, not affected.
  - Native Linux builds are based on Debian. gcc-15 will not hit Trixie, to be released later this year. This means that the first actual build using gcc-15 will be based on Debian 14 Forky which is about two years from now.
  - Flatpak builds are based currently on runtime 24.08, which does not use gcc-15. Perhaps we will use a  runtime for ocpn-13 in a year from now to build Flatpak which uses gcc-15.  This would then be the first usage.

All in all this is so far away that this bug is not that meaningful and I tend to think it should be closed.

OTOH, if you have patches for gcc-15 you want to upstream please feel free to make a PR or two. It would certainly not hurt.

nohal, 2025-04-30:
Fixed by https://github.com/OpenCPN/OpenCPN/commit/72e1a46a82391345547ba1f4ab9931a46f0ee05b

closed 2025-04-30 by nohal

filochard, 2025-04-30:
Hi Pavel

I have just seen that you have added #include `<cstdint>` inside 3files in commit https://github.com/OpenCPN/OpenCPN/commit/72e1a46a82391345547ba1f4ab9931a46f0ee05b
that's to say : 
- /gui/include/gui/AISTargetAlertDialog.h
- /libs/SQLiteCpp/include/SQLiteCpp/SQLiteCppExport.h
- /libs/sound/include/SoundFileLoader.h

Alec had already added this #include `<cstdint>` inside
/include/ocpn_plugin.h

When trying to build with GCC 15.1.0 I have found that this #include `<cstdint>` was needed by other files : 
- /libs/shapelib/sbnsearch.c	
- /libs/sqlite/include/sqlite3.h
- /libs/SQLiteCpp/include/SQLiteCpp/Column.h
- /libs/SQLiteCpp/include/SQLiteCpp/Statement.h

I hope this will help for the next version of OpenCPN, if it is built with GCC 15.1.0 
(nevertheless those changes are painless to build with previous versions of GCC they only correct a syntax that was accepted but will no more be accepted)

nohal, 2025-04-30:
Certainly none of the mentioned changes is needed on my  GCC 15.1 system

filochard, 2025-04-30:
They are neede for me to build OpenCPN 5.10.2 upon GCC 15.1
I thought they should be needed for the next version
Nevermind
I prefer to say something useless than missing something possibly useful

filochard, 2025-05-01:
Hi Pavel
Sorry for the noise
Thanks to you !!!
Indeed OpenCPN is ready now for GCC 15.1

I have built a rpm with a clone of the git with  your last commit (https://github.com/OpenCPN/OpenCPN/commit/72e1a46a82391345547ba1f4ab9931a46f0ee05b)
with GCC 15.1 on my system
(I have only rediffed the patches I add to 
/model/src/plugin_loader.cpp 
and /model/src/plugin_paths.cpp  
so that we can use the plugins installed by rpms )

It builds perfectly : the only problem I have encountered is that I needed to add 
**BuildRequires: pkgconfig(shapelib)** in my spec file
Which was not necessary for OpenCPN 5.10.2 when I built it with the previous version of GCC

I only wonder if this rpm may be used as a everyday version

filochard, 2025-05-03:
End of my struggle to build OpenCPN 5.10.2 with GCC 15.1.0 : 
I finally succeed with 3 patches adding
 #include <cstdint>
to 
- /libs/SQLiteCpp/include/SQLiteCpp/Statement.h
- /libs/sound/include/SoundFileLoader.h
- /include/ocpn_plugin.h

That will allow to provide OpenCPN 5.10.2 in the next version of our distribution, waiting to update OpenCPN to 5.12.0

Thanks to all of us