OpenCPN/OpenCPN #4521: OpenCPN 5.11.0 info in data monitor window and connections window wrong and crashes too.

antipole2, 2025-05-03:
**Describe the bug**
See attached screenshot.
NMEA0183 data is being sent on port 60003 OK (shown in blue) and indeed data is being received OK by the connecting device (actually iNavX on an iPhone)

**Issue 1 - time stamp**
I take it that that is a time stamp at the start of the entry in the data monitor window.  The time stamp is 18:30:58  but the time is actually 15:16 Danish time (UTC+2).

**Issue 2 - incorrect status**
The entry in the connections table shows this port status as X when it is clearly operating.

**Issue 3 - crash**
If I untick this connection OCPN crashes.  This is reproducible.  Crash report attached.

**Context**
The connection is being established by my iPhone connecting to OCPN.  So the status needs to be updated when a connection is established and when it disconnects.

[image]

Hakansv, 2025-05-03:
I agree on the time stamp error:
Here, UTC+2 i.e 21:18:xx and the time in DM is 08:47:29

[image]

leamas, 2025-05-03:
Reporting three separate issues in a single report (after first deleting the extremely helpful form) might become the start of a bumpy road.

Let's focus on the crash first. Could you please fill in the form, notably the exact steps 1, 2, 3 , ... leading up to the crash fir this issue?

leamas, 2025-05-04:
Issue 1: #4522

antipole2, 2025-05-04:
**To reproduce**

1. Create or choose an NMEA0183 output stream.  IP address 0.0.0.0.  Port something unique.  I use 60003.
2. Activate the stream.
3. In Data Monitor the output is red as not connected and in Connections status is X.  **Correct**
4. Activate a connection to take the output.  I use iNavX or Aqua Map or TZ iBoat on an iOS device. You could probably use a MFD.  Enter the IP address of the OCPN computer and the port number.
5. The Data Monitor now shows this port's data in blue, confirming is connected.  You can likely also see data in connecting device. **Correct**
6. But in Connections window the status is still X.  **It should have been updated.**
7.  Deactivate the stream. **Crash as per dump above**.

OCPN clearly knows the stream is active, since it is showing it in blue.  But it fails to set the correct status in Connections.  So when deactivating the connection, confusion is likely.

leamas, 2025-05-04:
Right. Another missing part from the form: Which version (build) are you using ?

antipole2, 2025-05-04:
> Right. Another missing part from the form: Which version (build) are you using ?

Apologies - thought I had included that. Tested for the issues today with 5.11.0-universal+cb8e39d7a.

leamas, 2025-05-05:
OK.

I¨m currently busy trying to create a new MacOS VM. This will take some time Apple being Apple.

In the meantime, could you please try to create  a full stacktrace from the crash according to https://opencpn-manuals.github.io/main/opencpn-dev/stacktrace.html#_macos? The dSYM file you need is available together with the build. That build may be gone, though, so you might need to use a more recent build then.

leamas, 2025-05-06:
Testing using current master  at 076edb48a

  - Using Håkan's test feed from signalk.stupan.se:10114 as input to get some data
  - Start a output connection to localhost:3344. Shown as  expected X
  - Start a local tcp sink using netcat `netcat -p 3344 -l localhost`
  - The connection changes to ✓, as expected
  - Deactivate the connection. Nothing bad happens
 

Did a new check on cb8e39d7a i. e., the same commit you have been testing. Same results. Have you any idea what the difference could be about?

Really need that stacktrace...

antipole2, 2025-05-06:
One possible difference is that I am connecting through a router and not direct.

I have tried using the TZ iBoat app instead of iNavX - both on an iPhone and also both on an iPad.
Tonight, it is not always crashing.  Sometimes I can disable the output by unticking and start it again by ticking.  When I tick to start again, the status briefly goes to 'O' before going back to 'X' but never to '✔️'.
(After resuming the output to iNavX, I have to re-enable the connection in the app.  With TZ iBoat,it picksup again automatically (mostly).)

There is no problem with the remote end disconnecting and then resumming.  The colour of the message in Data Monitor changes correctly.  Additionally, if I have a JavaScript running that pushes NMEA0183 to the port, the push fails if the port is disabled and starts to succeed again when reconnected.  So OCPN has got it right except in the connections panel. (By the way, I have checked and the issue exists even with JavaSCipt_pi disabled, so that is not the problem.)

I am on the boat with no WiFi - just a mobile signal subject to roaming limitations.  Getting ready for launch.  So not yet tried to follow your instructions for the full trace.

My hunch... the status showing as 'X' might be because of going through a router?  And if the status is wrong, maybe when unticking, it makes a bad decision about disabling a connection it already think is disabled.

Extra crash report from tonight:

[image]

leamas, 2025-05-07:
>  My hunch... the status showing as 'X' might be because of going through a router?

Doubt that. From the Mac's perspective there is just a TCP connection or not (unless some low level communication stack error which is really far fetched).

Might be about your equipment vs using a "clean" sink. One factoid which perhaps could be helpful is if you did the same check as I i. e., used netcat or something similar instead of your "real" equipment. `netcat ` is not available by default, `brew install netcat` makes the trick.

But what we really need the exact point for the crash(es).

antipole2, 2025-05-09:
Have installed netcat and trying to follow this by creating a sink for the data other than my iPad app.

> Start a output connection to localhost:3344. Shown as expected X
> Start a local tcp sink using netcat netcat -p 3344 -l localhost
> The connection changes to ✓, as expected
> Deactivate the connection. Nothing bad happens

But `-l` says to listen for connects.  In my situation, the remote app on iPhone/iPad makes the connection, so

> Tony@Laing ~ % netcat -p 60002 --s=localhost
> netcat: missing hostname argument
> Try `netcat --help' for more information.
> Tony@Laing ~ % netcat -p 60002 --s=192.168.1.137
> netcat: missing hostname argument

If using localhost, presumably it does not go via router, as in my problematic situation. So I tried giving the ip address.  Not sure what to try.

leamas, 2025-05-09:
Which makes my testing void. What is your setup? How is it configured, exactly?

antipole2, 2025-05-09:
So here is my setup - see the diagram below.

The W2K is a NMEA2000<->NMEA0183 WiFI converter.  Although it is capable of creating its own WiFi network, it is not a router.  So I have a separate router and the W2K connects in client mode.

OCPN receives data on port 60001.
Data out to the network is sent on port 60005.
Ignore the second port 60001 - not in use.

The various apps on iOS devices all work the same way.  In the apps you enter the TCP/IP address of the MacBook and a port 60002 or 60003 or 60004 and click on connect.  Output from OCPN starts flowing to the app and the relevant line in the data monitor turns blue.
If in the app you turn off the data flow, it stops and the entries in the data monitor turn red. You can start and stop data like this as often as you wish.  Attempts to write out on these ports in OCPN succeed or fail correctly according to whether the connection is established.

But, as I have reported, the status in the connections panel is always ‘X’.
If I disable one of these connections it often crashes, as reported.  If it has not crashed and I re-enable the connection, the status briefly gos to ‘O’  before returning to ‘X’ even though the data is flowing.

I hope that is clear.

[image]

antipole2, 2025-05-09:
> Which makes my testing void. What is your setup? How is it configured, exactly?

Please see above - my reply has appended to my previous rather than your question.

antipole2, 2025-05-12:
I have found that, when a connection to an iOS app is active, if I write to it explicitly, the data gets through.
But OCPN does not send NMEA0183 data routinely .  Presumably, this is because the port’s status is ‘X’.

This confirms my thesis that when these ports connect, OCPN knows this enough to be able to send data when it is written to,  but is failing to register the port as active.  I think all issues like the crash follow from this.

bdbcat, 2025-05-17:
I have tested this scenario as follows:
1.  linux "host", equivalent to Macbook in the diagram.  Input port valid N0183 data.  Output TCP port as described.
2. Windows "client", equivalent to iOS apps in the diagram.  Connected by TCP port on linux "host".
3. All these connections are on the same local network segment, through a router.  Client is on WiFi.

Results:
1.  Data flows as expected.
2. I confirm that the "X" on the server side (e.g. Macbook side) does not shift to "checkmark" when data is flowing.  THIS IS A LEGITIMATE BUG, nothing to do with MacOS or iOS.
3. Freely enable and disable connections from either side.  No crashes observed.

I will investigate the "X->checkmark" bug first, then do some tests on Mac.

bdbcat, 2025-05-17:
Tony, Just to be clear:
You are NOT reporting a failure to connect.
Rather, you are reporting:
1.  "X" mark is in-correct, because data is obviously flowing.
2. Sometimes the "server" crashes when the connection is disabled at the server side.

Do I have this right?

bdbcat, 2025-05-17:
Corrected the "checkmark" error with https://github.com/OpenCPN/OpenCPN/commit/73df5f0107ec545a76a91bf46a294793aa50e14d

antipole2, 2025-05-17:
Yes, that is correct.Am on remote anchorage with only limited mobile, so will test your patch when in port and have WiFi.Spent last evening recovering tracks lost because of having run v5.11.   (Switching between versions had rotated the xml files too many times 😟.) So reluctant to launch v5.11 again until I am in a more secure situation.

antipole2, 2025-05-18:
Using latest MacOS build 5.11.0-universal+17eadb653 all seems good.  The status settles on the correct value when enabled/disabled with tick mark or at remote end.
No crashes seen so far.

Will close this in 48 hrs.

Thanks.

antipole2, 2025-05-24:
All good thank you.

closed 2025-05-24 by antipole2