OpenCPN/OpenCPN #4522: Data Monitor: weird timestamps

leamas, 2025-05-04:
**Describe the bug**
As descrbed in #4521, issue 1.

**To Reproduce**
Steps to reproduce the behavior:
1. Start opencpn
2. Start the Data Monitor, for example using the E hotkey
3. See timestamps which does not reflect current time.

**Expected behavior**
This is indeed the question, see below

**Screenshots**
See #4521 

**Additional context**
Currently the time stamps are off from current time like 1-2 hours. This is clearly a bug.

The background is that std::chrono defines two basic clocks: system_clock and steady_clock.

`system_clock` is the wall time, basically used to answer the question "What time is it?". It is subject to changes like when user sets date/time, DST on/off, NTP time drift and what not.

`steady_clock` is guaranteed to be, well, steady. It is designed to answer the question "How long is the interval between t1 and t2"? It is not affected when user sets date/time, etc.

With this background:

1. Current code uses the steady clock to display wall time. This is obviously broken.
2. One working way is to use the steady clock, but just display a time stamp relative to for example when opencpn was started.
3. Another is to use the system clock and display the wall time.

Looking at these alternatives I tend to think that 2. is the correct solution. After all, what is the purpose of displaying wall time in the monitor window. 

User can indeed pause the output to check what time it is. However, there are simpler ways to do that. Besides this, I cannot see any use of the wall time in the monitor window. The reason it is there is IMHO just that it always has been. This does not means it's correct.

Using relative time stamps we avoid the time zones mess, and also get something which visiually is easier to inspect -- the typically leading zeroes are easier to disregard than in the wall clock case.

TL;DR: The correct solution here is to use the steady clock and relative time stamps.

bcn2, 2025-05-04:
From a user's point of view there are two valid options for time stamps:
- LCL, the system clock
- UTC
The time stamp shall indicate which is shown

leamas, 2025-05-04:
NO need, I have worked around this mess using relative time stamps. See #4524

bdbcat, 2025-05-04:
I am tending to think that UTC, properly annotated, is the correct solution.
For one reason, the logfile, if created, will contain unique timestamps, so are more durable.
Another:  We use wall clock time in OCPN master logfile, though we night consider shifting that to UTC also.

Do we not have a new config options that selects between LCL and UTC?  Why not honor that?

Other input?

leamas, 2025-05-04:
> For one reason, the logfile, if created, will contain unique timestamps, so are more durable.
Another: We use wall clock time in OCPN master logfile, though we night consider shifting that to UTC also.

> For analysis or replay (case of VDR) the timestamp of LCL or UTC is important. What happened when.

This PR is only about timestamps on screen. These are not used for later analysis or replay.  The logfile time stamps is a another story.

leamas, 2025-05-04:
> For analysis or replay (case of VDR) the timestamp of LCL or UTC is important. What happened when.

For vdr the format is as defined by  https://opencpn-manuals.github.io/main/vdr/log_format.html (using the raw "milliseonds since 1/1 1970" format"

bdbcat, 2025-05-04:
I would suggest that the logfile generated by data monitor should match the screen, and that both are reported in UTC.
When using this type of log, such as the standard OCPN log, I personally ignore visually the first few characters of the line when reading.  But it seems not right to discard this irreplaceable information from what might be a permanent retained log. I'm also thinking about "syslog", and friends.

I feel pretty strongly about this...

leamas, 2025-05-04:
> I would suggest that the logfile generated by data monitor should match the screen, and that both are reported in UTC.

So this is about two things

1. The log file and the screen should use the same format
2. The log file should use UTC.

I just cannot see the argument for 1. Why? 

As for 2.,  the log file time stamps is a different thing. We could of course discuss also that in this context, but the bug is actually  about the time stamps on screen as is #4524.

bdbcat, 2025-05-04:
1.  Why?  Coherence.  I, a simple user, would expect that the recorded logfile and what I see on screen should match timestamps.  Why would one want them to be different, intentionally?
When I come to post a couple of logfiles on CF, I'll need to type:  "The first file started at 3:00.  The second started at 3:30 and here is a screenshot taken at 3:40".
All of them need to use wall clock-time, in UTC.

leamas, 2025-05-04:
There is no such thing as the "logfile format",  there is currently three of them.  They have different purposes.

 1.  Is governed by spec above, it's purpose is that t should be possible to replay by the VDR plugin.  As a nice side effect it is possible to import it into spreadsheets like Excel.
 2. Is designed to be more easily machine parsed, the format in 1. is a bit convoluted. Targeting traditional command line tools for post processing.
 3. Is  aimed for humans.

Your remarks are AFAICT only relevant for the last format. As it is now, it also uses relative timings and would thus be consistent with the view on screen. The start  time is recorded in the beginning of log.

leamas, 2025-05-04:
> When I come to post a couple of logfiles on CF, I'll need to type: "The first file started at 3:00. The second started at 3:30 and here is a screenshot taken at 3:40".

Again: Why? I cannot see that the time when  a log was recorded  for example when tracking an error is that important.  And there is no need for users to print the date, all log formats has complete information. Either as absolute UTC time (the vdr format) or as the start time + relative timings.

bdbcat, 2025-05-04:
I am not a bot.  I am a human.  So I take doorway #3.
I like to see the logfiles, on-screen and in file, with full date/time information, preferably in UTC.
Am I the only one so afflicted?  I don't think so.

leamas, 2025-05-04:
If datetime should be used UTC is the reasonable choice., agreed.

The screen is not a logfile. You cannot scroll backwards, cannot save it, etc. It's just a window on a computer where we definitely know what time it is anyway.  My gut feeling that is that this is more about the cost of change than anything else. We are simply used to the old format.

But that does not mean it is the right one.  I actually think that the relative timing between different messages is the focus when reading those logs rather than the absolute UTC time. And this is why I keep pushing my case.

But we are approaching a point when we (you) have to make a decision. Let's do so after a night's sleep.

Hakansv, 2025-05-05:
> I like to see the logfiles, on-screen and in file, with full date/time information, preferably in UTC.
> Am I the only one so afflicted? I don't think so.

Really not. That is what I have tried to express in the past.

bdbcat, 2025-05-07:
Night's sleep did not change my opinion.
Please switch to wall clock (UTC) time reporting.
Thanks.

leamas, 2025-05-07:
Nor did it change my mind. But it's time to finish this up.

I don't want to change anything, merging things which I perceive as  buggy and inconsistent goes against basic instincts. So, I leave you with two PRs:  #4530 and #4524. Please make the decision to merge one of these and close the other. 

Hopefully I won't  have an opportunity to say "What did I say?"  in the near future ;) After a while, this is all forgotten

PR #4530 (2025-05-07): data monitor: Use system_clock - #4522
Use system_clock to resolve #4522. Marks log lines with current time which resolves the reported problem. However, the price is quite high:

  - Using system_clock means per definition that the relative time between messages is inconsistent/buggy.
  - Logs spanning over more than 24 hours are also inconsistent.

Closes: #4522
commit 9e4123b6 (2025-05-07): data monitor: Use system_clock - #4522

Use system_clock to resolve #4522. Marks log lines with current time
which resolves the reported problem. However, the price is quite high:

  - Using system_clock means per definition that the relative time
    between messages is inconsistent/buggy.
  - Logs spanning over more than 24 hours are also inconsistent.
closed 2025-05-07 by bdbcat