OpenCPN/OpenCPN #4523: 5.11.0 - Data Monitor Window not eresponsive after returning from the internal menu

bcn2, 2025-05-04:
**Describe the bug**
Data Monitor Window does not react after returning from the internal menu

**To Reproduce**
Steps to reproduce the behavior:
1. Open Data Monitor
2. Click on "Menu" down right - the three bars
3. Scroll down to "logging", open
4. Do nothing, just "Close"
5. Window is non-responsive, data flow visible. Keeps focus, so the only way to stop is killing OCPN

**Desktop (please complete the following information if applicable):**
 - OS: macOS
 - 15.4.1

[image]

[image]

[image]

leamas, 2025-05-04:
Hm... Linux and Windows are not affected by this. What build have you been using?

bcn2, 2025-05-04:
Dave's navobj.db commit from yesterday.

[image]

leamas, 2025-05-04:
OK, I can reproduce it. Seems to be MacOS specific which makes me feel nuncomfortable.

closed 2025-05-06 by leamas

leamas, 2025-05-06:
Turns out that this was a nasty bug, it should not work on any platform. But only MacOS rightfully misbehaved.

Thanks for reporting!

commit 076edb48 (2025-05-06): data monitor: Fix window references and Log Setup dialog close - #4523

Closes: #4523