OpenCPN/OpenCPN #4528: Weather Routing plugin GUI broken on Windows 11 – OpenCPN 5.10.2 + WR 1.15.26

arancioni, 2025-05-04:
I'm reporting a critical issue with the Weather Routing plugin's interface under Windows 11.

OpenCPN version: 5.10.2-0+12e5712

Plugin API: 118

Weather Routing plugin version: 1.15.26

OS: Windows 11 Pro (64-bit)

Installation: Official Plugin Manager

Main issues:

The plugin’s main window opens as a flat, collapsed rectangle (see attached photo), with no visible interactive elements.

In other cases, the routing interface loads but:

Dialog boxes are truncated or unreadable

Input fields are misaligned

Buttons are hidden or clipped, including critical controls like “Cancel”

The layout is completely broken

This makes the plugin non-functional.

Confirmed conditions:

Display resolution is standard (non-4K)

Default Windows scaling

No DPI overrides

System freshly rebooted

This appears to be a rendering/layout error linked to wxWidgets and plugin UI compatibility under Windows 11.

Attached:
Screenshots showing:

Collapsed main Weather Routing window

Truncated dialog for route position input

Please advise if a fix or workaround is available. I remain available for further technical details if needed.

Thankd

rgleason, 2025-05-04:
I don't believe this should be classified as a bug. WR 1.15.26 installs properly and and runs nicely showing the new WR Table, on my Windows 11 machine,  as it worked when I tested it before compiling.

[image]

Please uninstall the plugin completely and try reinstallation, using the default setup.  Please advise if this helps.

arancioni, 2025-05-04:
Hi Rick,

We've already installed and uninstalled the plugin multiple times, including full resets and default configurations.
The issue persists identically.

In addition, the WR interface is missing critical functionality:
the panel to create or edit polars does not appear at all, making the plugin unusable for routing setup.

Please review this as a serious regression on Windows 11 – this is not a user misconfiguration.

Thanks,

nohal, 2025-05-04:
I wonder why do you think this is an OpenCPN core application issue at all. Probably belongs to https://github.com/rgleason/weather_routing_pi
Maybe at least attaching the mentioned screenshots could shed some light on this.

rgleason, 2025-05-04:
Ok, I am trying to figure out what we can do to isolate the problem on your machine.
First of all we need a zipped file of your opencpn.log  uploaded here, use the meny "?" to find it and copy it.  If you have crashed, any zipped crash reports uploaded here would be useful.  C:\ProgramData\opencpn\CrashReports

### Clean Install

1. Do you have a clean install of OpenCPN 5.10.2 with no plugins enabled?
2. Does OpenCPN work as expected when all plugins are disabled?
3. Close Opencpn.
4. Have you removed all vestiges of WR including removing these directories?- Remove the WR plugin from the plugin directory and do not reinstall.
- Close OpenCPN
- C:\Users\[user]AppData\Local\opencpn\plugins\weather_routing_pi
- C:\ProgramData\opencpn\plugins\weather_routing   (you could rename it)
- C:\Program Files (x86)\OpenCPN\plugins  (Check, you shouldn't find any Weather Routing in here.)
- C:\ProgramData\opencpn\opencpn.ini  rename to opencpn.ini.old  (you can rename back later)
- C:\ProgramData\opencpn\plugins\cache\tarballs   Remove any weather_routing tarballs.

5. Now restart Opencpn, update the Plugin Master Catalog and install and enable WR using the standard installation.
6. Run a test.
7. Advise

rgleason, 2025-05-04:
Pavel is right, this should not be listed here, but under Weather_routing, but please respond in the next 24 hours or I will close this and you will need to report it under weather_routing. 
Thanks.

arancioni, 2025-05-04:
Hi Rick, hi Pavel,

We have already installed and uninstalled the WR plugin multiple times, using clean configurations and defaults.
The GUI problems persist identically, and the plugin is not usable on our Windows 11 system.

More importantly, the interface is missing key functionality:
there is no access to the Polar creation panel, which is essential for routing.

Screenshots were not attached initially due to email limitations, but they clearly show:

Collapsed or malformed plugin windows

Truncated dialogs and missing buttons

Absence of UI components related to polar management

This is not a usage error or a misunderstanding.
Please consider escalating or redirecting this issue to the plugin's repository if needed.

Thanks,
Adriano Bevilacqua

rgleason, 2025-05-06:
This needs to be closed by @bdbcat  or @nohal  or @leamas   I can't

leamas, 2025-05-06:
done

closed 2025-05-06 by leamas