OpenCPN/OpenCPN #4534: Add flag (Nation) in AIS target list

ironpowertga, 2025-05-08:
Problem:
It can be nice to have the possibility to view and order ship by flag

Resolution:
Add flag column in ais target list

RooieDirk, 2025-05-09:
We can already sort the list by mmsi no. The first 3 digits of the mmsi do give the country of register. Sorting on mmsi is also sorting on country.

ironpowertga, 2025-05-09:
Yes l’m agree, but it's easier to have directly the country show in a specific column

rgleason, 2025-05-09:
The would need program a lookup table. Can you provide it?

ironpowertga, 2025-05-09:
This is the actual view ais target list with my commit
[image]

rgleason, 2025-05-09:
Where is the PR?  The link appears to be circular.

ironpowertga, 2025-05-24:
I can now close the issue. Thank you for the merge.