OpenCPN/OpenCPN #4537: Development Manual - Index is screwed up can't find anything

rgleason, 2025-05-09:
**Describe the bug**
[See for exampe](https://opencpn-manuals.github.io/main/ocpn-dev-manual/0.1/building-on-armhf-linux-armbian-orange-pi.html)

I do not know why this keeps happening, but it is a real problem for new programmers.

nohal, 2025-05-09:
[image]

What is "screwed up"?

leamas, 2025-05-09:
Isn't the real problem that this old junk isn't removed?

leamas, 2025-05-09:
@rgleason : I miss the form data such as  :
  - Short description of the actual problem
  -  The buggy behaviour,
  -  Expected behaviour.

rgleason, 2025-05-09:
Next to the House button there is an index button.

[image]

When I click on the index button  I get this.

[image]

rgleason, 2025-05-09:
If clicking on the index symbol next to the Home button, goes to the List of Manuals, you have just created a circular event without the ability to open the relevant index.

nohal, 2025-05-10:
So all you are complaining is that when the browser window width is small enough that the index is collapsed and hidden (As you can see on my screenshot, there is no such button at all) the button that appears instead of it leads to the index page, right?

rgleason, 2025-05-10:
In the first screenshot in [this comment ](https://github.com/OpenCPN/OpenCPN/issues/4537#issuecomment-2867953483)  the index button, when clicked on by mouse,

Opens the second screenshot in [this comment](https://github.com/OpenCPN/OpenCPN/issues/4537#issuecomment-2867953483)  which is NOT the right index page!

This then leads to a circular path without ever seeing the correct index

nohal, 2025-05-10:
So your answer to my question is YES, right?

rgleason, 2025-05-10:
Ah ha. Thank you.

As you can see on my screenshot, there is no such button at all) the button that appears instead of it leads to the index page, right?

nohal, 2025-05-10:
Now I'm super confused. Is that button that appears when the window is not wide enough broken or not, according to you? According to me it is broken, and if you agree, I do not understand why you closed this issue? All I wanted to find out, what, in the waterfall of incomprehensible noise, is the actual thing you complain about, not make you close a valid issue.