OpenCPN/OpenCPN #4538: Impossible to import tracks saved in a .gpx file (from OpenCPN)

Ptizef, 2025-05-10:
HI,
Windows 11, compile from master yesterday

Tried to test the new navobj file system.
All things ok at start but :

1) Impossible to import tracks.  For routes, it works.

2) I deleted all my tracks. The .xml file's size was about 1000 ko and become 1 Ko, that seems correct as I had at that time no others objects, But the .db file stayed with his initial size of about 900 Ko.
Then I import some routes and the .xml file become 182 Ko, the .db file still has his initial size!
How to know what is really in this db and how to recover data from it in case of the .xml file system fails (maybe in case of a crash badly managed)
JP

Hakansv, 2025-05-10:
Confirmed on Win10 by a simple test:

1. I named an old track so it's easily recognized. (xxxx-mm-dd test)
2. The test track was, by "route and mark manager", exported as .gpx
3. I deleted the test rack.
4. Restarted O for no particular reason, but anyhow.
5. I imported the test track.
6. The test track is not visible on the track list. Still not after a O reboot, so apparently not saved in the track database.

LeaseOnLife, 2025-05-11:
Tracks that have been "send to peer" are also lost on "peer" after restarting O.

bdbcat, 2025-05-11:
Confirmed on both.  Please standby.

antipole2, 2025-05-16:
Problem on MacOS too.

I have to sail using v5.10 because of issue #4521 .  Having viewed my tracks in v5.11, none of my tracks are accessible in v5.10 - oh dear!

So I exported a couple I needed from v5.11 into GPX file and imported into v5.10.
Now 5.11 (5.11.0-universal+38e45e883) crashes on start up - crash report follows

[image]

bdbcat, 2025-05-18:
Import of navobjs corrected in commit c8c46e5641703b43db70c47e5c1ef2e8c95e9912
Please confirm and close.

antipole2, 2025-05-18:
Dave... Question:

I gather you are now using a database to store routes, tracks etc. rather than the navobj.xml files.  

Does this mean a user cannot revert from v5.11+ back to v5.10.2 without losing changes made using v5.11+ ?

LeaseOnLife, 2025-05-18:
my reported issue of tracks "send to peer" being lost seems to be fixed. That said, it took a very long tme for my navobj.xml  (lots of marks, lots of tracks) to be converted to navobj.db, I think the conversion needs a progress dialog.

Hakansv, 2025-05-18:
My test example to export - delete - import routes and tracks is also working now.

Hakansv, 2025-05-18:
On more comment:
I can agree on LOL's comment about some kind of user info about the ongoing navobj copying. On my onboard WinPC with some tracks and routes it took some time and a user might think the installation failed.
A progress bar may be to overdo it as a one-off event but a message box may do: "Rearranging routes and tracks. Please wait" ??

bdbcat, 2025-05-18:
LeaseOnLife/Hakan.
Will you please post or send me your "navobj.import_backup" file?  I'll test the import on this large file.  There are some optimizations in the import process that can be done...

bdbcat, 2025-05-18:
Tony...
If you meed to revert, you may export from O5.12, and then import to 5.10.

LeaseOnLife, 2025-05-18:
20MB email sent

bdbcat, 2025-05-19:
Added Progress Dialog, in master now.
Further thoughts?

Hakansv, 2025-05-19:
Nice progress bar!
Route WP names now shown if enabled in origin, else not. Very good.

Ptizef, 2025-05-19:
Hi all
I tested the last master.
Seems to work well.
As a test, I imported all my tracks for a total of 38Mo. It was reasonably long.
Deleting all in one time was the same, except the first time, O closed. Nothing was lost. I couldn't reproduce.
As another test, I imported all my routes for a total of 6Mo. The same. The only problem when setting / unsetting "Show all Routes" it takes about 2 Mn. If I click again in the dialog during the process, then O hangs.
The main problem I have:
I can't cancel the waypoints by the route manager. I delete all or one by one, they disappear, but when closing and reopening O, they are still here.
I can cancel them by the menu.
JP

LeaseOnLife, 2025-05-19:
xml2db workd fine on a PI4 4GB with some 3000 marks, many tracks. will be running it now as the daily driver.

Ptizef, 2025-05-19:
I would add that my tests were on Windows 11.
And I have a question.
Why the navobj.db file is still at 16 Mo without any objects except 2 MOBs I can't cancel at the moment due to the problem described above
JP

LeaseOnLife, 2025-05-19:
> Hi all I tested the last master. Seems to work well. As a test, I imported all my tracks for a total of 38Mo. It was reasonably long. Deleting all in one time was the same, except the first time, O [...]

There certainly is a problem deleting things. I tried to remove a bunch of marks, and after restart, O has them all back again.

bdbcat, 2025-05-19:
Ptizef...
Please send me, or post, your 16 Mo navobj.xml file for test.
Thanks

bdbcat, 2025-05-19:
RouteManagerDialog Mark delete button now corrected in github master, building now.
Thanks for the catch.

Ptizef, 2025-05-19:
Here my .db
I hope it'll go this time
Regards
JP

Ptizef, 2025-05-19:
1 - The delete marks works now. Thanks
- I send you my .db, but it didn't go due to size limits 
I hope this works

[navobj.7z.pdf](https://github.com/user-attachments/files/20300695/navobj.7z.pdf)

As a test, I removed this .db file. It was re-created at 64 Ko
I imported routes and tracks. 
I went up to 7,3 Mo
I deleted all. It stayed at 7,3 Mo
I imported again routes and tracks (some were the same as previously)
I went up to 7,5 Mo
I deleted all again.
It stays at this size
Can I conclude that objects are added to the file, but never removed?
JP

bdbcat, 2025-05-20:
Ptizef...
This is normal behavior of the database manager that we use, called SQLite.  It performs like many other database systems in this regard.
1.  On adding data the database grows, obviously.
2.  On deleting data, the table space in the database file previously occupied by deleted data is marked as "empty, available for re-use".  But the database file itself is not automatically compacted to remove that empty space.
3.  Next time data is added to the database, it will very likely use that "empty,available" table space for storage.
4.  So as a life cycle, the navobj.db file will grow to the largest active size that it has ever been, and stay at that size.

As a practical matter, there is no cause for the OpenCPN user to be concerned about the size of the navobj.db file at any point in time.  It will generally be somewhat smaller than the equivalent navobj.xml file as imported.

There are methods which may be used to semi-automatically compact the database file to reduce its active size.  All of these methods have drawbacks, the most significant of which is the potential for data loss if used while OpenCPN is running.  There are also application performance tradeoffs to consider.

Finally, there are third-party applications available to compact any SQLite database, if desired, without OCPN involvement.  This might be useful if one wishes to, say, distribute a pre-built navobj.db to a sailing club, for instance.

Otherwise, no practical need to monitor or worry about the navobj.db file size.

bdbcat, 2025-05-20:
Also, I loaded your large navobj.db database file.  It is essentially empty, as expected by your add/delete cycles.
On compaction, it reduces to the expected 64Ko size.
So I think we can say that there are no "orphan" objects accidentally created in the database.

bdbcat, 2025-05-20:
For those interested in the technical details, here is great article:
https://blogs.gnome.org/jnelson/2015/01/06/sqlite-vacuum-and-auto_vacuum/

Ptizef, 2025-05-20:
Many thanks for your answers and this very interesting article. It will help me to better understand how works à db...Anyway, I think that I can close this issue now.Thanks againJP
-------- Message d'origine --------De : bdbcat ***@***.***> Date : 20/05/2025  03:58  (GMT+01:00) À : OpenCPN/OpenCPN ***@***.***> Cc : JP Joubert ***@***.***>, Comment ***@***.***> Objet : Re: [OpenCPN/OpenCPN] Impossible to import tracks saved in a .gpx file (from OpenCPN) (Issue #4538) 
bdbcat left a comment (OpenCPN/OpenCPN#4538)
For those interested in the technical details, here is great article:
https://blogs.gnome.org/jnelson/2015/01/06/sqlite-vacuum-and-auto_vacuum/

bdbcat, 2025-05-23:
Please close, if satisfied.