OpenCPN/OpenCPN #4539: Mismatch in date and time format for route properties

Hakansv, 2025-05-12:
**Describe the bug**
There's a mix of date and time format, see shot
In the header ISO 8601 is used. Probably since I selected local time in options->Units.
In the properties table US date/time format is used.
Also has the time an extension that describe the time zone and a possible use of summer time. 

**Expected behavior**
A consistent use of date/time format.
No description of time zone and possible summer time. It takes up space with no use.
The time zone is also in local language, as advised by the OS even if I use Eng(US) in OCPN.

**Screenshots**

[image]

**Desktop (please complete the following information if applicable):**
 - OS: [Win 10]
 - OCPN Version [Git master 5.11]

**Additional context**
The new strange time format including time time zone and normal/summer time is disturbing more dialogs. It's unnecessary information and I suggest we go back to previous use.
Example of the new time:
[image]

sebastien-rosset, 2025-05-12:
'o'l'o'

sebastien-rosset, 2025-05-12:
If we could magically update all plugins to use the new `toUsrDateTimeFormat_Plugin` function, then the date/time format would be consistent everywhere in the UI. In that ideal scenario, we could remove the timezone from every widget, which would reduce the use of screen real estate.
We could also show "UTC" or "Local Time" in the status bar to make it clear what is the date/time global setting. But unfortunately it's going to take a while until all plugins are upgraded to use the new function, if ever. During that time, OpenCPN core (including grib/dashboard/wmm plugins) show date/time consistently, but other plugins do not.

Possible solutions:

1. Use timezone abbreviations: https://en.wikipedia.org/wiki/List_of_tz_database_time_zones. This requires importing a list of timezone abbreviations, because as you've shown, the builtin `wxDateTime.Format("%Z")` returns the full name of the timezone on Windows 10.
2. Use UTC offset, e.g. `+03:00`
3. Don't format date/time with timezone info. I.e. format the date/time according to global setting (UTC or Local Time), but don't show the timezone. The upside is that it takes less screen real estate. The downside is that if the user has loaded a plugin that does not format date/time consistently, it is confusing.

Hakansv, 2025-05-12:
We could make the tzname optional when using "getUsrDateTimeFormat()". So if anyone think it's valuable info, also on Win, it can be free to use?
I think Dave has some thoughts about this as well. Maybe on Z?

sebastien-rosset, 2025-05-12:
> We could make the tzname optional when using "getUsrDateTimeFormat()". So if anyone think it's valuable info, also on Win, it can be free to use? I think Dave has some thoughts about this as well. Maybe on Z?

Do you mean providing a global option to show timezone when formatting date/time? I.e. in addition to selecting whether date/time should be formatted in UTC or local time, there would be an option to specify whether timezone is included or not?

bdbcat, 2025-05-12:
Leaving aside plugins, for now.  They can be updated off-cycle, whenever.  We need to get the core workable.
I prefer option 1) above, to give consistency across all platforms, without platform specific options.  Do not use the "system %Z" specifier at all.

 A bit more codespace, but worth it in this case.

Hakansv, 2025-05-13:
> Do you mean providing a global option to show timezone when formatting date/time? I.e. in addition to selecting whether date/time should be formatted in UTC or local time, there would be an option to specify whether timezone is included or not?

No, that's not my intention. Rather make it optional for the developer of a GUI. But so far I've not seen any example where this information adds something valuable. A far-fetched benefit could be in connection with tidal management, but even there is often a lack of space.
So, if you ask me we could take out all about tzName from datetime.cpp.

rgleason, 2025-05-13:
Sebastien is working very hard to make this happen in Weatherrouting, which we really need, so don't screw this up.

leamas, 2025-05-13:
On a sidenote, %Z is specified to use a time zone abbreviation. Håkan's example is not, we are probably facing some kind of msvc bug.

C++ tends to be more standardized.  `std:;:put_time`  seems to have the same basic capabilities as  strftime which I assume is what's used here (?)  If so, a put_time test on windows might be worthwhile, using a standard function is less work to code and maintain.

bdbcat, 2025-05-13:
From docs:
 "Z" : writes locale-dependent time zone name or abbreviation, or no characters if the time zone information is not available 

Windows does "locale-dependent time zone name "
Most everything else does something like "EDT", or so.   I think this is what we want for platform independence of screen layouts.

Hakansv, 2025-05-13:
> Developers can do that already. There are formatting options to specify custom date/time. 

How can we specify to not use tzName for local time?

sebastien-rosset, 2025-05-13:
As a side note, the `toUsrDateTimeFormat` function uses locale-dependent formatting of date and time, as long as OpenCPN uses wxWidgets 3.1.6+
It properly respects locale-dependent date/time formatting for improved internationalization. For instance, US users will see dates like "Mon 5/13/2025 2:30 PM EDT", while German users will see "Mo. 13.05.2025 14:30 MESZ" and Japanese users will see "月 2025/05/13 14:30 JST". 

This of course assumes the wxWidget library is correctly honoring instructions like `wxLOCALE_LONG_DATE_FMT` and `wxLOCALE_SHORT_DATE_FMT`.

Additionally, the `toUsrDateTimeFormat` function is used to format date/time as a string. I.e. it can be used to format the date/time as highlighted in green color below. The format of the date/time picker is a separate piece of code.

[image]

@Hakansv , in your system, what is the locale-dependent time format? is it 24-hour or 12-hour?

Hakansv, 2025-05-13:
My Win system is CET and Swedish so ISO 8601 i.e. 24 hours.
For OCPN units I've selected "Local Time".
In OCPN I use language ENG(US). That's why route leg uses the US date/time format.
If I in OCPN change language to Swedish (SE) the route leg date/time format will be ISO 8601

sebastien-rosset, 2025-05-13:
> My Win system is CET and Swedish so ISO 8601 i.e. 24 hours. For OCPN units I've selected "Local Time". In OCPN I use language ENG(US). That's why route leg uses the US date/time format. If I in OCPN change language to Swedish (SE) the route leg date/time format will be ISO 8601

Thanks. So just to confirm, the formatting of the date/time highlighted in green is correct for your configuration. What is not correct is the time picker format. It should use 12-hour time format and show AM/PM because you have configured OCPN to use language ENG(US).

sebastien-rosset, 2025-05-13:
> How can we specify to not use tzName for local time?

I take this back. I've just added an option to show/hide the timezone. I'm also working on timezone abbreviation for Windows.

Hakansv, 2025-05-13:
> Thanks. So just to confirm, the formatting of the date/time highlighted in green is correct for your configuration. What is not correct is the time picker format. It should use 12-hour time format and show AM/PM because you have configured OCPN to use language ENG(US).

Yes, confirmed.
As mentioned some time before I'd like to have an option to always use ISO 8601 independent of selected OCPN language, but that is beyond this issue. 
Next would be convincing the US to adapt to ISO 8601, but that is beyond OCPN I suppose. :)

rgleason, 2025-05-14:
Yes we are a backward country now.

transmitterdan, 2025-05-19:
On Windows we are now throwing wx exceptions in funciton ```bool NavObj_dB::UpdateDBRoutePointAttributes(RoutePoint* point)``` at this line ```time_t etd = point->GetETD().GetTicks();``` (about line no. 1275) the GetEDT() can (and does) return wxInvaldDateTime which when passed on to GetTicks() throws an exception. Do we expect GetETD() to never return an invalid date/time?

bdbcat, 2025-05-19:
Good catch.
point->GetETD() is actually not used very often, usually is uninitialized wxInvalidDateTime.  
I'll fix that.

transmitterdan, 2025-05-19:
I haven't seen an obvious problem with the TZ displayed for ETD on the Route Properties table.

bdbcat, 2025-05-23:
Ready to close?

Hakansv, 2025-05-23:
Although the discussions here have solved other things the original issue is still there.
It is still ISO8601 format in the header but language adapted in the table.
Personally, I prefer ISO 8601 and think it can be used everywhere but maybe it should be consistent throughout the dialogue?

**English-(US)**
[image]

**Swedish**
[image]

Hakansv, 2025-06-01:
There is no clear reaction to this issue during the ongoing Beta, so it may be that I am being unnecessarily consistent. 
Personally, as I said, I appreciate ISO 8601 and am not bothered by that format. So if those who are not used to seeing that format have not commented, it is probably accepted.