OpenCPN/OpenCPN #4542: Restarting OCPN 5.11 latest loses tracks and abbreviates route names

antipole2, 2025-05-18:
**Describe the bug**
Working on MacOS 5.11.0-universal+17eadb653..
Tracks are being lost
Route names being truncated to 25? characters

**To Reproduce**
Steps to reproduce the behavior:
1. Examine and note tracks and routes.  Ensure some rout names longer than 25 characters.
2. Quite OCPN
3. Restart OCPN
4. Examine tracks and routes

I find many of my tracks lost.
Route names have been abbreviated to 25 characters e.g.
'2025: Præstø to Vordingborg' -> '2025: Præstø to Vordingbo'

bdbcat, 2025-05-18:
Is this two bugs?
1.  Name truncation
2. Tracks lost.
Or are the two effects related somehow?

antipole2, 2025-05-18:
I don't know whether related or not.  Do you need me to split this into two issues?

bdbcat, 2025-05-18:
Yes, please.  Easier to track.  One error is easy to fix, I think.  The other, not so easy...
More info on the tracks loss, please.  Old tracks lost in import?  New tracks lost after import?

antipole2, 2025-05-18:
This one now just about abbreviation of route names.

Losing tracks to follow when I have time - must sail tomorrow

antipole2, 2025-05-18:
Now not seeing abbreviation when testing just that.  Maybe they are related and it was mistake to separate.

Sorry - no more time just now.  Must sail with last of fair wind.  Will check again when I can.

bcn2, 2025-05-18:
just tested on macOS and no truncation...

Hakansv, 2025-05-18:
On W10 tested step 1 to 4
The route name: "To test and very loooooooooooooong route name" was not truncated. (46 chr)

bcn2, 2025-05-18:
Dave...

from yesterdays commits routes are now as well SQL based - right? So when testing against the latest builds routes shall be imported/exported from the db?

antipole2, 2025-05-18:
Rather than deleting the navobj.xml file, perhaps it should be rotated - becoming navobj.xml.1
That way, there is a way back, should it be needed.

bdbcat, 2025-05-19:
For very first time O5.12 users (i.e. not Alpha testers like you), the navobj.xml file is backed up once as "navobj.xml.import_backup".  This is treated as a "write-once" file, never deleted of overwritten by OCPN, and contains the navobj.xml collection present when O5.12 is first installed.  This file could be copied to navobj.xml, and re-used for O5.10.  

We officially abandon the idea of rotating navobj backups.  Experience suggests that they are not very useful, and so not often used.

I'm pretty sure that we will all be very happy to never think about navobj,xml ever again...

antipole2, 2025-05-19:
@bdbcat   That very helpful.  I have now worked out how to switch between versions, sailing with the reliable v5.10.2:

1. Make passage under v5.10.2
2. Quit v5.10.2
3. Save copy of navobj.xml file somewhere
4. DELETE any navobj.db to ensure fresh import
5. Launch v5.11 - nice conversion progress window
6. See all my routes and tracks (but see what follows re abbreviation)
7. Play with v5.11
8. Quit v5.11
9. Restore saved navobj.xml
10. Launch v5.19.2
11. Make new passage

antipole2, 2025-05-19:
Still seeing abbreviated route and track names

Under v5.10.2

[image]

[image]
Switch to v5.11 as above - get conversion progress window.
Here are the routes and tracks

[image]

[image]

Very weirdly, some of the routes are not being abbreviated - the ones starting '2025 plan:..' whereas the ones starting '2025:..' are.  These latter were edited as they moved from outline plan to actual routes to be sailed. I have no idea why one is impacted and not the other.
I have checked in the xml file and the names are correct - not abbreviated.
In v5.11 they are abbreviated and if I look at their properties, the name is abbreviated there too - so it is not a display issue.
My only suggestion is to send you the xml file

bdbcat, 2025-05-19:
Please do send me the XML file.  I will investigate.

antipole2, 2025-05-19:
Dave...Think you have been sent a DropBox link for the file.  If not, I can email you it myself.

bdbcat, 2025-05-19:
Tony...
Route name clipping issue resolved, will push shortly.
Root cause:  You may have noticed that the name was truncated when it contained a special character, such umlaut.  The database code needed to be taught how properly to convert from wxString to const char*.  All good now.
Updated build should be available in a few hours.

bcn2, 2025-05-19:
Dave..
what I see including an umlaut like Ö is that the string is truncated exactly by one char for each umlaut

bdbcat, 2025-05-19:
Exactly correct.  Database records deal in bytes, not characters.  UTF-8 for umlaut character is two bytes, not one.  So the byte length of the string is not equal to the character length of the string, if special characters are used.
Thanks for confirmation.

antipole2, 2025-05-20:
I confirm that the latest build 5.11.0-universal+53af9dfb3 has fixed the abbreviation issue - thanks.

However, I encountered strange behaviour not seen before.

1. It treated me as a first time user, taking me through choice of units etc.
2. It also looked for nav inputs.  I did not have any switched on so it took a long time to give up.  A skip button would be useful if you are launching OCPN not on a boat.
3. It asked for chart location and somehow forgot all my charts in the process.

It seems it did not find my .ini file and it did not import my navobj.xml.  In fact, it did not even create an empty db file.  So I ended up with a vanilla system.

I went back to the previous build and it still did not import the .xml stuff.

Thanks to the wonders of TimeMachine, I rolled back my .ini and o_charts_pi folder and then it imported the xml file and I was able to rebuild all charts.

This may have been a one-off quirk but I mention it in case it was not just me.

antipole2, 2025-05-21:
Let's assume my oddity a one-off and close this.

closed 2025-05-21 by antipole2