OpenCPN/OpenCPN #4543: Generate an OpenCPN Message upon routepoint creation

TwoCanPlugIn, 2025-05-19:
Excellent news on the adoption of SQLite as the underlying store for routes, tracks and routepoints.

Upon creation of a routepoint can core publish a OCPN_WAYPOINT_CREATED message.

Something like:
```
UpdateDBRoutePointAttributes(point);
.....
wxJSONValue v;
v[("Name")] = point->GetName();
v[("GUID")] = point->m_GUID;
v[("Latitude")] = point->GetLatitude();
v[("Longitude")] = point->GetLongitude();
wxString msg_id(("OCPN_WPT_CREATED"));
g_pi_manager->SendJSONMessageToAllPlugins(msg_id, v);
```
This would enable plugins, such as TwoCan to automatically "export" waypoints to third party MFD's over NMEA 2000 using PGN 130074.

As an aside, for backup of the SQLite database, rather than a file copy that seems to have been  implemented, why not use the SQLite Backup API?

```
bool NavObj_dB::Backup(wxString backupFile) {
  sqlite3_backup* pBackup;
  sqlite3* backupDatabase;

  if (sqlite3_open(backupFile..ToStdString().c_str(), &backupDatabase) == SQLITE_OK) {
    pBackup = sqlite3_backup_init(backupDatabase, "main", m_db, "main");
    if (pBackup) {
      int result = sqlite3_backup_step(pBackup, -1);
      if ((result == SQLITE_OK) || (result == SQLITE_DONE)) {
        if (sqlite3_backup_finish(pBackup) == SQLITE_OK) {
          return true;
        }
      }
    }
  }
  ReportError("Backup Failed");
  return false;
}
```

bdbcat, 2025-05-19:
TwoCan...
I'll consider the plugin message idea.

On backups:  Backups of SQL database us  trivial.  The backup/restore API does bring much beyond a file copy.  However, a coherent backup/restore policy, with related GUI, is not.
Start with a simple question:  When should a full backup of the navobj.db be performed?

bdbcat, 2025-05-19:
TwoCan...
The current implementation uses standard rollback logging.  So far, the performance seems OK without any funny business here.  So if O should die (or be killed) while inside a db transaction, the db will be automatically repaired when next accessed.
The chances of the SQLite library crashing are pretty remote.  If O crashes due to an unrelated problem, the database will be secure.
We update the db on (almost) all user object manipulation _as it happen_s in real time.  The only exception is object "dragging" on-screen.  We defer the db update until the drag is finished, mouse button released.
Track points are recorded immediately.
All db operations are done on the main app thread.

TwoCanPlugIn, 2025-05-19:
Isn't it wonderful what a decent database brings to the table! Hopefully no more navob.xml woes.

leamas, 2026-01-09:
While the need of a "somethting happened" in the plugin API is obvious I'm not sure that a message is the correct solution. The reason is basically that they don't scale that well.

A perhaps better solution could be  to directly register a callback. Something like this in the API:
```

RegisterEventCallback(const std::string plugin_name, int evt_mask,
                      std::function<void(int)> callback);
```
That is, the  callback is invoked with an integer argument which contains the event (bit) which happened. The mask argument defines which events (bits) the plugin is interested in.

How to design this in detail is  a trade off between usability in this particular use case vs how  general the callback should be.  The proposal above is to just provide the info "there is a new waypoint available" rather than providing  the actual properties.

If we keep it in a general form like above the interface could be expanded to all sorts of events  by just adding more bits.

Registering a callback this way is way scales better than the "send message" approach since the only thing  to deal with is the list of subscribers rather than all plugins. And even if json serializing/deserialializng *is* fast just providing an integer bit mask is of course more efficient.

quinton-hoole, 2026-01-09:
> Yes. Things like these needs time to iron out details. I just created new milestone 5.16, assigning to it. 
> 
> This is not  set in stone and might certainly be changed if need be.

Agreed. I can immediately see a few holes in the argument for a generic "something happened" callback, especially if the typical response from the plugin is to call back into the core (e.g. to retrieve the new waypoint details). 

So yes, I agree that it's a non-trivial piece of functionality that needs to be thought out carefully, and exposed to debate before implementation.