OpenCPN/OpenCPN #4545: Data Monitor, Edit Filters, Layout - macOS 15.5

bcn2, 2025-05-21:
**Describe the bug**
- Layout window does hide partially under Main Data Window. Clicking does not change that, stays under the main. Have to drag the main window sideways. Z-order?

- Filter setting single items are hidden. Narrowing the Window and enlarging it will bring them up (partly). The last screenshot shows the maximum extension. Missing horizontal sliders? Which would be ugly here anyhow.

**To Reproduce**

[image]

[image]

[image]

[image]

[image]

**Desktop (please complete the following information if applicable):**
 - OS: macOS 15.5, M4

leamas, 2025-05-21:
> Layout window does hide partially under Main Data Window. Clicking does not change that, stays under the main. Have to drag the main window sideways. Z-order?

The Data Monitor (DM) is running  using FLOAT_ON_PARENT which is necessary to avoid that it gets hidden under the main canvas window (which is the parent). This basically works as expected.

OTOH, the dialog is not modal so placing it on top of the DM is not really what you want. The nice thing is that changes in the dialog immediately is reflected in the DM (if editing the active filter). This means that it is perfectly sane to have the DM on top of the dialog after doing some edits,  but when actually editing you  want the dialog on top. Ergo: not a z ordering issue.

So the question then becomes how OS treats the dialog when initially opened. GTK and Windows places it on top, if I understand you correctly MacOS does not. But since a defined z -ordering  isn't really what we want, I don't see the solution. 

Do you?

leamas, 2025-05-21:
> Filter setting single items are hidden. Narrowing the Window and enlarging it will bring them up (partly). The last screenshot shows the maximum extension. Missing horizontal sliders? Which would be ugly here anyhow.

What is the actual problem here?  IIUC the window is created using it's natural size showing all elements.  Narrowing as per your screenshot is a user action. Given that user narrows the window like this the result should be no surprise. 

The window should  be narrow enough to fit also on small screens. If this does not hold, we certainly need horizontal sliders. But AFAICT is should hold (?)

bcn2, 2025-05-21:
On the first run even with sufficient horizontal space, the single items are hidden to the right. You need to narrow the window and broaden it again to make the content visible. Even then some chars are still disappearing to the right.

On a second run during the same session, the lay-out is correct with content.
In a new session the same game as described above.

Some more testing under macOS would be helpful perhaps - to discard that this an isolated issue only on our system.

bcn2, 2025-05-21:
On our Apple M1 with Sequoia 15.2 the same game:

[image]

leamas, 2025-05-21:
I can actually reproduce the mess on my too old MacOS VM. However, the build chain on it is broken.

Looking at the code I see no obvious errors. Ans both GTK and Windows  does the right thing here. However, those windows uses delayed creating which, IIRC, has caused problems on MacOS earlier.  

You might want to try patch below. It's just a shot in the dark.
```
diff --git a/gui/src/filter_dlg.cpp b/gui/src/filter_dlg.cpp
index 07a63b151..079261b7c 100644
--- a/gui/src/filter_dlg.cpp
+++ b/gui/src/filter_dlg.cpp
@@ -340,13 +340,12 @@ class SetPanel : public wxPanel {
 public:
   SetPanel(wxWindow* parent, std::set<T>& set, std::function<void()> on_update,
            const std::string& label, std::function<wxArrayString()> get_choices)
-      : wxPanel(),
+      : wxPanel(parent, wxID_ANY),
         m_set(set),
         m_on_update(std::move(on_update)),
         kEditBtnId(wxWindow::NewControlId()),
         kListboxId(wxWindow::NewControlId()),
         kListLabelId(wxWindow::NewControlId()) {
-    wxPanel::Create(parent, wxID_ANY);
     auto flags = wxSizerFlags(0).Border();
     auto vbox = new wxBoxSizer(wxVERTICAL);
     auto hbox = new wxBoxSizer(wxHORIZONTAL);

```

bdbcat, 2025-05-22:
Tried the patch locally, no help.

quinton-hoole, 2025-05-22:
See also useful discussion in https://github.com/rgleason/weather_routing_pi/issues/397

It seems like this might be a known bug in wxwidgets on macOS, for which there seems to possibly be a good workaround. More thought needed.

quinton-hoole, 2025-05-22:
https://github.com/wxWidgets/wxWidgets/issues/19129

leamas, 2025-05-22:
Seems to be a known issue: https://discuss.wxpython.org/t/checklistbox-overly-left-padding-inserted-strings/34545/9

bdbcat, 2025-05-23:
Corrected, by my local testing.
Pretty typical, but non-sensible, changes required to force full layout on complex MacOS dialogs.

quinton-hoole, 2025-05-23:
> @quinton-hoole  The z-ordering issue reported here is basically moot, see [above](https://github.com/OpenCPN/OpenCPN/issues/4545#issuecomment-2897486872).
> 
> The remaining issue is about (initial) layout of some panels. This is IMHO not related to the wxWidgets bug mentioned or   https://github.com/rgleason/weather_routing_pi/issues/397

I'm fairly certain that there is still a z-ordering problem on macos, as per https://github.com/rgleason/weather_routing_pi/issues/397. I will try to narrow it down/fix it in that issue.

bcn2, 2025-05-23:
Layout now confirmed ok:

[image]

bcn2, 2025-07-18:
Confirmed looking very acceptable (tested DE as well) - the few truncated items are still correctly understood

closed 2025-07-18 by bcn2