OpenCPN/OpenCPN #4554: Occasional crash around N2K driver when quitting OCPN

antipole2, 2025-05-27:
**Describe the bug**
Sometimes when I quit OCPN I get a crash related to the N2K driver

**To Reproduce**
Steps to reproduce the behaviour:
I am using a particular sequence during plugin developement
1. Compile plugin to get tarball
2. In OpenCPN import the tarball
3. Quit OCPN (I seem to need to do this if the import is not the first time)
4. Restart OCPN and test
About 20% of the time,I get a crash at 3 when I quit.
**Expected behavior**

**Screenshots**
Crash report

[Crash on quit after plugin import.zip](https://github.com/user-attachments/files/20452191/Crash.on.quit.after.plugin.import.zip)
It is always at CommDriverN2KSerial::Close( ).
I do have an N2K stream active but no hardware connected, so it is not active.

[image]

**Desktop (please complete the following information if applicable):**
 - OS: MacOS
 - Version 15.4.1

**Additional context**
Using test builds of OCPN - latest seen with build 5.11.0-universal+859ca42e5

antipole2, 2025-06-19:
Will do, but it is only an occasional crash for me, so will need to give it a few days before I can opine.

antipole2, 2025-06-22:
Dave…. Had a disastrous (navigation-wise) passage today using a recent OCPN build  (19 June 5.11.2-universal+2ef04cc61).  I realise this was at my own risk.  But am detailing here in the hope it contributes to bug hunting.

Fortunately, my at the wheel station iPad running TZ iBoat carried on using its own GPS and I have recovered my track record from there.

Firstly, the input from WiFi did not start itself.  But every time I turned a comms port off, I got a crash - maybe 5 or 6 tries. Also the ship’s instruments not making sense,
Eventually, in desperation, I turned off all instruments, including N2K network, and switched back on. Then all came well.

Conclusion: My N2K system had got in a twist and none of the inputs to OCPN was functioning.  So turning off a non-functioning network connection was crashing OCPN.
Two crash reports attached.

[image]

[image]

After everything was sorted and we finally berthed, I deactivated my route and got another crash.  Report attached.

[image]

Finally, I imported my route exported as GPX from TZ iBoat and OCPN hung.  No crash report.  If I see this again, I will report separately.

bdbcat, 2025-06-22:
Tony...
For the three crash report:
Did you have any output connections active at the time?

bdbcat, 2025-06-24:
Tony...
From the above I surmise that the corrupted N2K WiFi connection was the root cause of the crashes.  If so, not a good thing, but pretty hard to reproduce.

Which brings us back around to the top.
I am looking for confirmation that the original source of the original crashes (in this issue) is resolved.
I think it is.  I reproduced it locally.  I fixed it.  I have never seen it again.
Do you agree?

If we have crashes in otherwise clean conditions, please start a new issue.  Otherwise, this one is becoming impossible to follow in any useful way.

Thanks
Dave

antipole2, 2025-06-24:
Good tack, Dave.
I commented out the `nmeaSend.push();`  and ran the script ten times without a problem.
I then re-enabled that step and it crashed on the second run.

So looks like the send may not be cleaning up properly and crashes on repeat?

bdbcat, 2025-06-24:
Tony...
I added more protection on serial N2K driver during shutdown.
This will probably prevent the crash at N2K shutdown, but may not reveal the source of the original forced exit.
Building now.

bdbcat, 2025-06-25:
I will, thanks.  I'll work on linux, might crash in a spot I understand.

antipole2, 2025-06-28:
Not quite there yet...
Using Beta 3 I can run the test script I gave you to try and push PGN 59904 to a non-functioning N2K port and no crash.

However, I have now got my W2K-1 converter receiving  N2K data over WiFi as per the Actisense instructions for OpenCPN.  I can see N2K data arriving in the data monitor window and, if I run my WhoGoesThre scriot without sending, I get a full report on the N2K data received.

But, if I push out PGN 59904 twice, OCPN quits cleanly.  The difference here is that previously the N2K port was not working.  Now it is working and I get a quit.  This is true whether the port is set to input only or input and output.
Nothing helpful in the logfile.

> 13:30:18.607 MESSAGE base_platform.cpp:273 PlugInManager: Using data dirs from: /Applications/OpenCPN.app/Contents/SharedSupport/plugins;/Users/Tony/Library/Application Support/OpenCPN/Contents/SharedSupport/plugins;/Applications/OpenCPN β3.app/Contents/PlugIns
> 13:30:18.608 MESSAGE base_platform.cpp:285 PlugInManager: using data dir: /Users/Tony/Library/Application Support/OpenCPN/Contents/SharedSupport/plugins/JavaScript_pi
> 13:30:18.613 MESSAGE base_platform.cpp:273 PlugInManager: Using data dirs from: /Applications/OpenCPN.app/Contents/SharedSupport/plugins;/Users/Tony/Library/Application Support/OpenCPN/Contents/SharedSupport/plugins;/Applications/OpenCPN β3.app/Contents/PlugIns
> 13:30:18.613 MESSAGE base_platform.cpp:285 PlugInManager: using data dir: /Users/Tony/Library/Application Support/OpenCPN/Contents/SharedSupport/plugins/JavaScript_pi
> 13:30:18.614 MESSAGE base_platform.cpp:273 PlugInManager: Using data dirs from: /Applications/OpenCPN.app/Contents/SharedSupport/plugins;/Users/Tony/Library/Application Support/OpenCPN/Contents/SharedSupport/plugins;/Applications/OpenCPN β3.app/Contents/PlugIns
> 13:30:18.614 MESSAGE base_platform.cpp:285 PlugInManager: using data dir: /Users/Tony/Library/Application Support/OpenCPN/Contents/SharedSupport/plugins/JavaScript_pi
> 13:30:18.685 MESSAGE ocpn_app.cpp:1995 opencpn::MyApp starting exit.
> 13:30:18.686 MESSAGE ocpn_app.cpp:2035 LOGBOOK:  2025-06-28 11:30:18 UTC OFF: Lat   54.37359 Lon    9.08442 COG ----- SOG   0.02 kts
> 13:30:18.686 MESSAGE TCDataSource.cpp:41 UnLoading Tide/Current data source: /Applications/OpenCPNv5.8.app/Contents/SharedSupport/tcdata/harmonics-dwf-20210110-free.tcd
> 13:30:18.688 MESSAGE TCDataSource.cpp:41 UnLoading Tide/Current data source: /Applications/OpenCPNv5.8.app/Contents/SharedSupport/tcdata/HARMONICS_NO_US.IDX
> 13:30:18.709 MESSAGE ocpn_app.cpp:2063 opencpn::MyApp exiting cleanly...

antipole2, 2025-06-28:
[image]

antipole2, 2025-06-28:
If you want the full script that analyses all inputs and looks at what is out there, see here https://github.com/antipole2/WhoGoesThere

bdbcat, 2025-06-28:
Hmmm...
I have run the script you gave me earlier on a live N2K connection, Actisense in N2K mode.  No trouble.
Unlikely that the current fault is due to the comm driver, since we are getting a clean exit.
I'll try the full script next.

bdbcat, 2025-06-28:
Tony...
I am on Windows, using Actisense W2K-1, which is a WiFi  gateway, configured as N2K raw data.
N2K PGNs are received and decoded, position is correct.

Now I run the full WhoGoesThere script.
It runs without trouble, producing a report of the one detected GNSS source on my little N2K network.
Thoughts?

antipole2, 2025-06-28:
****Firstly****, If you look at the output from the full script just before the NMEA2000 samples heading, there is a count of the NMEA2000 126996 responses.
I am seeing
`No NMEA2000 126996 responses.`
Are you seeing no responses?

I suspect this is related to this issue https://github.com/antipole2/WhoGoesThere/issues/6
When I developed this script, my N2K interface was the Actisense NGT-1.  The issue discusses that not all N2K interfaces transmit 126996, I.e. Yacht Devices YDWG-2.

I have no confirmation that the W2K actually sends out 126996.  If not, then no responses.

**Secondly**,  this is awfully complicated…
When I wrote the NMEA2000 object, I inadvertently used a very obsolete branch of the Canboat project and several PGNs do not decode.  This is what JS 2.1.1 is using.

For the JavaScript_pi 2.2 (not yet released) I adopted the current Canboat schema.  This was a major re-write of the NMEA2000 object.

I am running 2.2 and you are running 2.1.1.

I have tried going back to 2.1.1 but have been unable get it working properly and I do not want to mess too deeply whilst cruising - we sail again tomorrow.
This quitting MIGHT be related to my updates in 2.2.

When I quite OCPN normally, the JS plugin logs the start and end of its deinit process.
When OCPN quits on a second push of 126996, these entries are missing from the log.
Maybe the plugin is quitting and triggering OCPN to shut down.  There is no quit capability in the plugin APIs but perhaps the code is quitting.

**Summary**

(1) If you are seeing zero responses to 126996, that looks like an issue with the W2K.

(2) If you can confirm that you can run the script multiple times without an exit, I must accept that there is an issue with my rewrite of NMEA2000 object or JS 2.2.  That maybe one for when I am back ashore.

PS For your interest, this what I get in my report.  The missing 126996 responses would show what devices the different sources are.

```
**OpenCPN messages received since plugin started**
AIS
GLOBAL_SETTINGS_UPDATED
OCPN_CORE_SIGNALK
OCPN_DRAW_PI_READY_FOR_REQUESTS
OCPN_OPENGL_CONFIG
OpenCPN Config

**NMEA0183 samples**
Counts	Sentence
10		$SDDBT,5.77,f,1.76,M,0.96,F
10		$SDDPT,1.76,-0.18,
10		$GPGGA,190924,5422.4151,N,00905.0650,E,1,11,0.92,45.89,M,45.19,M,,
10		$GPGLL,5422.4151,N,00905.0650,E,190924,A,A
10		$GPGNS,190924,5422.4151,N,00905.0650,E,A,11,0.92,45.9,45.2,,,V
10		$GPGRS,190924,1,-0.0,-0.0,0.0,0.0,-0.0,-0.0,-0.0,-0.0,0.0,0.0,-0.0,,,
8		$GPGSV,3,3,12,20,47,210,39,27,08,024,33,08,10,058,32,07,43,066,47
20		$HCHDM,228.7,M
10		$IIMDA,,I,,B,18,C,16.4,C,,,,C,,T,,M,,N,,M
10		$IIMTW,16.4,C
10		$WIMWV,104.78,R,11.68,N,A
11		$GPRMC,190924,A,5422.4151,N,00905.0650,E,0.04,0,121105,,,A,V
10		$TIROT,0.41,A
10		$IIRSA,-0.4,A,,V
10		$IIVHW,,T,228.7,M,0,N,0,K
10		$IIVLW,17902.77,N,17902.77,N,,N,,N
10		$GPVTG,0,T,,M,0.04,N,0.07,K,A
10		$WIVWR,105.5,R,11.9,N,6.1,M,22.1,K
10		$GPZDA,190923,12,11,2005,,

No NMEA2000 126996 responses

**NMEA2000 samples**
Counts	Source	PGN
3		0		60928	ISO Address Claim
9		5		61184	Seatalk: Wireless Keypad Light Control
22388     5		126720	Seatalk1: Pilot Mode
10		6		126992	System Time
101		5		127245	Rudder
102		5		127250	Vessel Heading
102		5		127251	Rate of Turn
51		5		127257	Attitude
10		35		128259	Speed
10		35		128267	Water Depth
10		35		128275	Distance Log
101		6		129025	Position, Rapid Update
40		6		129026	COG & SOG, Rapid Update
10		6		129029	GNSS Position Data
2		7		129038	AIS Class A Position Report
10            6		129539	GNSS DOPs
10		6		129540	GNSS Sats in View
101		10		130306	Wind Data
20		35		130311	Environmental Parameters
5		10		130312	Temperature
5		10		130314	Actual Pressure
End of report
```

antipole2, 2025-06-29:
Dave...  I am having a **terrible** time with OCPN beta3 and been in serious difficulty with navigation today.  I am communicating with you through this issue because I do not want to sound off on the public forums.

Firstly, in the Connections panel, my NMEA0183 input keeps on flitting from tick to '!'.  Sometimes it goes back to tick but often I have to turn the channel off and back on again to get the tick back - not easy when sailing and needing to be at the helm.
I think '!' means malformed data and if I filter in the Data Monitor to show malformed data only, there is an awful lot of it.
But this allegedly malformed data has ticks by it, which indicates checksum OK I think.

How does OCPN judge NMEA0183 data to be malformed, other than by checksum?   Either OCPN is falsely flagging data as malformed, or my navigation system has suddenly started generating bad data but is putting a valid checksum on it.

Secondly, to try and cope with this, I have tried to make adjustments in the Connections panel.  I have moved away from in/out to separate in and out channels to the same port to try and separate the issues.  But just about every time I touch that panel, I gt a crash - so many I will send only a couple of samples.

Crashes when I disable a channel are common, especially if they do not have a tick status.  Some channels no longer connect to a valid destination and anything I do with them gets a crash.  I have tried binning such channels but I get a crash that way too.

So trying to sort out the first issue of malformed data while getting crashes whenever I touch anything has been a nightmare, particularly when navigating in confined waters.

Here are a few crash reports

[crashes.zip](https://github.com/user-attachments/files/20969981/crashes.zip)

bdbcat, 2025-06-29:
Is the malformed data coming from N2K, or N0183?

bdbcat, 2025-06-30:
Tony...
A breakthrough.  It turns out that your plugin is the first time anyone has tested N2K output to a network connection (like W2K-1).  And there is a bug in SendMessage() for N2K network connections.
Detail:  N2K serial output expects your "chopped" payload.  But N2k net expects the full message, including the header  Obviously, this is wrong.  They must both be consistent. But it does match your test results.
So, I have some homework to fix this.
Thanks for the analysis, and the exercise.

Does the latest OCPN build in Cloudsmith help with your connections edit crashes?

bdbcat, 2025-06-30:
For your thinking and planning:
I am planning to standardize on the "clipped" payload model.
This mode already in use by Douwe's  AutoTrackRaymarine_pi plugin, in addition to your js_pi for serial port.
So, as the function argument says, this is a "payload", not a full message.

bdbcat, 2025-06-30:
Tony...
I appreciate your flexibility.  I really prefer not to break the one or two existing Production N2K support plugins.
This change will come overnight tonight, for all drivers accessible to WriteCommDriverN2K() interfaces.
Thanks

bdbcat, 2025-06-30:
Tony...
Can you tell me the N2K mode that your W2K-1 is using for PC side communication?

bdbcat, 2025-07-01:
Tony...
Some experimentation with W2K-1 Actisense.  I learned several things:
1.  The PGN 59904 has one 3-byte data field.  This is the PGN that is being requested.  So in your script, you probably want to say PGN59904(126996).
2. The W2K is a bit fussy about operating mode.  The one that works for me is "N2K ASCII", bidirectional.  See screenshot.

Here is the C code I have used to test this.

```
     // Get  comm handle
      std::vector<DriverHandle> drivers = GetActiveDrivers();

      auto list = std::vector<int>();
      list.push_back(59904);
      RegisterTXPGNs(drivers[0], list);

      auto data =
          std::make_shared<std::vector<unsigned char>>();
      // Push 126996 0x1F014
      data->push_back(0x14);
      data->push_back(0xF0);
      data->push_back(0x01);

      WriteCommDriverN2K(drivers[0], 59904, 255, 5, data);

```

When I execute this code, I see on my external N2K monitor that PGN 59904 has been sent, with three byte data field 126996.
And I see all members of my small N2K bus respond with PGN 126996.

So all of this says that the "chopped" data vector is the correct one for all N2K drivers.[

[image]

Good Luck
Dave

bdbcat, 2025-07-01:
" I am still seeing data coming from the W2K-1 via NMEA0183 frequently flagged as malformed,"
As I mentioned before, this flag does not really mean "malformed".  It means that no data have been transferred within the last two seconds.  It is possible that the gateway conversion function of the W2K is bottl-enecked?  I will try that interface here, as well.

bdbcat, 2025-07-02:
Tony...
Finally, the stars aligned, and I was able to re-produce the clean exit fault on second PGN transmission.  Everything has to be setup just so...
Anyway, I found the fault, corrected it.  Related to logging of output messages.
Tested on MacBook, using my little code snip above.  Works fine.  PGN-55904(126996) goes out on the net, and several PGN-126996 reports soon arrive.  I can repeat at will.
New code in github, building now.
Sorry for the misdirection and fussing around in the wrong place.

antipole2, 2025-07-04:
Dave... Had a difficult passage (from the OCPN perspective) from the River Eider to Wangeroog across the shipping lanes into Hamburg, Bremerhavn and Wilhelmshaven with no AIS on my cockpit TZ iPad display.  Ynskje supervised from OCPN below over headphones!

I have today reconfigured from scratch and have a working system again - provided I stay away from N2K.

## Lessons
I am not confident in having both NMEA0183 and N2K going.
I cannot yet be clear where the conflict lies, possibly in the area of priorities.

I suspect having the Connections display open may cause the '!' status to appear and stay on.
Disabling the connection and re-enabling it seems to get it back to ✔️

I had a crash disabling the N2K connection (crash report attached).  I have not seen any crashes when enabling/disabling NMEA0183 connections.

With N2K active, I can send  PGN 126996 multiple tines and it is seen by the W2K-1 but I am not getting any responses so doubt it is going out on the N2K network.  I am in contact with Actisense over that.  As I am Mac only, I cannot view the N2K logs.

When/if I solve this, I will document.  Meanwhile, can you please let me know which version of the W2K-1 you are using, as instructions will be different for the latest version.

Sorry if this is not more exact.  I will look out for more exact pointers.
Tony

bdbcat, 2025-07-04:
Here is my device info.

[image]

bdbcat, 2025-07-05:
Tony...
If you can recall, will you show me you connections setup at the time you got a crash by deactivating an active N2K connection?

antipole2, 2025-07-05:
Dave... Attached is a screenshot of my connections after my revision.

[image]

60001 is the NMEA0183 input from the W2K-1 via my router
60002 is NMEA0183 output to the W2K-1 and inc back onto the N2K network
60003 is NMEA0183 output to TZ iBoat on my iPhone but not in use t that time
60004 is NMEA0183 output to TZ iBoat on my iPad
60010 is N2K output to the W2K-1.  Got. Crash whenI disabled this.

It would really help if this table included the description text (or at least the first 10 or so characters).  I had an RFE on the old system about this.  I can raise one as an Issue.  I know you are not taking new features at this time but it is a hassle having to remember which port does what - and I have type all this to explain to you.

Just had another unfortunate passage with 60010 off - so no N2K ASCII involved.

The 60001 input keeps hanging up. Often resumes after 10-20 seconds but sometimes not at all until I turn it off and back on, when it starts working again.

This causes significant delays in the outputs.  My Garmin instrument stops showing autopilot information and the iPad stops updating via 60004 until 60001 resumes.

Today I did not realise the position on the iPad had stopped updating and I went too far before turning to take a narrow gap in a sand bank and I hit the bottom.

After that, I had my iPhone displaying the chart with WiFi off so it was using its own GPS.  I could then use that when the iPad plot was stuck.

I had several crashes (attached).  **I think this happens when I toggle any other input while 60001 is hung up.**

[Crashes.zip](https://github.com/user-attachments/files/21074900/Crashes.zip)

We are in Spiekeroog for a couple of days.  Unless you offer a fix to try, next passage I will try with N2K input and no 60001 input to hang up.

Also attached is this morning’s passage over the Watt at high tide. We closely followed the route (blue with diamond marks).  The yellow line is the recorded track which has straight lines from where things stopped recording to where I got it going again.  I haveincluded the notifications for good measure.

[image]

I hope this helps, Tony

antipole2, 2025-07-05:
Thanks for your thoughts on this.  I understand your argument about small screens.  I suppose to make it configurable would be complicated.

A good idea to use the /etc files to give the address and port names.  However, these names would not apply in the W2K-1 or the TZ iBoat app on iOS devices and to use names on just the Mac would, I fear, be inviting confusion or miss-matches.  It is helpful to see the port number or IP address matches at both ends.

antipole2, 2025-07-05:
Dave... I have knocked up a script 'Deadman' that monitors and logs loss or regain of fix and am testing under different load conditions.  For example, with just the 60001 input only, it looks OK.  I will run that over night in port.
I will also look at connecting directly to the W2K-1 without the router if I can find a situation where loading is a problem that can be matched without the router.

bdbcat, 2025-07-06:
In the build queue.  Maybe 15 minutes.

bdbcat, 2025-07-06:
I think that looping  the OCPN data back to N2K via 60002 may be effectively overloading the W2K-1.  
If the 0183->N2K conversion in the W2K does its job, that will mean essentially that every N2K message may be duplicated.  That could activate a N2K->N0183 conversion, and the results N0183 message will be sent out on 60001.  Which is repeated by 60002, And so on.  
So you might be getting a loopback situation that eventually overwhelms some channel buffers.
Does this make sense at all?

bdbcat, 2025-07-06:
A assume that you want this loopback from OCPN to add in something from js_pi?

bdbcat, 2025-07-07:
Tony...
I will dig into this.
Have you considered/tried making 60004 into a UDP connection, instead of  TCP?

antipole2, 2025-07-07:
Will think about UDP.  I used to use iNavX which does not support UDP.  TZ iBoat does.

I have just resurrected the FollowOCPN script that I used to use for iNavX.  It gives me complete control over the data to send and the handling of errors when the other device has disconnected.

**Question**
The options for a channel are 

- Receive input
- Output on this port (as autopilot repeater)

Surely, autopilot repeater is a sub-option of output.  Otherwise, how to I have an output port without OCPN sending whatever it fancies?

- Receive input
- Output on this port
- Output autopilot data on this port

antipole2, 2025-07-07:
Of course UDP is the answer.  I am rather embarrassed not to have realised this.

In mitigation, when I worked with connecting to iNavX, it did not support UDP.  (It now does).  So I had to develop complex script techniques to cope with TCP connections and disconnections.  And I have got caught carrying these concepts forward.

In a brief test this evening, it all works fine with UDP.  This is just what UDP was designed for.  I will try it in earnest in the coming days.

It would still be good, in the interests of robustness, to avoid the clogging up of input when another port disconnects but UDP avoids the problem I think.

antipole2, 2025-07-08:
**N2K input & output**

Dave… I have been thinking about the issue of N2K and the possibility of loop-back.

I recall a conversation with an Actisense engineer in January 2020 in which he explained that there was no loop- back.
Input:      NMEA0183 -> N2K bus
Output:  N2K bus -> NMEA0183

Presumably this is achieved by not outputting PGNs whose origin matches that of the W2K-1.  In the W2K-1 there are multiple servers and I presume they avoid feedback through both N2K ASCII and N2K routes.

Anyhow, to keep things simple for these tests , I have set my N2K channel to input only, so no loop-back possible.

**Observations**

With no N2K active, all is well and I get NMEA0183 output via 60002 back into the ship’s N2K instruments and via the UDP port  to my iPad apps.  This is good enough for me and how I will be using it on coming passages.

With only N2K input, OCPN gets its fixes and other course data etc. OK But there are no NMEA0183 outputs, so my iPad apps cannot play.

With both NMEA0183 and N2K inputs, the NMEA0183  channels soon clog up.  To get things working again, I have to disable N2K input and must then toggle off and on each of the NMEA0183 channels to get them working again.  By the way, NMEA0183 has priority over N2K input all cases.
Running both NMEA0183 and N2K  inputs is not good news.

To repeat, I have a satisfactory working set up.  This report is more about helping debug more complicated situations.

antipole2, 2025-07-10:
Still having problems with losing my connections and hence position.
Another problematic passage today.
To recap, I have the following connections via a router:
60001 NMEA0183 input from WK2-1
60002 NMEA0183 output to WK2-1
10000 NMEA0183 UDP output

I suspect problems may be linked to brief interruptions to the WiFi connection.

Testing in port, I briefly switch the Mac's WiFi connection to something other than my router and back again.  The OCPN log has

```
17:43:05.964 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.723   Lon: 7.36683
17:43:15.268 MESSAGE comm_drv_n0183_net.cpp:448 TCP NetworkDataStream connection established: TCP:192.168.1.162:60002
17:44:07.080 MESSAGE comm_drv_n0183_net.cpp:448 TCP NetworkDataStream connection established: TCP:192.168.1.162:60002
```
So the fix has been missed but there is nothing logged about 60001 errors.
However, it appears that 60002 is noted as failing because it attempts to reopen that port.

It sometimes recovers with lots of switching  back and forth between '!' and '✔️' on all the channels.
Sometimes it recovers but sometimes not.  And if not, OCPN fails to update its position again.
I have to go below and toggle the channels off and on to get it working again.

This is a serious issue for me with this set up.

bdbcat, 2025-07-12:
Tony...
Probably you are tired of experimenting, and want to just enjoy your cruise, but...
I discovered that if a net connection is set for output-only (like 60002), then the activity watchdog is not managed well.
Further, if there is no input activity processed, then the no-data watchdog may be triggered.
Corrected this, in github now, build should be available soon.
Good Luck
Dave

antipole2, 2025-07-14:
Dave...  I think things may be better since your patch.  But I am focussing first on getting it working reliably with 60001 for both input and output.

Just has a passage with it somewhat better but many times the displayed position lagged behind the reality on the water, sometimes just by a minute but sometimes it felt much longer - like minutes.  When manoeuvring in confined spaces or avoiding sandbanks, this is serious.

You fixed a problem with watchdog and output to my 60002.  I think there may be a similar issue with my input & output on 60001.

See this from the OCPN log
```
13:00:00.567 MESSAGE ocpn_frame.cpp:5692 LOGBOOK:  2025-07-14 11:00:00 UTC  GPS Lat   53.65272 Lon    7.05648 COG  213.56000 SOG   3.65 kts
13:00:09.217 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6527   Lon: 7.05648
13:00:25.666 MESSAGE comm_drv_n0183_net.cpp:410 NetworkDataStream connection lost: TCP:192.168.1.162:60001
13:00:30.815 MESSAGE comm_drv_n0183_net.cpp:448 TCP NetworkDataStream connection established: TCP:192.168.1.162:60001
13:01:19.217 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6515   Lon: 7.05398
13:02:31.219 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6511   Lon: 7.05302
13:03:15.219 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.651   Lon: 7.05232
13:03:20.150 MESSAGE comm_drv_n0183_net.cpp:448 TCP NetworkDataStream connection established: TCP:192.168.1.162:60001
13:03:30.219 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6506   Lon: 7.04991
13:05:03.289 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6504   Lon: 7.0476
13:05:26.218 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6503   Lon: 7.04726
13:06:47.220 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6501   Lon: 7.04479
13:08:16.219 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6493   Lon: 7.04226
13:08:20.398 MESSAGE comm_drv_n0183_net.cpp:448 TCP NetworkDataStream connection established: TCP:192.168.1.162:60001
13:10:04.220 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6474   Lon: 7.03924
13:10:26.260 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6473   Lon: 7.03899
13:11:01.222 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6472   Lon: 7.03873
13:11:08.540 MESSAGE comm_drv_n0183_net.cpp:448 TCP NetworkDataStream connection established: TCP:192.168.1.162:60001
13:11:28.222 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6461   Lon: 7.03606
13:11:40.222 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6461   Lon: 7.036
13:13:39.223 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6449   Lon: 7.03327
13:14:30.223 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6445   Lon: 7.03253
13:15:23.223 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6439   Lon: 7.03163
13:18:52.324 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6396   Lon: 7.02703
13:18:56.343 MESSAGE comm_drv_n0183_net.cpp:448 TCP NetworkDataStream connection established: TCP:192.168.1.162:60001
13:19:37.225 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6389   Lon: 7.02345
13:19:41.596 MESSAGE comm_drv_n0183_net.cpp:448 TCP NetworkDataStream connection established: TCP:192.168.1.162:60001
13:22:40.227 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6387   Lon: 7.01783
13:22:52.225 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6388   Lon: 7.01767
13:25:59.226 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6394   Lon: 7.01035
13:26:38.226 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6395   Lon: 7.0093
13:26:42.231 MESSAGE comm_drv_n0183_net.cpp:448 TCP NetworkDataStream connection established: TCP:192.168.1.162:60001
13:27:25.227 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6411   Lon: 7.00659
13:28:20.228 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6417   Lon: 7.0061
13:29:04.227 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6423   Lon: 7.00575
13:29:50.227 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6432   Lon: 7.00623
13:30:00.401 MESSAGE ocpn_frame.cpp:5692 LOGBOOK:  2025-07-14 11:30:00 UTC  GPS Lat   53.64350 Lon    7.00663 COG   41.05000 SOG   6.61 kts
13:30:26.228 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6439   Lon: 7.00725
13:32:52.229 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6466   Lon: 7.01399
13:33:31.229 MESSAGE comm_bridge.cpp:240    ***GPS Watchdog timeout at Lat:53.6471   Lon: 7.01538
13:33:39.204 MESSAGE comm_drv_n0183_net.cpp:448 TCP NetworkDataStream connection established: TCP:192.168.1.162:60001
```

The positions reported in the Watchdog records are changing, so the position is being received - so why is it timing out?

I also logged some data from the Monitor window but cannot tied the twp together easily as the timestamp on the logged data seems wrong.

```
# timestamp_format: EPOCH_MILLIS
# Created at: Mon Jul 14 13:30:08 2025
 
received_at,protocol,msg_type,source,raw_data
19:16:52.622 ↔ ✓ Internal Plugin ok Cursor-pos: 53c2b064,23N  7c2b01,29E
19:16:52.623 ↔ ✓ Internal Plugin ok Cursor-pos: 53c2b064,22N  7c2b01,29E
19:16:52.818 ⇤ ✓ Internal AppMsg ok appmsg::gps-watchdog
[... cut ...] "lost" : true,   "Type" : "Information",   "shipname" : "Unknown             ",   "hdg" : 511}
19:16:53.231 ↔ ✓ Internal Plugin ok AIS: {   "Msg" : "AIS Target",   "class" : 1,   "mmsi" : 244073839,   "callsign" : "    
```

In the above sample, logging was started at Jul 14 13:30:08 2025 into a new file but the timestamps on the records start at 19:16:52.622
Or am I missing something?
The `Cog nan Sog nan` in this sample is probably because that was recorded while the position was frozen.

Tony

bdbcat, 2025-08-20:
Tony...
Old thread, let's refresh the current state.
Do I understand the situation correctly?
1.  N2K connection to Actisense (through a router) in TCP mode.
2.  Occasionally OCPN just crashes, due to a problem in N2K driver shutdown.
3. The driver shutdown was not caused by any overt user interaction.
4.  Now suspect some latent problem in the router, which displeases OCPN.

All correct so far?

antipole2, 2025-08-21:
No.  See the connections screenshot in this thread above (8 days ago).  Only output is on the UDP port.

bdbcat, 2025-08-24:
Tony...
Finally got a chance to look at this again, and study the crash report above.
While the fault at first appears to be related to N2K driver, there is something more insidious happening.  Further down in the crash report, we see "exit+44".  This is telling us that OCPN is crashing for some as yet unknown reason, invoking the "exit" method.  The fault with N2K driver shutdown is incidental.
So, how to isolate the root cause?
Following standard debugging practices, the first think to do is to simplify the OCPN configuration as much as possible, while changing the external environment us little as possible.  The obvious target here is the connections config.
Is it possible to navigate and test with one single simple N0183 interface active, to obtain standard N0183 GNSS messages?

antipole2, 2025-08-25:
Dave... just done a long passage in easy conditions, so able to do some testing.  This was before your last, so not with that latest build.

(1) Ran with N2K input from NGT-1 via USB only. No other inputs or outputs - router not even powered on
Saw many of the closing N2K driver crashes - every hour or so.
Tried disabling all plugins apart from o-charts.  Also without route activated.  No difference.

(2) Then instead of the NGT-1, listened only on 60001 NMEA0183 input.  No crashes but it lost connection fairly often.
Then tried only from W2K-1 via router.   Both with NMEA0183 connection and N2K. Also lost connection at times.

(3) WifI direct from W2K-1 without going through router.  (a) N2K input OK - no problems.  (b) NMEA0183 input lost position once.

Conclusion: (1) The crash with NGT-1 is real and not a byproduct of the others 
(2) These errors could be because of my router and might be addressed by your latest build.  They do not result in a crash.
(3a) is solid and OK.  Possible problem with  (3b).

Will test with your latest build, but we have just completed our last passage for this season and it will be 'in harbour' only.

bdbcat, 2025-08-26:
I ran v5.12.2 with NGT-1 receiving live input over-night.  No other inputs. No faults.
Considering next steps....