OpenCPN/OpenCPN #4559: ECN chart text display squeezed

Chma-sail, 2025-05-29:
**Describe the bug**
ECN chart text is squished together. All attempts to adjust display screen width, font, chart scaling, etc., have no effect. Extensive search for solution results in no joy.
Running OpenCPN 5.10.2-0 on HP Elitebook 850 G3 with Windows 10 Pro ver. 2009. Using two displays but changing OpenCPN window from large display to small display makes no difference. All other text and scaling features appear to work as intended.

[image]

Hakansv, 2025-05-29:
Did you try:
Options->Charts->Vector chart display:  [x] De-Cluttered text ?

[image]

bdbcat, 2025-05-29:
What charts are these, please?

Chma-sail, 2025-05-30:
Sorry, did not mean to close - newbie error. Reopened.

bdbcat, 2025-05-30:
1.  Please try with "no-quilt" mode.  Use the "q" key.  Any difference?
2. Please tell us more about your displays.  Do you use two displays?
3. Please try with the lowest resolution display, and no other display enabled.
4. Are you using Windows Display Scaling?  If so, what scale value?

leamas, 2026-08-19:
Without more input this is not actionable. Closing.

Please add notes to the closed issue if it should be reopened.

closed 2026-08-19 by leamas