OpenCPN/OpenCPN #4565: Route Point HTML links not persisted, either during import or during creation/update nor retrieved

TwoCanPlugIn, 2025-06-02:
No corresponding INSERT  or UPDATE statement to add routepoints with a HTML link into the routepoint_html_links table, nor any SELECT statements to display HTML links in the corresponding UI.

Also, for the table definition I don't think there is a need to have a separate guid for the html link.
The current table definition is.
```
CREATE TABLE IF NOT EXISTS routepoint_html_links (
    guid TEXT PRIMARY KEY,
    routepoint_guid TEXT NOT NULL,
    html_link TEXT,
    html_description TEXT,
    html_type TEXT,
    FOREIGN KEY (routepoint_guid) REFERENCES routepoints(guid) ON DELETE CASCADE
);
```
I think it could be simplified by making the primary key also the foreign key reference.

```
CREATE TABLE IF NOT EXISTS routepoint_html_links (
    guid TEXT PRIMARY KEY,
    html_link TEXT,
    html_description TEXT,
    html_type TEXT,
    FOREIGN KEY (guid) REFERENCES routepoints(guid) ON DELETE CASCADE
);
```

bdbcat, 2025-06-03:
There may be multiple HTML links per Routepoint.

TwoCanPlugIn, 2025-06-03:
OK, understood. My apologies.

Dumb question, Why is "Description" duplicated in the Waypoint properties dialog?

bdbcat, 2025-06-03:
Close now?

closed 2025-06-04 by TwoCanPlugIn

nohal, 2025-06-04:
> I should always remind myself never to question the novel UI paradigms that OpenCPM has adopted.

Your constant bitching about this would perhaps be more productive if you a) explained what exactly you think is dement and b) provided an alternative that is not. But I have already told you this too many times, so should probably remind myself that it is a waste of time.