OpenCPN/OpenCPN #4567: Table definitions allow NULL values for GUID's

TwoCanPlugIn, 2025-06-02:
Seems to be a long acknowledged bug/omission with SQLite.  PRIMARY KEY implies NOT NULL, but it is not implemented for non INTEGER column types. Refer to the [fourth paragraph in Section 3.5,](https://sqlite.org/lang_createtable.html)

Suggest changing the table definitions to explicitly set the primary key to be NOT NULL (Discovered "accidentally" when trying to insert a route without a GUID just for the hell of it!)

```
CREATE TABLE IF NOT EXISTS routes (
            guid TEXT PRIMARY KEY NOT NULL,
            name TEXT,
            start_string TEXT,
            end_string TEXT,
            description TEXT,
            planned_departure TEXT,
            plan_speed REAL,
            time_format TEXT,
            style INTEGER,
            width INTEGER,
            color TEXT,
            visibility INTEGER,
            shared_wp_viz INTEGER,
            created_at DATETIME DEFAULT CURRENT_TIMESTAMP
        );

CREATE TABLE IF NOT EXISTS routepoints (
            guid TEXT PRIMARY KEY NOT NULL,
            lat REAL,
            lon REAL,
            Symbol TEXT,
            Name TEXT,
            description TEXT,
            TideStation TEXT,
            plan_speed REAL,
            etd INTEGER,
            Type TEXT,
            Time TEXT,
            ArrivalRadius REAL,
            RangeRingsNumber INTEGER,
            RangeRingsStep REAL,
            RangeRingsStepUnits INTEGER,
            RangeRingsVisible INTEGER,
            RangeRingsColour TEXT,
            ScaleMin INTEGER,
            ScaleMax INTEGER,
            UseScale INTEGER,
            visibility INTEGER,
            viz_name INTEGER,
            shared INTEGER,
            isolated INTEGER,
            created_at DATETIME DEFAULT CURRENT_TIMESTAMP
        );

CREATE TABLE IF NOT EXISTS tracks (
            guid TEXT PRIMARY KEY NOT NULL,
            name TEXT,
            description TEXT,
            visibility INTEGER,
            start_string TEXT,
            end_string TEXT,
            width INTEGER,
            style INTEGER,
            color TEXT,
            created_at DATETIME DEFAULT CURRENT_TIMESTAMP
        );
```
Also could improve the referential integrity by enforcing a foreign key on point_guid

 ```
CREATE TABLE IF NOT EXISTS routepoints_link (
            route_guid TEXT,
            point_guid TEXT,
            point_order INTEGER,
            PRIMARY KEY (route_guid, point_guid),
            FOREIGN KEY (route_guid) REFERENCES routes(guid) ON DELETE CASCADE,
            FOREIGN KEY (point_guid) REFERENCES routepoints(guid)
        );
```

And I'm wondering from a performance perspective (not that it would matter much), given that a route may have the same waypoint included multiple times, whether the PRIMARY_KEY should span route_guid and point_order columns?

Also do each of the html_link tables (track_html_links, route_html_links, routepoint_html_links) need to have two guids? Given that each row refers to a single respectve track, route or routepoint surely one guid which is the same as the corresponding guid is sufficient and the foreign key enforces the referential integrity.

```
CREATE TABLE IF NOT EXISTS track_html_links (
            guid TEXT PRIMARY KEY NOT NUL,
            html_link TEXT,
            html_description TEXT,
            html_type TEXT,
            FOREIGN KEY (guid) REFERENCES tracks(guid) ON DELETE CASCADE
        );
```
It might be too late (at least for the beta testers) nut why is there two time columns in routepoints ? It seems like at least during upgrade) that Time contains the original creation time of the routepoint from navobj.xml, and createdAt is the time that the object was inserted into navobj.db. I would have assumed that createdAt would reflect the time at which the routepoint was originally created in navobj.xml? Just because it has been defined as a default doesn't mean that you can't supply a value.

Finally for track_points, is there a need for a point order given that tracks are inserted in chronological order?

bdbcat, 2025-06-02:
Thanks for the review.
I'll implement the "NOT NULL" constraint.

bdbcat, 2025-06-05:
Thanks for the comments.
"as optimal as can be" is a pretty high bar.  I tend to design to a lowest common readable denominator, so that one does not need to be a guru to read and understand the code and structures.
So far in beta testing, we have had no performance or storage problems with db's of hundreds of MBytes.  Above a certain size, the db will always be smaller than an equivalent .XML representation, never mind the redundancy of the rolling backups we kept.
So I think we are there.
Closing.

closed 2025-06-05 by bdbcat