OpenCPN/OpenCPN #4572: GetpSharedDataLocation() returns NULL;

TwoCanPlugIn, 2025-06-04:
This plugin API is broken.

It is the underlying cause for the issue reported for [Dashboard Tactics on CF](https://www.cruisersforum.com/forums/f134/general-support-question-294709.html#post3999748)

I have also confirmed in my test plugin that it returns NULL; 

The workaround for Dashboard Tactics is perhaps to use GetPluginDataDir(plugin_name);

However at some stage GetpSharedDataLocation() must have worked, else DashboardTactics (and others) would have never worked.

rgleason, 2025-06-04:
Can we get this API call fixed? It is the most practical solution IMHO

leamas, 2025-06-04:
Of course it should be fixed. However, it's  kind of obsolete and I cannot see any usecase where it makes sense for a plugin and it sjhould probably be marked as deprecated.

@TwoCanPlugIn : If you need a location where plugin can store state you **must** use ` GetPluginDataDir(plugin_name)`, the plugin handler requires it. All sorts of interesting problems will occur otherwise.

bdbcat, 2025-06-04:
Indeed.
Respectfully, I can see no problems with GetpSharedDataLocation().
It is used in many existing plugins, like e.g. GRIB_pi and o-charts_pi.

bdbcat, 2025-06-04:
How may I exercise Dash-T to activate this code path related to influxDB?

TwoCanPlugIn, 2025-06-04:
Told you I wasn't a programmer.....

My guess then is that the problem with Rick's Dashboard Tactics is that it is looking in wrong directory for the  streamout_template.json file. It was looking in GetpSharedDataLocation() instead of the plugin's data directory.

Clearly the InfluxDB component of Dashboard Tactics has not been used in a long time, otherwise the problem would have been reported.

Absent a fix to Rick's plugin, the advice to the CF user may be to copy streamout_template.json into the SharedDataLocation.
I know that is c:\ProgramData\OpenCPN on Windows and ~/.opencpn on Linux, not sure what it would be on a Mac 

Anyway, I'll close this now.

Sorry for the panic attack.